espressif/esp-idf #10678: cannot find file: esp-idf/components/lwip/port/esp32/netif/dhcp_state.c

xobs, 2023-02-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3153-g2707c95a5f

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Projects should continue to build

### What is the actual behavior?

During a compile, the following error was emitted:

```
-- Resolving GIT Version
-- GIT hash: 853203a-dirty
-- Configuring done
CMake Error at C:/Users/Sean/esp/esp-idf/tools/cmake/component.cmake:479 (add_library):
  Cannot find source file:

    C:/Users/Sean/esp/esp-idf/components/lwip/port/esp32/netif/dhcp_state.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  C:/Users/Sean/esp/esp-idf/components/lwip/CMakeLists.txt:162 (idf_component_register)

CMake Error at C:/Users/Sean/esp/esp-idf/tools/cmake/component.cmake:479 (add_library):
  No SOURCES given to target: __idf_lwip
Call Stack (most recent call first):
  C:/Users/Sean/esp/esp-idf/components/lwip/CMakeLists.txt:162 (idf_component_register)

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1, output of the command is in the E:\Code\Farpatch\farpatch\build\log\idf_py_stderr_output_35924 and E:\Code\Farpatch\farpatch\build\log\idf_py_stdout_output_35924
```

### Steps to reproduce.

1. Attempt to build with the attached sdkconfig file
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/10563814/sdkconfig.zip)

### Build or installation Logs.

```plain
> idf.py menuconfig
Executing action: menuconfig
Running cmake in directory C:\Users\Sean\esp\esp-idf\examples\network\network_tests\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Users\Sean\.espressif\python_env\idf5.1_py3.10_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 C:\Users\Sean\esp\esp-idf\examples\network\network_tests"...
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.35.2.windows.1")
-- Component directory C:/Users/Sean/esp/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
[... 38 lines cut ...]
-- Configuring done
CMake Error at C:/Users/Sean/esp/esp-idf/tools/cmake/component.cmake:479 (add_library):
  Cannot find source file:

    C:/Users/Sean/esp/esp-idf/components/lwip/port/esp32/netif/dhcp_state.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  C:/Users/Sean/esp/esp-idf/components/lwip/CMakeLists.txt:162 (idf_component_register)

CMake Error at C:/Users/Sean/esp/esp-idf/tools/cmake/component.cmake:479 (add_library):
  No SOURCES given to target: __idf_lwip
Call Stack (most recent call first):
  C:/Users/Sean/esp/esp-idf/components/lwip/CMakeLists.txt:162 (idf_component_register)

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1
>
```
```

### More Information.

This just started in last night's nightly build, which means it was caused by a merge yesterday.

xobs, 2023-02-02:
To add: The sdkconfig is from an existing project -- specifically, from https://github.com/farpatch/farpatch/blob/main/sdkconfig.defaults

For testing, I copied this sdkconfig to `examples/network/network_tests` and ran `idf.py build`

xobs, 2023-02-03:
This appears to be caused when `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` is enabled.

The following patch works around the issue:

```patch
diff --git a/components/lwip/CMakeLists.txt b/components/lwip/CMakeLists.txt
index 22ceb1de84..671da81e20 100644
--- a/components/lwip/CMakeLists.txt
+++ b/components/lwip/CMakeLists.txt
@@ -156,7 +156,7 @@ if(CONFIG_LWIP_DHCPS)
 endif()

 if(CONFIG_LWIP_DHCP_RESTORE_LAST_IP)
-    list(APPEND srcs "port/esp32/netif/dhcp_state.c")
+    list(APPEND srcs "port/esp32xx/netif/dhcp_state.c")
 endif()

 idf_component_register(SRCS "${srcs}"
```

david-cermak, 2023-02-03:
@xobs Thanks for your report and finding the rootcase -- Please feel free to post your fix in https://github.com/espressif/esp-idf/pulls

An action point from our side is to extend our test configuration to cover this config  `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` as well

xobs, 2023-02-03:
Thanks @david-cermak -- I opened #10690

PR #10690 (2023-02-08): lwip: fix dhcp_server path (IDFGH-9309)
The path for dhcp_server.c has changed from esp32/ to esp32xx/

This fixes the build when CONFIG_LWIP_DHCP_RESTORE_LAST_IP is enabled, and closes #10678
closed 2023-02-08 by espressif-bot