espressif/esp-idf #10680: esp_http_client  get 请求https 域名返回200 获取长度为0

Wvirgil123, 2023-02-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

环境：esp32s3 + esp-idf 5.0 
问题：使用正确的证书情况下，使用 esp_http_client  api   请求 www.timeapi.io 域名返回200 但获取长度为0。

代码：
`
static void http_redirect_to_https(void)
{
    esp_http_client_config_t config = {
        .url = "https://www.timeapi.io/api/TimeZone/ip?ipAddress=237.71.232.203",
        .event_handler = _http_event_handler,
        .cert_pem = howsmyssl_com_root_cert_pem_start,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    esp_err_t err = esp_http_client_perform(client);
    if (err == ESP_OK) {
        ESP_LOGI(TAG, "HTTP redirect to HTTPS Status = %d, content_length = %lld",
                esp_http_client_get_status_code(client),
                esp_http_client_get_content_length(client));
    } else {
        ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
    }
    esp_http_client_cleanup(client);
}
`
输出结果：
`
(9899) HTTP_CLIENT: HTTP redirect to HTTPS Status = 200, content_length = -1
(9899) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
`

正确情况:

[image]

[image]

ESP-YJM, 2023-02-02:
@Wvirgil123 这个是 chunked 的数据，没有 content-length 字段，你工具上也显示的是 chunked 的。

Wvirgil123, 2023-02-02:
> @Wvirgil123 这个是 chunked 的数据，没有 content-length 字段，你工具上也显示的是 chunked 的。

我打印 chunk_length  也是为0， 通过perform 方式可以获取到正确的 chunk 长度和数据吗？

> if (err == ESP_OK) {
>     ESP_LOGI(TAG, "HTTP GET Status = %d, content_length = %lld",
>             esp_http_client_get_status_code(client),
>             esp_http_client_get_content_length(client));
>     int len = 0;
>     bool ret = esp_http_client_is_chunked_response(client);
>     err = esp_http_client_get_chunk_length(client, &len);
>     ESP_LOGI(TAG, "is_chunked:%d, chunk_length:%d, ret:%d", ret, len, err);
> } else {
>     ESP_LOGE(TAG, "HTTP GET request failed: %s", esp_err_to_name(err));
>     return -1;
> }  

响应结果：
I (30804) : HTTP GET Status = 200, content_length = -1
I (30804) : is_chunked:1, chunk_length:0, ret:0

ESP-YJM, 2023-02-02:
不可以，chunk 的数据就是无法获取长度的。

Wvirgil123, 2023-02-02:
> 不可以，chunk 的数据就是无法获取长度的。

https_request_example_main.c   demo 中的方式可以获取到chunk 数据吗

ESP-YJM, 2023-02-02:
可以的

Wvirgil123, 2023-02-02:
> 可以的

esp_tls_conn_read 函数 获取的数据中， body 多了128 和0 两个数字，这个是bug 吗

[image]

[image]

ESP-YJM, 2023-02-02:
能看下你打印这个串口日志的地方吗？

Wvirgil123, 2023-02-02:
> 

[image]

ESP-YJM, 2023-02-02:
`esp_tls_conn_read` 读的是包含 http 头和 http body 的原始数据，而浏览器上显示的是经过 http 报文编码解析后的数据。前者把原始数据转化为 str 时是会打印出 128 和 0，需要将原始数据通过 http 编码后打印出来就没有 128 和 0 了。建议你使用 `esp_http_client `组件，该组件里会把 http 原始数据进行编码，就没有问题了。所以这个不是 bug。

closed 2023-02-15 by Alvin1Zhang