espressif/esp-idf #10682: Example MQTT SSL-DS: openssl.exe not included with ESP-IDF on Windows

profeldoo, 2023-02-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I expect "openssl" command to work out of the box with ESP-IDF 5.0 CMD (i.e., to be in "PATH").

### What is the actual behavior?

Within example found at https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ssl_ds the user needs to call the openssl program (from the command line) to create an RSA certificate. However, the command is not recognized. If we type "where openssl", we get nothing. "openssl.exe" normally comes bundled with Git, but IDF's embedded version lacks it.
[image]

### Steps to reproduce.

1. Open ESP-IDF 5.0 CMD
2. Type in "openssl help".
3. Get nothing.

AdityaHPatwardhan, 2023-02-20:
Hi @profeldoo The generic dependencies which are used by any core IDF component are usually added in the build for the env.
But in this case, the `openssl` dependency is only used in this particular example.(Or a couple of other examples).
It just increases the build size for all the users who are probably never going to use that.
This dependency is only applicable to the example, Hence it is not included in the default IDF build.

AdityaHPatwardhan, 2023-02-27:
I think my answer above should answer the query
I will go ahead and close the issue, Please feel free to re-open it if required.
Thanks,
Aditya

closed 2023-02-27 by AdityaHPatwardhan