espressif/esp-idf #10683: ESP32S3-WROOM-1 FLASH and PSRAM Mode configuration are not supported

Mbded, 2023-02-02:
Helo.  I try lunch little project with usage  **SPI connected RAM** but during build project I get some error each tell me about mismatched spped clk CPU and SPI. I tried a few combination but effect is always this same.
Below mu setup parameters, and I also attachment sdkconfig

```
ENV
ESP_IDF 4.4.3
```
```
module:
ESP32S3-WROOM-1 N8R8 
```
Log form build application

`====================[ Build | app | Debug ]=====================================
/home/rafal/clion/bin/cmake/linux/bin/cmake --build /home/rafal/esp/esp-idf/examples/peripherals/i2c/i2c_simple/cmake-build-debug --target app -- -j 8
[  0%] Built target memory_ld
[  1%] Built target partition_table_bin
[  1%] Built target _project_elf_src
[  1%] Built target model
[  1%] Built target custom_bundle
[  2%] Built target __idf_ulp
[  2%] Built target __idf_esp_serial_slave_link
[  3%] Built target __idf_sdmmc
[  3%] Built target __idf_esp_https_ota
[  3%] Built target __idf_esp_http_server
[  4%] Built target __idf_esp_http_client
[  4%] Built target __idf_tcp_transport
[  5%] Built target __idf_esp-tls
[  7%] Built target __idf_nghttp
[  7%] Built target __idf_app_trace
[  7%] Built target __idf_cxx
[  8%] Built target __idf_newlib
[ 10%] Built target __idf_freertos
[ 11%] Built target __idf_esp_timer
[ 11%] Built target __idf_esp_common
[ 12%] Built target __idf_xtensa
[ 15%] Built target __idf_esp_hw_support
[ 17%] Built target __idf_soc
[ 17%] Built target __idf_heap
[ 17%] Built target __idf_log
[ 24%] Built target __idf_lwip
[ 25%] Built target __idf_console
[ 26%] Built target __idf_esp_wifi
[ 32%] Built target __idf_wpa_supplicant
[ 32%] Built target __idf_esp_event
[ 33%] Built target __idf_esp_netif
[ 33%] Built target __idf_tcpip_adapter
[ 34%] Built target __idf_esp_eth
[ 34%] Built target __idf_vfs
[ 38%] Built target __idf_hal
[ 39%] Built target __idf_esp_rom
[ 41%] Built target __idf_esp_system
[ 41%] Built target __idf_esp_phy
[ 41%] Built target __idf_espcoredump
[ 42%] Built target __idf_esp_gdbstub
[ 42%] Built target __idf_pthread
[ 43%] Built target __idf_nvs_flash
[ 43%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/esp32s3/spi_timing_config.c.obj
[ 43%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_chip_boya.c.obj
[ 43%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_chip_th.c.obj
[ 43%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/memspi_host_driver.c.obj
[ 43%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_chip_mxic_opi.c.obj
[ 43%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_chip_winbond.c.obj
[ 43%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/cache_utils.c.obj
[ 43%] Building C object esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_timing_tuning.c.obj
In file included from /home/rafal/esp/esp-idf/components/spi_flash/esp32s3/spi_timing_config.c:15:
/home/rafal/esp/esp-idf/components/spi_flash/esp32s3/spi_timing_config.h:174:1: error: static assertion failed: "FLASH and PSRAM Mode configuration are not supported"
 _Static_assert(CHECK_POWER_OF_2(SPI_TIMING_CORE_CLOCK_MHZ / SPI_TIMING_PSRAM_MODULE_CLOCK), "FLASH and PSRAM Mode configuration are not supported");
 ^~~~~~~~~~~~~~
esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/build.make:88: polecenia dla obiektu 'esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/esp32s3/spi_timing_config.c.obj' nie powiodły się
make[3]: *** [esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/esp32s3/spi_timing_config.c.obj] Błąd 1
make[3]: *** Oczekiwanie na niezakończone zadania....
In file included from /home/rafal/esp/esp-idf/components/spi_flash/spi_flash_timing_tuning.c:19:
/home/rafal/esp/esp-idf/components/spi_flash/esp32s3/spi_timing_config.h:174:1: error: static assertion failed: "FLASH and PSRAM Mode configuration are not supported"
 _Static_assert(CHECK_POWER_OF_2(SPI_TIMING_CORE_CLOCK_MHZ / SPI_TIMING_PSRAM_MODULE_CLOCK), "FLASH and PSRAM Mode configuration are not supported");
 ^~~~~~~~~~~~~~
esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/build.make:101: polecenia dla obiektu 'esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_timing_tuning.c.obj' nie powiodły się
make[3]: *** [esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/spi_flash_timing_tuning.c.obj] Błąd 1
CMakeFiles/Makefile2:3992: polecenia dla obiektu 'esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/all' nie powiodły się
make[2]: *** [esp-idf/spi_flash/CMakeFiles/__idf_spi_flash.dir/all] Błąd 2
CMakeFiles/Makefile2:2651: polecenia dla obiektu 'CMakeFiles/app.dir/rule' nie powiodły się
make[1]: *** [CMakeFiles/app.dir/rule] Błąd 2
Makefile:203: polecenia dla obiektu 'app' nie powiodły się
make: *** [app] Błąd 2
`

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10568592/sdkconfig.txt)

Icarus113, 2023-02-03:
Hi @Mbded , the ESP32S3-WROOM-1 N8R8  is a `F4R8` board. For `F4R8`, you can refer to [this guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html). There's a table illustrating supported Flash / PSRAM mode / freq match.

Icarus113, 2023-02-06:
Hi @Mbded , the ESP32S3-WROOM-1 N8R8 is a module with a quad flash and an octal PSRAM. So you should not enable `CONFIG_ESPTOOLPY_FLASHMODE_OPI=y`. 

FYI a newer IDF version will support automatically detect if the flash is a quad one or an octal one. (Should be released in v5.1). So that time if you're not familiar with the flash devices, the driver will auto-detect a default one.

Mbded, 2023-02-07:
solved

closed 2023-02-07 by Mbded