espressif/esp-idf #10686: How to fix a xTaskPriorityDisinherit

f-hoepfinger-hr-agrartechnik, 2023-02-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Bad boot
https://gist.github.com/f-hoepfinger-hr-agrartechnik/7295e71d664265988b65d3499ad780c1

Good Boot
https://gist.github.com/f-hoepfinger-hr-agrartechnik/1521c58baebf7244ed539477a5c19f27

in 1 of 10 Starts a bad boot happens ... it it happens it only recovers after 3 resets most times. 
after this it happens not for 7-10 resets and so on. 

My Question: 
how to dig deeper in such a Problem? 

```
assert failed: xTaskPriorityDisinherit tasks.c:4976 (pxTCB == pxCurrentTCB[xPortGetCoreID()])

Backtrace: 0x4008244a:0x3ffc89e0 0x40094425:0x3ffc8a00 0x4009a45d:0x3ffc8a20 0x40096dee:0x3ffc8b40 0x4009480d:0x3ffc8b60 0x40094dcc:0x3ffc8b80 0x40083611:0x3ffc8bc0 0x400836c1:0x3ffc8bf0 0x40083732:0x3ffc8c10 0x401ff0b1:0x3ffc8c40 0x401f681c:0x3ffc8c60 0x40220100:0x3ffc8c80 0x401310d2:0x3ffc8cf0 0x400db5eb:0x3ffc8d40 0x400db2aa:0x3ffc8dd0 0x400daef3:0x3ffc8e10 0x4021fd5f:0x3ffc8e30 0x40097f56:0x3ffc8e60
0x4008244a: panic_abort at /home/franz/git/hr/esp-idf/components/esp_system/panic.c:412

0x40094425: esp_system_abort at /home/franz/git/hr/esp-idf/components/esp_system/esp_system.c:135

0x4009a45d: __assert_func at /home/franz/git/hr/esp-idf/components/newlib/assert.c:78

0x40096dee: xTaskPriorityDisinherit at /home/franz/git/hr/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4976 (discriminator 1)

0x4009480d: prvCopyDataToQueue at /home/franz/git/hr/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2261

0x40094dcc: xQueueGenericSend at /home/franz/git/hr/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:861

0x40083611: lock_release_generic at /home/franz/git/hr/esp-idf/components/newlib/locks.c:192

0x400836c1: _lock_release at /home/franz/git/hr/esp-idf/components/newlib/locks.c:198

0x40083732: __retarget_lock_release at /home/franz/git/hr/esp-idf/components/newlib/locks.c:334

0x401ff0b1: __tz_unlock at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/time/tzlock.c:50

0x401f681c: localtime_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/time/lcltime_r.c:123

0x40220100: esp_log_system_timestamp at /home/franz/git/hr/esp-idf/components/log/log_freertos.c:84

0x401310d2: AppTask::AbstractTask::create_config(char const*, int, int, int, unsigned long) at /home/franz/git/hr/CTIS2/HR-Agrar-ISO-OS/components/AppTask/src/AbstractTask.cpp:62

0x400db5eb: App_Main::isobus_main() at /home/franz/git/hr/CTIS2/HR-Agrar-ISO-OS/components/AppThreading/src/App_Main.cpp:113

0x400db2aa: start_app() at /home/franz/git/hr/CTIS2/HR-Agrar-ISO-OS/components/AppMain/src/AppMain.cpp:319

0x400daef3: app_main at /home/franz/git/hr/CTIS2/HR-Agrar-ISO-OS/main/main.cpp:14

0x4021fd5f: main_task at /home/franz/git/hr/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x40097f56: vPortTaskWrapper at /home/franz/git/hr/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

```

ESP-Marius, 2023-02-03:
@f-hoepfinger-hr-agrartechnik Would you mind trying to apply the commit from https://github.com/espressif/esp-idf/pull/10649 and see if this fixes your issue?

f-hoepfinger-hr-agrartechnik, 2023-02-03:
> @f-hoepfinger-hr-agrartechnik Would you mind trying to apply the commit from #10649 and see if this fixes your issue?

tnx. 

i will try in this moment.

f-hoepfinger-hr-agrartechnik, 2023-02-08:
Yes, Problem seems solved !

closed 2023-02-08 by f-hoepfinger-hr-agrartechnik