espressif/esp-idf #10687: (help) delay time for web server communication doesn't go down below 30 ms

Inverter-Jo, 2023-02-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

kit : ESP32_WROOM_32E
idf version : v5.0

Based on ws_echo_server example.
The following changes have been made:

esp_wifi_set_ps(false);
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED = y -> false;
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED = y -> false;
CONFIG_HTTPD_ERR_RESP_NO_DELAY= false

and i make the client a local client with index_html.

I thought the delay time for web server communication would be less than 10 ms.

but COMMUNICATION LATENCY doesn't go down below 30 ms.

How can I make it's latency 10ms down?

chipweinberger, 2023-02-03:
30ms is pretty good for wifi.

- add more logs to determine when the request handler is called, and when it completes

- find the https task and increase it's priority

Inverter-Jo, 2023-02-05:
We are conducting a simple test of connecting from a web client, sending packets to the server (esp32) at 24 ms intervals through settimeout, and receiving callbacks.

Unintentional, but now there is one more packet between the packets being sent and received from the client's point of view.

Below are the results of reading in wireshark and leaving end as log at the end of start at the beginning of handler.
Even when the transmission interval is changed, more transmission packets are placed between the transmission packet and the received packet.

What's the problem?
[image]
[image]

chipweinberger, 2023-02-06:
Oh I just noticed you are using websockets.

In the WS echo example you should put logs here:

- *after* `esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);`
- *before* `httpd_ws_send_frame_async` 
- *before* `sess->send_fn` (inside the `httpd_ws_send_frame_async` function)

Inverter-Jo, 2023-02-06:
@chipweinberger
Here are the WIRESHARK results and LOG that apply what you said.

CHECKED//
1. DISABLE DEBUG LOGS
2. log
  FIRST : after esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
  SECOND : before httpd_ws_send_frame_async
  THIRD : before sess->send_fn (inside the httpd_ws_send_frame_async function)
  
 wireshark
[image]

 
 log
 I (99702) websocket_server: byte_array_handler Handshake done, the new connection was opened
I (100510) websocket_server: FIRST
I (100510) websocket_server: SECOND
I (100510) httpd_ws: THIRD
I (100514) websocket_server: FIRST
I (100514) websocket_server: SECOND
I (100518) httpd_ws: THIRD
I (100522) websocket_server: FIRST
I (100526) websocket_server: SECOND
I (100530) httpd_ws: THIRD
I (100534) websocket_server: FIRST
I (100538) websocket_server: SECOND
I (100542) httpd_ws: THIRD
I (100546) websocket_server: FIRST
I (100550) websocket_server: SECOND
I (100554) httpd_ws: THIRD
I (100558) websocket_server: FIRST
I (100562) websocket_server: SECOND
I (100566) httpd_ws: THIRD
I (100570) websocket_server: FIRST
I (100574) websocket_server: SECOND
I (100578) httpd_ws: THIRD
I (100582) websocket_server: FIRST
I (100586) websocket_server: SECOND
I (100590) httpd_ws: THIRD
I (100594) websocket_server: FIRST
I (100598) websocket_server: SECOND
I (100602) httpd_ws: THIRD
I (100610) websocket_server: FIRST
I (100610) websocket_server: SECOND
I (100610) httpd_ws: THIRD
I (104326) httpd_ws: THIRD
I (106530) websocket_server: byte_array_handler Handshake done, the new connection was opened
I (107734) websocket_server: FIRST
I (107734) websocket_server: SECOND
I (107734) httpd_ws: THIRD
I (107734) websocket_server: FIRST
I (107738) websocket_server: SECOND
I (107742) httpd_ws: THIRD
I (107746) websocket_server: FIRST
I (107750) websocket_server: SECOND
I (107754) httpd_ws: THIRD
I (107766) websocket_server: FIRST
I (107766) websocket_server: SECOND
I (107766) httpd_ws: THIRD
I (107786) websocket_server: FIRST
I (107786) websocket_server: SECOND
I (107786) httpd_ws: THIRD
I (107814) websocket_server: FIRST
I (107814) websocket_server: SECOND
I (107814) httpd_ws: THIRD
I (107834) websocket_server: FIRST
I (107834) websocket_server: SECOND
I (107838) httpd_ws: THIRD
I (107862) websocket_server: FIRST
I (107862) websocket_server: SECOND
I (107862) httpd_ws: THIRD
I (107886) websocket_server: FIRST
I (107886) websocket_server: SECOND
I (107886) httpd_ws: THIRD
I (107914) websocket_server: FIRST
I (107914) websocket_server: SECOND
I (107914) httpd_ws: THIRD

chipweinberger, 2023-02-06:
well those times are all pretty good!

could be delay from your router!

are you using STA or AP wifi mode?

id try AP mode

Inverter-Jo, 2023-02-07:
Thank you for your response.

Proceeding to Sta.
Even when rssi was experimented with -15 dB, it did not go below 25 ms.
Then can't the minimum communication delay go below 20 ms?

- [are you using STA or AP wifi mode? ] 
Will the mode affect the delay time?

chipweinberger, 2023-02-07:
In STA mode your home wifi router will add additional delay, yes.

Inverter-Jo, 2023-02-08:
@chipweinberger
Thank you for your continued answer to the question.

I lack understanding of wifi station mode and wifi ap mode.
Is it possible to mix ws_echo_server example and wifi_ap_mode?
Could you give me a rough example?

chipweinberger, 2023-02-08:
Yes, you can mix the code. 

First you need to create your wifi access point (AP) mode, then start the http server after you do that.

Inverter-Jo, 2023-02-09:
@chipweinberger

It's sad, but the device that contains the code I make needs to access the Wi-Fi.

Can you tell me how much LATENCY is when you set it to AP MODE?

Let me close this topic with this question.

Thank you for answering our questions!

hmalpani, 2023-02-13:
Hello @Inverter-Jo
commented [here](https://github.com/espressif/esp-idf/issues/10665#issuecomment-1427642662) with the suggestion.

hmalpani, 2023-02-16:
Closing this issue. Feel free to reopen

closed 2023-02-16 by hmalpani