espressif/esp-idf #10688: Why is "esp_netif_sntp.h" missing from v5.0 release?

pback34, 2023-02-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

esp_netif_sntp.h should be in /components/esp_netif/include  

### What is the actual behavior?

File is not there. I am wondering why this file is missing from this version tag, when it is in the latest docs.

### Steps to reproduce.

checkout v5.0 tag

pback34, 2023-02-03:
i see for some reason when i checkout v5 in the docs, this is also not mentioned. I'm now confused though as to how to best use the sntp feature.

shentu, 2023-02-06:
Maybe use "esp_sntp.h" instead.

david-cermak, 2023-03-24:
If you're using IDF `v5.0`, please check the docs for this particular version: https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/index.html

`esp_netif_sntp.h` was only added into `v5.1`. Please use `"esp_sntp.h"` on previous releases.