espressif/esp-idf #10692: How Do I upload a bootloader to a brand new ESP32-S3 chip on my newly designed and produced Board?

SenelCanik, 2023-02-03:
Hello Everyone,

We made our own board with ESP32-S3 (FN8 Model with 8 MB Flash internal). We see bootloop when we first turn it on. After looading bootloader.bin (and many different *.bin files) we still see bootloop problem and we get the following messages from COM port. We use Windows CMD for esptool.exe to load *.bin files.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1634
load:0x403b6000,len:0xe74
load:0x403ba000,len:0x31c8
Checksum failure. Calculated 0x8e stored 0xde
ets_main.c 329
... (loops) ...

How can we solve this problem?

Any help is appreciated.

Thank You

shentu, 2023-02-06:
1. Check the firmware bins size. Does the size match the partition table?
2. Try earse the flash before flashing.
3. Try change `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL`.

Otherwise, try searching 'Checksum failure' on github, can get lots of help.

SenelCanik, 2023-02-06:
Thankyou Shentu.

1) I think, there should not be size problem. Because we have this problem even after I erase all flash.

2) I erased the flash:
------------------------------
C:\Pub\BootLoader>esptool -p COM39 --chip esp32s3 erase_flash
esptool.py v4.4
Serial port COM39
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:68:2d:98
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.9s
Hard resetting via RTS pin...

When I check serial port COM39:
-------------------------------------
invalid header: 0xESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048839
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
--------(loops)----------

When I checked Device Manager, COM39 disaappear and comes back with a sound from pc speaker continiously.

3) Can you help me how to change CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL?

By the way we have two flash on the board. First one is 8 MB internal flash of ESP32-S3 FN8 Model Chip. Second one is external 16 MB (W25Q128JVSIQ) beside the ESP32-S3 chip on the board. Is this may be source of problem?

Thank you very much for helping.

Senel

mythbuster5, 2023-02-15:
May I know what flash you are using? You can get this information via `esptool.py flash_id`

closed 2023-08-23 by Alvin1Zhang