espressif/esp-idf #10696: Websocket client does not allow sending 0 byte packets

haakonnessjoen, 2023-02-04:
https://github.com/espressif/esp-idf/blob/e8bdaf91986a41678adc6be13888fc037b1acb68/components/esp_websocket_client/esp_websocket_client.c#L849

This line prevents the websocket client from sending 0 byte packets, which should be valid, and some implementations use this feature for ping/pong packets, reducing overhead by having no data in the packet.

This seems to be present in all versions of esp-idf, up until 4.4.4. (I cannot find websocket client in 5.x)

Alvin1Zhang, 2023-06-20:
Related PR was merged, close the issue, feel free to reopen. Thanks.

closed 2023-06-20 by Alvin1Zhang

AxelLin, 2023-06-20:
@Alvin1Zhang Just remind that v4.x branches do not include this fix.

komagroup, 2024-06-27:
> @Alvin1Zhang Just remind that v4.x branches do not include this fix.

Also on 4.4.7 there is no `esp_websocket_client_send_with_opcode` so I can't even send ping or pong.