espressif/esp-idf #10698: Panic! Using a SPI LCD and CONFIG_SPI_MASTER_ISR_IN_IRAM=y (IDF 4.4.4)

kriegste, 2023-02-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I use CONFIG_SPI_MASTER_ISR_IN_IRAM=y (which is default) and I have an LCD connected to the SPI bus. It is initialized with
> intr_flags = ESP_INTR_FLAG_IRAM

When updating my firmware, the flash access triggers a

> Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

> 0x40162e5d: spi_lcd_prepare_cmd_buffer at esp_lcd_panel_io_spi.c:?
> 0x40084842: spi_intr at esp-idf/components/driver/spi_master.c:613
> 0x40082ff5: _xt_lowint1 at esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1114
> 0x40085dca: spi_flash_op_block_func at esp-idf/components/spi_flash/cache_utils.c:123
> 0x40083771: ipc_task at esp-idf/components/esp_ipc/src/esp_ipc.c:77

So the LCD functions are not IRAM_ATTR. Are we not allowed to put the SPI ISR into IRAM when an LCD is used?
https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/peripherals/lcd.html
does not mention this limitation. Maybe there should be a warning?

Instead of getting the ISR out of IRAM, I'd rather put all the necessary LCD stuff in there, too.
Which LCD functions do I need to make IRAM_ATTR? I don't want to rely on trial and error. This is why I am asking.

kriegste, 2023-02-06:
If possible, the best solution would be an sdkconfig flag to put the necessary LCD functions into IRAM.

suda-morris, 2023-02-09:
Hi @kriegste 

> When updating my firmware, the flash access triggers a
Guru Meditation Error: Core 0 panic'ed (Cache disabled but cached memory region accessed).

This indicates that when you doing OTA which will disable the Flash cache, functions (e.g. this `spi_lcd_prepare_cmd_buffer)` that are put in the Flash are not available at that moment. SPI LCD usually has the GRAM to hold the frame buffer,  it can keep the screen showing a static image when you doing OTA. But if  you want to display an animation when doing OTA, then that would need to put a lot of functions (e.g. `panel_io_spi_tx_color`) into the IRAM, which is not economic IMO.

But if you're working with ESP32S3 with PSRAM attached, an alternative approach to work around that issue is to enable the following two options:

```
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y
```

suda-morris, 2025-03-10:
Alterntaively, we recommend to enable the `CONFIG_SPI_FLASH_AUTO_SUSPEND`. By this, you don't have to place a bunch of functions into IRAM while the cache also won't be disabled during flash erase. Then you can do e.g. "showing a progress bar of the OTA" on your screen.

closed 2025-03-10 by suda-morris