espressif/esp-idf #10701: arduino-as-component

RodrigoRamosRibeiro, 2023-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

fatal error: Arduino.h: No such file or directory
    1 | #include "Arduino.h"
      |          ^~~~~~~~~~~
compilation terminated.

#include "Arduino.h"

extern "C" void app_main()
{
    initArduino();
    pinMode(4, OUTPUT);
    digitalWrite(4, HIGH);
    // Do your own thing
}

Test on Windows 10, pop OS, and Mint.

igrr, 2023-02-06:
Hi @RodrigoRamosRibeiro,

Could you please add more information to help us troubleshoot this issue:

- What steps have followed to set up the project with Arduino-esp32 as a component?
- Which versions of ESP-IDF and arduino-esp32 are you using? You can run `git describe` command in ESP-IDF and arduino-esp32 directories to get this.
- What is the content of your project's CMakeLists.txt?
- What is the content of main/CMakeLists.txt in your project?
- Please run `idf.py fullclean`, then `idf.py build` and attach the entire output you get from that command. (Preferably, as a text file.)

RodrigoRamosRibeiro, 2023-02-06:
[logs.txt](https://github.com/espressif/esp-idf/files/10669331/logs.txt)
New project - "Arduino-as-component" template-compile.
idf 5.0
CMakeLists.txt project 👍 
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(project-name-teste)

main/CMakeLists.txt

idf_component_register(
    SRCS "main.cpp"
    INCLUDE_DIRS ""
)

I made a video and it is linked below:
https://youtu.be/ECQuXF6eFtY
pop os

igrr, 2023-02-06:
The main issue, as far as I can tell, is your project doesn't include and doesn't have any reference to arduino-esp32 component.

You can confirm it by looking at "Components: " and "Component paths: " lines in logs.txt. With arduino-esp32 component added successfully, you should see its name in that list. Please refer to the docs for using arduino-esp32 as an IDF component: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html.

The second issue which you aren't running into, yet, is that you are trying to use arduino-esp32 with IDF v5.0. At the moment, arduino-esp32 project uses IDF v4.4.3, and will not compile successfully with IDF v5.x. So you will need to use IDF v4.4.x if you'd like to use arduino-esp32 as an IDF component.

georgemichalis, 2023-02-09:
i am running various projects with arduino as a component. It is as @igrr said, you have to register the component, and also have both the arduino as well as the esp-idf on the correct version. 
This is not exactly trivial, because lets say that you are developing something on esp-idf 5 and another project on the esp-idf 4, you would have to have both of them installed. To make things worse, sometimes the idf version that the arduino build was based it wasn't a specific release of idf but a random commit. This was at the time of arduino 1.0.6, so i am not sure if this has changed right now.  To avoid this you have to also use the esp-idf as a component on the project folder.

In any case for the instructions i assume:

1. that you have installed the ESP-IDF 4.4.3. 
2. that you are using git

To make it clearer you have to do the following steps. 

1.  Download arduino on the project folder. 
`mkdir -p components && \
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \
cd ../..
`
2. add the arduino as a submodule. 
`git submodule add 'components/arduino'`

3. Change the arduino in the correct commit. By default you got the master branch, which is probably not the correct version. Go to the folder of arduino and checkout the 2.0.6 tag, then another `git submodule update --init --recursive`

4. commit your project changes on git, and now the submodule should be on the correct version from now on. 
5. run `idf.py menuconfig` to configure the project. You need to enable some settings for the arduino to compile. This may be inaccurate nowdays and this step is probably NOT needed, because last time i tried that i was still on 1.0.6, but better to give you all the information. 

 - `Arduino Configuration / Autostart arduino setup and loop on boot`
- `Component configuration / MbedTLS / TLS key exchange methods / Enable preshared-key-ciphersuits / enable PSK based Ciphersuit modes`

- change on the component configuration/freertos/tick rate and set it ot 1000
 because delay() function will only work in multiples of 10ms. That is because, by default, esp-idf handles task events 100 times per second. To fix that behavior you need to set FreeRTOS tick rate to 1000Hz in `Component config -> FreeRTOS -> Tick rate.`

- enable on `Component configuration/bluetooth the bluetooth module.`

6. finally on the CMakeList.txt you need to have something equivalent:
```
idf_component_register(SRCS "whatever.c"
 INCLUDE_DIRS "."
 REQUIRES arduino)
```

7. Any arduino libraries you want to add need a similar approach + a make file
for example a component of the "Adafruit sensor" needs the following in its own `CMakeLists.txt`
```
idf_component_register(SRCS "Adafruit_Sensor.cpp"
                       INCLUDE_DIRS "."
                       REQUIRES arduino)
```
8. Finally, this applies only the normal development boards. if you have anything from another manufacturer (for example maybe an esp32 firebeetle from dfrobot, you also need to replace the `arduino/variants/esp32/pins_arduino.h` file from the file of the correct variant. 

if you do this you need to edit the `.gitmodule` file to set the submodule status as `dirty` for the arduino
```
[submodule "components/arduino"]
	path = components/arduino
	url = https://github.com/espressif/arduino-esp32.git
	ignore = dirty
```

Good luck. Personally after messing a while with all that (and some other issues that i had with the arduino wrappers). , i found it easier to just rewrite any driver to pure esp-idf and avoid arduino all together.

RodrigoRamosRibeiro, 2023-02-09:
run idf.py menuconfig does not work until manually change sdkconfig to:CONFIG-FREERTOS-HZ=1000

closed 2024-05-10 by igrr