espressif/esp-idf #10702: unity example project does not work properly

real-hao, 2023-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I try to  run `unity_test/test` under the example folder and didn't modify any code. The expected behavior shown in the `unity_test/README.md` is
```
#### Executing one test by its name #####

Running Mean of an empty array is zero...
/Users/ivan/e/esp-idf/examples/system/unit_test/components/testable/test/test_mean.c:16:Mean of an empty array is zero:PASS

-----------------------
1 Tests 0 Failures 0 Ignored 
OK

#### Running tests with [mean] tag #####

Running tests matching '[mean]'...
Running Mean of an empty array is zero...
/Users/ivan/e/esp-idf/examples/system/unit_test/components/testable/test/test_mean.c:16:Mean of an empty array is zero:PASS
Running Mean of a test vector...
/Users/ivan/e/esp-idf/examples/system/unit_test/components/testable/test/test_mean.c:22:Mean of a test vector:PASS
Running Another test case which fails...
/Users/ivan/e/esp-idf/examples/system/unit_test/components/testable/test/test_mean.c:35:Another test case which fails:FAIL: Expected 2147483647 Was -1

-----------------------
3 Tests 1 Failures 0 Ignored 
FAIL

#### Running tests without [fails] tag #####

Running tests NOT matching '[fails]'...
Running Mean of an empty array is zero...
/Users/ivan/e/esp-idf/examples/system/unit_test/components/testable/test/test_mean.c:16:Mean of an empty array is zero:PASS
Running Mean of a test vector...
/Users/ivan/e/esp-idf/examples/system/unit_test/components/testable/test/test_mean.c:22:Mean of a test vector:PASS

-----------------------
2 Tests 0 Failures 0 Ignored 
OK

#### Running all the registered tests #####

Running Mean of an empty array is zero...
/Users/ivan/e/esp-idf/examples/system/unit_test/components/testable/test/test_mean.c:16:Mean of an empty array is zero:PASS
Running Mean of a test vector...
/Users/ivan/e/esp-idf/examples/system/unit_test/components/testable/test/test_mean.c:22:Mean of a test vector:PASS
Running Another test case which fails...
/Users/ivan/e/esp-idf/examples/system/unit_test/components/testable/test/test_mean.c:35:Another test case which fails:FAIL: Expected 2147483647 Was -1

-----------------------
3 Tests 1 Failures 0 Ignored 
FAIL
```

### What is the actual behavior?

However, it unable to run unit test normally and seems to be unable to find `TEST_CASE` in the component `testable`. In addtion, I didn't modify any code in folder `unity_test`.
[image]

### Steps to reproduce.

1. clone `unity_test` folder to `Desktop`
2. then `cd Desktop/unity_test/test`
3. and `idf.py build`
4. last, `idf.py flash monitor`

### Debug Logs.

```plain
none
```

### More Information.

the `Application project` can run well, and outputs the following:
```
In main application. Collecting 32 random numbers from 1 to 100:
   5   66   86   98   62   66   40   42    7   62 
  33   77   40   54    8   75   39   82   73    4 
  22    5   72   78   53    1    6   48   23    1 
  84   55 
Mean: 45
```

ESP-Marius, 2023-02-06:
Hmm, this is strange. I tested a few times (on windows with PS as well), and as far as I could see it works fine.

Could you double check that you didnt modify anything? Does it work if you build it and run it directly from the example folder?

real-hao, 2023-02-06:
> Hmm, this is strange. I tested a few times (on windows with PS as well), and as far as I could see it works fine.
> 
> Could you double check that you didnt modify anything? Does it work if you build it and run it directly from the example folder?

I reinstalled the whole framework and resolved it. The reason may be that I accidentally modified the framework code when viewing the source code

closed 2023-02-06 by real-hao