espressif/esp-idf #10703: CONFIG_ESP32_WIFI_ENABLED=n for BLE only ?

f-hoepfinger-hr-agrartechnik, 2023-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

can we set CONFIG_ESP32_WIFI_ENABLED=n for a BLE only Application ? 

Thanks, 
Franz

xiewenxiang, 2023-02-06:
@f-hoepfinger-hr-agrartechnik 
Yes, you can set the `CONFIG_ESP32_WIFI_ENABLED=n` for a BLE only Application.

f-hoepfinger-hr-agrartechnik, 2023-02-11:
i was thinking so yes. 

but did not take it.

closed 2024-03-22 by Alvin1Zhang