espressif/esp-idf #10704: Support for BOOTLOADER_CUSTOM_RESERVE_RTC missing for ESP32C2

HomeACcessoryKid, 2023-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

support for `CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC` in ESP32-C2

### What is the actual behavior?

It is missing as an option from `idf.py menuconfig`

when compiling code that works fine on other platforms, this is the essential hint that it is missing:
```
/project/LCM4ESP32/main/ota.c:117:9: error: implicit declaration of function 'bootloader_common_get_rtc_retain_mem_reboot_counter'
```

### Steps to reproduce.

1. use the code of HomeACcessoryKid/LCM4ESP32 and set device to esp32-c2
2. compile the app code: `idf.py app`
3. observe compile error
4. go into menuconfig/bootloader options and missing RTC memory options

### Build or installation Logs.

```plain
root@25e7affaca0f:/project/LCM4ESP32# idf.py app
Executing action: app
Running ninja in directory /project/LCM4ESP32/build
Executing "ninja app"...
[2/8] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/ota.c.objFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/ota.c.obj 
ccache /opt/esp/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -I/project/LCM4ESP32/build/config -I/project/LCM4ESP32/main -I/opt/esp/idf/components/newlib/platform_include -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/include -I/opt/esp/idf/components/freertos/esp_additions/include/freertos -I/opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/opt/esp/idf/components/freertos/esp_additions/include -I/opt/esp/idf/components/esp_hw_support/include -I/opt/esp/idf/components/esp_hw_support/include/soc -I/opt/esp/idf/components/esp_hw_support/include/soc/esp32c2 -I/opt/esp/idf/components/esp_hw_support/port/esp32c2/. -I/opt/esp/idf/components/esp_hw_support/port/esp32c2/private_include -I/opt/esp/idf/components/heap/include -I/opt/esp/idf/components/log/include -I/opt/esp/idf/components/soc/include -I/opt/esp/idf/components/soc/esp32c2/. -I/opt/esp/idf/components/soc/esp32c2/include -I/opt/esp/idf/components/hal/esp32c2/include -I/opt/esp/idf/components/hal/include -I/opt/esp/idf/components/hal/platform_port/include -I/opt/esp/idf/components/esp_rom/include -I/opt/esp/idf/components/esp_rom/include/esp32c2 -I/opt/esp/idf/components/esp_rom/esp32c2 -I/opt/esp/idf/components/esp_common/include -I/opt/esp/idf/components/esp_system/include -I/opt/esp/idf/components/esp_system/port/soc -I/opt/esp/idf/components/esp_system/port/include/riscv -I/opt/esp/idf/components/esp_system/port/include/private -I/opt/esp/idf/components/riscv/include -I/opt/esp/idf/components/lwip/include -I/opt/esp/idf/components/lwip/include/apps -I/opt/esp/idf/components/lwip/include/apps/sntp -I/opt/esp/idf/components/lwip/lwip/src/include -I/opt/esp/idf/components/lwip/port/esp32/include -I/opt/esp/idf/components/lwip/port/esp32/include/arch -I/opt/esp/idf/components/esp_ringbuf/include -I/opt/esp/idf/components/efuse/include -I/opt/esp/idf/components/efuse/esp32c2/include -I/opt/esp/idf/components/driver/include -I/opt/esp/idf/components/driver/deprecated -I/opt/esp/idf/components/esp_pm/include -I/opt/esp/idf/components/mbedtls/port/include -I/opt/esp/idf/components/mbedtls/mbedtls/include -I/opt/esp/idf/components/mbedtls/mbedtls/library -I/opt/esp/idf/components/mbedtls/esp_crt_bundle/include -I/opt/esp/idf/components/esp_app_format/include -I/opt/esp/idf/components/bootloader_support/include -I/opt/esp/idf/components/bootloader_support/bootloader_flash/include -I/opt/esp/idf/components/esp_partition/include -I/opt/esp/idf/components/app_update/include -I/opt/esp/idf/components/spi_flash/include -I/opt/esp/idf/components/pthread/include -I/opt/esp/idf/components/esp_timer/include -I/opt/esp/idf/components/app_trace/include -I/opt/esp/idf/components/esp_event/include -I/opt/esp/idf/components/nvs_flash/include -I/opt/esp/idf/components/esp_phy/include -I/opt/esp/idf/components/esp_phy/esp32c2/include -I/opt/esp/idf/components/vfs/include -I/opt/esp/idf/components/esp_netif/include -I/opt/esp/idf/components/wpa_supplicant/include -I/opt/esp/idf/components/wpa_supplicant/port/include -I/opt/esp/idf/components/wpa_supplicant/esp_supplicant/include -I/opt/esp/idf/components/esp_wifi/include -I/opt/esp/idf/components/unity/include -I/opt/esp/idf/components/unity/unity/src -I/opt/esp/idf/components/cmock/CMock/src -I/opt/esp/idf/components/console -I/opt/esp/idf/components/http_parser -I/opt/esp/idf/components/esp-tls -I/opt/esp/idf/components/esp-tls/esp-tls-crypto -I/opt/esp/idf/components/esp_adc/include -I/opt/esp/idf/components/esp_adc/interface -I/opt/esp/idf/components/esp_adc/esp32c2/include -I/opt/esp/idf/components/esp_adc/deprecated/include -I/opt/esp/idf/components/esp_eth/include -I/opt/esp/idf/components/esp_hid/include -I/opt/esp/idf/components/tcp_transport/include -I/opt/esp/idf/components/esp_http_client/include -I/opt/esp/idf/components/esp_http_server/include -I/opt/esp/idf/components/esp_https_ota/include -I/opt/esp/idf/components/esp_https_server/include -I/opt/esp/idf/components/esp_lcd/include -I/opt/esp/idf/components/esp_lcd/interface -I/opt/esp/idf/components/protobuf-c/protobuf-c -I/opt/esp/idf/components/protocomm/include/common -I/opt/esp/idf/components/protocomm/include/security -I/opt/esp/idf/components/protocomm/include/transports -I/opt/esp/idf/components/esp_local_ctrl/include -I/opt/esp/idf/components/esp_psram/include -I/opt/esp/idf/components/espcoredump/include -I/opt/esp/idf/components/espcoredump/include/port/riscv -I/opt/esp/idf/components/wear_levelling/include -I/opt/esp/idf/components/sdmmc/include -I/opt/esp/idf/components/fatfs/diskio -I/opt/esp/idf/components/fatfs/vfs -I/opt/esp/idf/components/fatfs/src -I/opt/esp/idf/components/idf_test/include -I/opt/esp/idf/components/idf_test/include/esp32c2 -I/opt/esp/idf/components/ieee802154/include -I/opt/esp/idf/components/json/cJSON -I/opt/esp/idf/components/mqtt/esp-mqtt/include -I/opt/esp/idf/components/spiffs/include -I/opt/esp/idf/components/wifi_provisioning/include -march=rv32imc  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Og -fmacro-prefix-map=/project/LCM4ESP32=. -fmacro-prefix-map=/opt/esp/idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/ota.c.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/ota.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/ota.c.obj -c /project/LCM4ESP32/main/ota.c
/project/LCM4ESP32/main/ota.c: In function 'ota_rtc_read_task':
/project/LCM4ESP32/main/ota.c:117:9: error: implicit declaration of function 'bootloader_common_get_rtc_retain_mem_reboot_counter' [-Werror=implicit-function-declaration]
  117 |     if (bootloader_common_get_rtc_retain_mem_reboot_counter()) { //if zero, RTC CRC not valid
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/project/LCM4ESP32/main/ota.c:118:34: error: implicit declaration of function 'bootloader_common_get_rtc_retain_mem'; did you mean 'bootloader_common_get_chip_ver_pkg'? [-Werror=implicit-function-declaration]
  118 |         rtc_retain_mem_t* rtcmem=bootloader_common_get_rtc_retain_mem(); //access to the memory struct
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                  bootloader_common_get_chip_ver_pkg
/project/LCM4ESP32/main/ota.c:118:34: warning: initialization of 'rtc_retain_mem_t *' from 'int' makes pointer from integer without a cast [-Wint-conversio]
[... cut ...]
./components/esp_system/ld/esp32s3/memory.ld.in:#ifdef CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC
./components/esp_system/ld/esp32s3/memory.ld.in:#define ESP_BOOTLOADER_RESERVE_RTC (CONFIG_BOOTLOADER_RESERVE_RTC_SIZE + CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC_SIZE)
./components/esp_system/ld/esp32s2/memory.ld.in:#ifdef CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC
./components/esp_system/ld/esp32s2/memory.ld.in:#define ESP_BOOTLOADER_RESERVE_RTC (CONFIG_BOOTLOADER_RESERVE_RTC_SIZE + CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC_SIZE)
```

ESP-Marius, 2023-02-06:
Hi,

This is actually as intended. C2 doesnt have any RTC memory, so this kconfig option is disabled. (You can actually see its dependency in kconfig `depends on SOC_RTC_FAST_MEM_SUPPORTED`)

HomeACcessoryKid, 2023-02-06:
If C2 doesn't have any RTC, then the documentation of this feature must be made a lot clearer.
Based on that documentation I decided to raise this bug.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c2/api-reference/kconfig.html#config-bootloader-custom-reserve-rtc
Considering that you have dynamic documentation per platform, then you have to make the conclusion clear:  
_There is no RTC on a C2_. Just write it down.

Having said that, is there an alternative (short of abusing flash memory) to transfer information across a powered reset?

As always, thanks for the wonderful development platform that the ESP is!

HomeACcessoryKid, 2023-02-06:
Clear answer, thank you.

Having said that, and considering that the CUSTOM_RTC feature was able to address 16 bytes only anyhow, is it not possible to carve out 16 bytes out of this RTC_CNTL_STOREX_REG space? Or even 4?
Not that I know what that implies, but 16 seems a very small number, right?
Because you rightfully wrote "good" when referring to flash. Let alone that NVS is hard (if at all) to use sending info from or to the bootloader.

ESP-Marius, 2023-02-07:
You can see what these registers are used for [here](https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp32c2/rom/rtc.h#L59). It doesn't look like any of them are likely to available for users, but maybe there are some circumstances where we could free up one or two of them. I'm not that familiar with the details of the RTC related code, but i'll poke the guys responsible for this and ask them.

>  CUSTOM_RTC feature was able to address 16 bytes only anyhow
We actually recently fixed this (as you implied, this size limit was pretty arbitrary... ), so you are now free to use as much RTC memory as you want for this.

ESP-Marius, 2023-02-07:
Asked around and unfortunately it doesnt seem like we'll be able to release any of these registers for end-user use-cases.

HomeACcessoryKid, 2024-05-13:
Hi Marius,
Picking this up again with a new idea.
Upon reading the code for the registers, I think that there is a lot of free namespace for the RESET_REASON definitions.
So, with a 15 bit space (with the half-word and top bit stuff), the definitions could go as high as 32767, right? So plenty of space.

It would be nice if there would be an extension of the esp_restart(void) function where a parameter is given to indicate a reboot from another ota partition than the current selected one. We know that RTC_SW_CPU_RESET is encoded as 12.

Simple encoding could be done with a value of 28 (12+16) to indicate RTC_SW_CPU_RESET_OTA1 instead of RTC_SW_CPU_RESET.
Or, to cover all ota definitions and also factory reset you could assign 28 to factory and 32-47 to OTA0-OTA15

Other ideas can valid as well, but with plenty of space, these could be valuable functions.

Thanks for your attention...

ESP-Marius, 2024-07-24:
Hey, sorry for the slow reply. 

I think we are still reluctant to use these available bits, partially due to that as soon we open these up then we can never reclaim them in the case we need them to work around a bug or limitation in the future.

But how about this: would you mind opening a feature request describing what you want and why you want it? It might not result in us caving in a doing it, but at least it will give a proper place where we can discuss it, as well as giving other users the chance to upvote/explain their needs.

If we see that there is a big demand for adding a certain bit etc to one of the registers then we are more likely to be able to convince the stakeholders.

closed 2024-09-04 by SoucheSouche

HomeACcessoryKid, 2024-09-04:
Thanks @ESP-Marius , I still have it on my to-do list to open a feature request, but for now too busy with other stuff.