espressif/esp-idf #10705: ESP32-C6 IDF-5346: RC_FAST_CLK definitions don't match the TRM

ajf58, 2023-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In components/soc/esp32c6/include/soc/clk_tree_defs.h , `RC_FAST_CLK` is described as being ~17.5MHz. In the TRM, this is described as being ~20MHz (see section 5.2 of the TRM). I think this is an error in copying work from the information about the ESP32-C3, where RC_FAST_CLK is described as being 17.5MHz in both the source code, and the ESP32-C3's TRM.

I think one of the following is true:
1. The C6's  TRM is correct, and the source code is wrong.
2. The source code is correct, and the TRM is wrong.
3. Both are wrong.

ajf58, 2023-02-05:
Similarly I think the `SOC_CLK_RC_SLOW_FREQ_APPROX` has been copied from the ESP32-S2's clk_tree_defs.h as it report's 90kHz, not 150kHz as documented in the TRM.

songruo, 2023-02-21:
Hi @ajf58, 17.5MHz and 136kHz are closer to the actual frequencies of the clocks, comparing to 20MHz and 150kHz. We will update the TRM accordingly.

Alvin1Zhang, 2023-06-20:
The TRM has been updated, close the issue, feel free to reopen. Thanks.

closed 2023-06-20 by Alvin1Zhang