espressif/esp-idf #10706: Feature request: IP_EVENT_AP_STAIPASSIGNED should also deliver host-name

JB-DX, 2023-02-06:
In Soft-AP mode DHCP-event IP_EVENT_AP_STAIPASSIGNED delivers the ip-adr.
Could this be expanded to also deliver the host-name of the client?
I would like to reject clients with host-names that don't match a partial string.
Or is there another solution to find host-name and its ip-adr?

another related question:
How can an AP access its host-name / ip-adr table in order to search there?

hansw123, 2024-03-01:
Hi @JB-DX 
Sorry for the delay in replying to you, if I understand you correctly, the functionality you require can be achieved by modifying the lwip code, which you can do entirely on your own

hansw123, 2024-03-05:
hi @JB-DX 
you can use  esp_err_t esp_netif_get_hostname(esp_netif_t *esp_netif, const char **hostname)
get the host name
use  esp_err_t esp_netif_set_hostname(esp_netif_t *esp_netif, const char *hostname) set hostname
in code the default hostname is CONFIG_LWIP_LOCAL_HOSTNAME  "espressif"

AxelLin, 2024-03-05:
> hi @JB-DX you can use esp_err_t esp_netif_get_hostname(esp_netif_t *esp_netif, const char **hostname) get the host name use esp_err_t esp_netif_set_hostname(esp_netif_t *esp_netif, const char *hostname) set hostname in code the default hostname is CONFIG_LWIP_LOCAL_HOSTNAME "espressif"

The request is to get the **host-name of the client**, not the host-name of AP.

hansw123, 2024-03-05:
> The request is to get the **host-name of the client**, not the host-name of AP.

sorry for mistake,The current code does not have this feature yet, only the function to retrieve the MAC address is available. We will evaluate whether this feature should be implemented by us or by the client.

JB-DX, 2024-03-05:
Getting the name of the client  is more useful than its MAC address in order to accept only wanted clients. So implementing this feature would be really helpful.

hansw123, 2024-03-06:
hi @JB-DX 
We'll implement this feature，You will be notified as soon as it is realised

Alvin1Zhang, 2025-11-28:
Thanks for reporting and sorry for the very slow turnaround, commit on master branch is at https://github.com/espressif/esp-idf/commit/75e0d6273904336515c555677eeff34746b11854, feel free to reopen.

closed 2025-11-28 by Alvin1Zhang