espressif/esp-idf #10707: [esp32c3] build error when primary console is JTAG

chegewara, 2023-02-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

d29e53dc0c59d1b1b01c5e2c3ca1b22113828587

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build success.

### What is the actual behavior?

Build fail.

### Steps to reproduce.

Build with example `wifi/ftm`.

### Build or installation Logs.

```plain
/media/chegewara/upwork/esp/master/components/console/esp_console.h:91:18: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
   91 |     .baud_rate = CONFIG_ESP_CONSOLE_UART_BAUDRATE
```
```

deepaknairrpf, 2023-02-08:
I am facing the same issue on my Adafruit Feather esp32 s2 device. When I change the "channel for console output" to USB CDC, I get the above mentioned build error. With UART, I don't get the ftm> console.

Is there any known hardware on which I can successfully run this example to calculate distances between two devices?

As I'm a complete beginner trying to understand WiFi FTM, any help is highly appreciated.

Thanks

deepaknairrpf, 2023-02-16:
My bad.
I was able to change the code in main to use the `esp_console_dev_usb_cdc_config_t` data-type and the `esp_console_new_repl_usb_cdc` function.  I'm able to now run the FTM code on esp32-s2 target.

closed 2023-05-08 by espressif-bot

Massimiliano-solutiontech, 2023-09-27:
Hi,
I'm facing the same issue using `ESP-IDF v5.1.1`

```sh
esp-idf/components/esp_hw_support/sleep_modes.c:153:61: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
  153 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
```

baylf2000, 2023-09-28:
> Hi, I'm facing the same issue using `ESP-IDF v5.1.1` on a `ESP32-C6`
> 
> ```shell
> esp-idf/components/esp_hw_support/sleep_modes.c:153:61: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
>   153 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
> ```

Also facing this same issue with V5.1.1 and an ESP32-C6.

marciogranzotto, 2023-10-04:
same issue on `master` (272b4091f1f1ff169c84a4ee6b67ded4a005a8a7) and ESP32-C6

KonstantinKondrashov, 2023-10-06:
Hi @baylf2000 @Massimiliano-solutiontech @marciogranzotto!
Yes, I confirm that there is a build issue happening on V5.1 and master branches for ESP32-C6.

There is a fix already in this commit 8545f8595f2591e1d172e127e13d2eb0f6145149, but it has not been released to GitHub yet. This commit closes https://github.com/espressif/esp-idf/issues/11924 issue. I give the changes that fix this issue here. Backport for 5.1 will be done too.

```diff
--- a/components/esp_hw_support/sleep_modes.c
+++ b/components/esp_hw_support/sleep_modes.c
@@ -488,12 +488,12 @@ FORCE_INLINE_ATTR void resume_uarts(void)
 FORCE_INLINE_ATTR bool light_sleep_uart_prepare(uint32_t pd_flags, int64_t sleep_duration)
 {
     bool should_skip_sleep = false;
-#if !SOC_PM_SUPPORT_TOP_PD
+#if !SOC_PM_SUPPORT_TOP_PD || !CONFIG_ESP_CONSOLE_UART
     suspend_uarts();
 #else
     if (pd_flags & PMU_SLEEP_PD_TOP) {
         if ((s_config.wakeup_triggers & RTC_TIMER_TRIG_EN) &&
             // +1 is for cover the last charactor flush time
             (sleep_duration < (int64_t)((UART_LL_FIFO_DEF_LEN - uart_ll_get_txfifo_len(CONSOLE_UART_DEV) + 1) * UART_FLUSH_US_PER_CHAR) + SLEEP_UART_FLUSH_DONE_TO_SLEEP_US)) {
```

marciogranzotto, 2023-10-23:
@KonstantinKondrashov I think that it's already merged on `master`, but not on `release/v5.1`. Can you check that?

cwbyun, 2024-05-22:
Hi,
I'm facing the same issue using ESP-IDF v5.2.1 on a ESP32-C6