espressif/esp-idf #10709: PSRAM/mass production

vvasiu, 2023-02-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

Configuring PSRAM clock to 80MHz for maximizing performance should be suitable for mass production.

### What is the actual behavior?

Instead, the bootloader raises a warning at runtime, `W (279) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version`

### Steps to reproduce.

1. Set `CONFIG_SPIRAM_SPEED_80M` in sdkconfig
2. Build any application
3. Run the application and inspect the bootloader logs

### More Information.

I would like to have an official answer from Espressif with the future esp-idf release where this issue is not present anymore, meaning enabling the PSRAM clock at 80MHz is suitable for mass production.

vvasiu, 2023-02-06:
Thanks for the confirmation, @Icarus113 , I'm ok with your answer.

Alvin1Zhang, 2023-02-07:
Will update once v5.0.1 releases, thanks.

KonssnoK, 2023-02-21:
same issue is in v4.4,
@Icarus113 please backport
```

I (414) opi psram: vendor id : 0x0d (AP)
I (414) opi psram: dev id    : 0x02 (generation 3)
I (414) opi psram: density   : 0x03 (64 Mbit)
I (418) opi psram: good-die  : 0x01 (Pass)
I (422) opi psram: Latency   : 0x01 (Fixed)
I (427) opi psram: VCC       : 0x01 (3V)
I (432) opi psram: SRF       : 0x01 (Fast Refresh)
I (437) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (443) opi psram: BurstLen  : 0x01 (32 Byte)
I (448) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (454) opi psram: DriveStrength: 0x00 (1/1)
W (459) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (469) spiram: Found 64MBit SPI RAM device
I (474) spiram: SPI RAM mode: sram 80m
I (478) spiram: PSRAM initialized, cache is in normal (1-core) mode.
```

Icarus113, 2023-02-21:
It's already backported, you should see it in v4.4.5. @KonssnoK

KonssnoK, 2023-02-21:
@Icarus113 could you reference the commit please?
we are currently on top of dab3f38f0f966437c95e35f2c27e20d9a2a18fe7 and it's still there :)

Icarus113, 2023-02-22:
@KonssnoK The v4.4.5 isn't released yet. I'm sorry it didn't catch the v4.4.4 release.

This is the original [commit](https://github.com/espressif/esp-idf/commit/7b3dc053087750dbd3d2b924a4731c1d820a11d4)

I think you can ignore this warning on your branch, as the 80MHz DDR timing tuning parameters is verified.

KonssnoK, 2023-02-22:
@Icarus113 aha, well, good to know that it's just a placeholder. Thanks.

closed 2023-04-24 by Icarus113