espressif/esp-idf #10710: Clocks configuration through `menuconfig`

diablo1281, 2023-02-06:
### Is your feature request related to a problem?

More advanced clock configuration through `menuconfig` or another method to have more control other them in a build and configuration process.

### Describe the solution you'd like.

Would be great to have proper CLK options in IDF menuconfig. Setting peripherals clocks and others and configuring CPU properly from the start.

### Describe alternatives you've considered.

The workaround is to start on 80MHz and configure it by hand with `rtc_clk_cpu_freq_set_config` and correct config.
This method has a huge downside because FreeRTOS remain configured with 80MHz in mind.

### Additional context.

Using ESP-IDF I try to configure ESP32 to run at 40MHz from the start, which seems impossible to do easily. MenuConfig does not support lower than 80MHz and selecting CLK sources. Correcting by hand is impossible due to CMake.

igrr, 2023-02-06:
> Using ESP-IDF I try to configure ESP32 to run at 40MHz from the start, which seems impossible to do easily. MenuConfig does not support lower than 80MHz and selecting CLK sources.

The reason why selecting 40 MHz CPU frequency for the ESP32 is not supported via menuconfig is that:
- ESP32 radio peripherals (Wi-Fi and BT) can only function correctly when the APB clock is 80 MHz and the CPU clock is >= 80 MHz
- We haven't paid much effort to supporting the use case when ESP32 is used as a microcontroller, without radio peripherals, in ESP-IDF.

On future chips where the CPU clock can be lower than the APB clock (or on the chips which don't have the radio peripherals at all), lowering the default CPU clock will be supported through the same "CPU Frequency" menuconfig option.

If you explain the use case, or the goal you are trying to achieve, I might be able to suggest you an alternative solution you could use in the meantime.

diablo1281, 2023-02-06:
Thanks for the quick response.
ESP32-WROOM-32E was chosen for the project, but we don't utilize WiFi or BT for now. While testing module for EMC (electromagnetic compatibility) at 80MHz, we encourage some interferences which disappear when clocking it down to 40MHz.
Project originally was wrote in Arduino framework, but due to testing lower frequency, I rewriting it on ESP-IDF to gain more control.
It is my first to work with ESP so any tips would be appreciated.

igrr, 2023-02-06:
What you can do without modifying IDF is: utilize the power management feature, enabling CONFIG_PM_ENABLE in menuconfig.

Then in the application, set both the maximum and the minimum CPU frequency to 40 MHz:
https://gist.github.com/igrr/14ecd8a324173a7bf3091d36f95638f2

Some of the peripheral drivers may still calculate the divisors incorrectly, since this mode of operation is not part of our test plan. If you run into some problem, please don't hesitate to open an issue and we'll try to fix it.

diablo1281, 2023-02-06:
I just tried it, seems to be working at 40MHz and FreeRTOS times also work fine. Will see what EMC test shows.

Thank you very much for your help. :)

igrr, 2023-02-06:
> P.S. Just out o curiosity, is it possible to disable PLL_CLK if it is not used as a source clock?

I think there is no official API for that yet, and the Power Management module doesn't disable the PLL on ESP32 since it then it takes some time to get it enabled again. You can call the function `rtc_clk_cpu_freq_set_xtal(void)` declared in `soc/rtc.h`, it should do the trick.

igrr, 2023-02-08:
I'll close this issue since the question has been answered. If you run into another problem, please don't hesitate to open a new issue.

closed 2023-02-08 by igrr