espressif/esp-idf #10711: esp32 sntp同步网络时间耗时很久，有时候连接好wifi1分钟后都没同步好，请问这个怎么让它同步得快一点

CYCHEN97, 2023-02-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

esp32 sntp同步网络时间耗时很久，有时候连接好wifi1分钟后都没同步好，请问这个怎么让它同步得快一点

### What is the actual behavior?

esp32 sntp同步网络时间耗时很久，有时候连接好wifi1分钟后都没同步好，请问这个怎么让它同步得快一点

### Steps to reproduce.

esp32 sntp同步网络时间耗时很久，有时候连接好wifi1分钟后都没同步好，请问这个怎么让它同步得快一点

ESP-YJM, 2023-02-06:
@CYCHEN97 请问你用的 SNTP 服务器是哪个？

CYCHEN97, 2023-02-07:
@ESP-YJM[image]

CYCHEN97, 2023-02-07:
一直在这里面很久才能同步好[image]

ESP-YJM, 2023-02-07:
@CYCHEN97 可以抓包看下 NTP 的交互流程吗？然后把 SNTP_DEBUG 打开，看下日志。

CYCHEN97, 2023-02-07:
请问通过什么可以抓取NTP的交互流程？LWIP_DEBUG打开，SNTP_DEBUG就打开了是不？

CYCHEN97, 2023-02-07:
@ESP-YJM

ESP-YJM, 2023-02-07:
LWIP_DEBUG 是总开关，然后还要再打开 SNTP_DEBUG。抓包的话可以用抓包路由器或者抓包网卡抓空中包。

CYCHEN97, 2023-02-07:
@ESP-YJM 你好，这是debug的log文件，麻烦帮我分析一下，谢谢大佬
[SNTP_LOG.TXT](https://github.com/espressif/esp-idf/files/10675855/SNTP_LOG.TXT)

ESP-YJM, 2023-02-08:
从你的日志看来是解析 SNTP 域名服务器的域名解析超时，导致的 NTP 流程超时。你可以先尝试在自己电脑上 ping NTP 域名服务器，然后把域名解析出的 IP 填入 sntp_setservername 里，测试下 NTP 的流程。另外，你需要检查下你网络的 DNS 是否工作正常，或者通过调用 `dns_setserver` 设置一个常用的 DNS 域名解析服务器，比如`114.114.114.114`,`8.8.8.8`.

Alvin1Zhang, 2023-03-07:
@CYCHEN97 Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

CYCHEN97, 2023-03-08:
@Alvin1Zhang 测试完我认为这跟家里的路由器相关，连接手机的热点时同步时间都是很快的

Alvin1Zhang, 2023-03-09:
感谢反馈。