espressif/esp-idf #10718: msc_example triggers panic_abort when USB stick connected

skanky-dev, 2023-02-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using the esp-idf sample project "msc_example_main.c" (usb > host > msc)
No changes made, targeting ESP32S3, our own custom PCB with the ESP32S3 on it.
When I run it via the serial connetion, the ESP32 continually resets as soon as I plug in a USB memory stick (it gets to "I (361) example: Waiting for USB stick to be connected")
Its fine without the USB memory stick inserted.
Breakpointing with JTAG, the function wait_for_msc_device() and its call to xEventGroupWaitBits() I can get to, but somewhere beyond there execution halts in panic.c __attribute__((noreturn, no_sanitize_undefined)) panic_abort(const char *details)
I've been trying to step through, but can't get to a specific point it fails at, step into doesn't work properly for me.
Searching I can't find any known issues with this example project, I'm wondering if anyone can point me in any directions to try and debug the issue?

roma-jam, 2023-02-07:
Hi @skanky-dev, thanks for the delivering the problem. 

Firstly, full log can helps a lot, cause it contain the variables which can be used for the debug. 
Secondly, debugging USB-related part with the JTAG debugger not always can be achievable, because a USB has its' own timings. 
For enable esp-idf debug you can use "idf.py menuconfig" and enable the verbose debug level: "Component config -> Log output -> Default log verbosity" and set it to Verbose. 

Then rebuild the project and launch it again. 
Anyway, further log/logs could also help a lot to find the problem. 

Thanks.

skanky-dev, 2023-02-07:
Hey that's a great tool I wasn't aware of, thanks!

When I use that this is what I get:
`I (816) example: Waiting for USB stick to be connected
D (1126) HUB: Root port reset
D (1126) HUB: Stage done: START
D (1126) HUB: Stage done: GET_SHORT_DEV_DESC
E (1126) HUB: Bad transfer status: CHECK_SHORT_DEV_DESC
E (1126) HUB: Stage failed: CHECK_SHORT_DEV_DESC
D (1136) HUB: Stage done: CLEANUP_FAILED

***ERROR*** A stack overflow in task usb_events has been detected.

Backtrace: 0x40375d52:0x3fcea6d0 0x40379d25:0x3fcea6f0 0x4037d0f6:0x3fcea710 0x4037ba7b:0x3fcea790 0x40379ddc:0x3fcea7b0 0x40379dd2:0x3fcea6c0 0x00000002:0x00000010 |<-CORRUPTED
0x40375d52: panic_abort at C:/Espressif/esp/esp-idf/components/esp_system/panic.c:412

0x40379d25: esp_system_abort at C:/Espressif/esp/esp-idf/components/esp_system/esp_system.c:135

0x4037d0f6: vApplicationStackOverflowHook at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:388

0x4037ba7b: vTaskSwitchContext at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3648

0x40379ddc: _frxt_dispatch at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:418

0x40379dd2: _frxt_int_exit at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:213

ELF file SHA256: 3aea023dace96357

Rebooting...`

I'll try playing with the stack size, although seems odd this would be happenin with an untouched example project

roma-jam, 2023-02-07:
@skanky-dev Thanks for the log sharing. It helps a lot. 

At the first glance it looks like the problem with the same nature: [#10538](https://github.com/espressif/esp-idf/issues/10538)
But the result a little bit different, because the examples are not the same. 

As I have already posted in the 10538 issue, the solution for string device descriptor presence is already in a review state and should be in master as soon as it can be. 
But here the problem can be somewhere else, need to investigate more. 

Probably, @esp-saurabhbansal can help here. 

Thanks.

esp-saurabhbansal, 2023-02-08:
@skanky-dev 
Thankyou for the bug report.

The last log that you reported shows two problems:
1. stack overflow when you changed the verbosity level to 'Verbose' - This can be resolved by increasing the stack size of task usb_events from 2048 to 3072/4096. An increased stack is required for Verbosity Level of logs.
2. Second is of 'Stage failed: CHECK_SHORT_DEV_DESC' is being investigated upon. On my device, it is not getting reproduced.

Thanks.

skanky-dev, 2023-02-08:
Thanks for coming back on this. I'm not sure exactly what you mean by "stack size of task usb_events", but I've used "ESP-IDF: SDK Configuration editor" and changed "Minimal allowed size for shared stack" from 2048 to 4096 in the hope that is the right setting?
This is what I get now (resettng issue remains of course):

`--- idf_monitor on \\.\COM10 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x167c
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x2e98
entry 0x403c9904
I (25) boot: ESP-IDF v5.0 2nd stage bootloader
I (25) boot: compile time 16:47:33
I (25) boot: chip revision: v0.1
I (26) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (91) boot_comm: chip revision: 1, min. application chip revision: 0
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0e4f8h ( 58616) map
I (117) esp_image: segment 1: paddr=0001e520 vaddr=3fc92100 size=01af8h (  6904) load
I (118) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=30614h (198164) map
I (159) esp_image: segment 3: paddr=0005063c vaddr=3fc93bf8 size=012b8h (  4792) load
I (160) esp_image: segment 4: paddr=000518fc vaddr=40374000 size=0e020h ( 57376) load
I (177) esp_image: segment 5: paddr=0005f924 vaddr=50000000 size=00010h (    16) load
I (183) boot: Loaded app from partition at offset 0x10000
I (183) boot: Disabling RNG early entropy source...
I (197) cpu_start: Pro cpu up.
I (197) cpu_start: Starting app cpu, entry point is 0x40375250
0x40375250: call_start_cpu1 at C:/Espressif/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_stDr(19 )pefcpu up.U
V_BCA2H__ERR4 iEGlegal ed r biis tr clr g  ena ma k is[0 0
x f20V)CeCHsE_ERn :FUco_Be 2 access _rEGor intr c8rbi& enta minsk ith: 0 bf
[0m
D (220) clk: RTC_SLOW_CLK calibration value: 3736986
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERR: core 0 access error intr clr & ena mask is: 0x1f
I (229) cpu_start: Pro cpu start user code
I (232) cpu_start: cpu freq: 160000000 Hz
I (237) cpu_start: Application information:
I (242) cpu_start: Project name:     usb-msc
I (247) cpu_start: App version:      1
I (251) cpu_start: Compile time:     Feb  8 2023 16:47:06
I (257) cpu_start: ELF file SHA256:  04c19dd5868569f1...
I (263) cpu_start: ESP-IDF:          v5.0
V (268) memory_layout: reserved range is 0x3c04e4e8 - 0x3c04e508
D (274) memory_layout: Checking 5 reserved memory ranges:
D (280) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc92100
D (286) memory_layout: Reserved memory range 0x3fc92100 - 0x3fc95930
D (292) memory_layout: Reserved memory range 0x3fceee34 - 0x3fcf0000
D (299) memory_layout: Reserved memory range 0x40374000 - 0x40382100
0x40374000: _WindowOverflow4 at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1742

D (305) memory_layout: Reserved memory range 0x600fe000 - 0x600fe010
D (312) memory_layout: Building list of available memory regions:
V (318) memory_layout: Examining memory region 0x40374000 - 0x40378000
0x40374000: _WindowOverflow4 at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1742

0x40378000: spi_flash_mmap_pages at C:/Espressif/esp/esp-idf/components/spi_flash/flash_mmap.c:222
[... 102 lines cut ...]
D (1138) HUB: Stage done: CLEANUP_FAILED

***ERROR*** A stack overflow in task usb_events has been detected.

Backtrace: 0x40375d52:0x3fcea6d0 0x40379d25:0x3fcea6f0 0x4037d0f6:0x3fcea710 0x4037ba7b:0x3fcea790 0x40379ddc:0x3fcea7b0 0x40379dd2:0x3fcea6c0 0x00000002:0x00000010 |<-CORRUPTED
0x40375d52: panic_abort at C:/Espressif/esp/esp-idf/components/esp_system/panic.c:412

0x40379d25: esp_system_abort at C:/Espressif/esp/esp-idf/components/esp_system/esp_system.c:135

0x4037d0f6: vApplicationStackOverflowHook at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:388

0x4037ba7b: vTaskSwitchContext at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3648

0x40379ddc: _frxt_dispatch at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:418

0x40379dd2: _frxt_int_exit at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:213

ELF file SHA256: 04c19dd5868569f1

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403756f1
0x403756f1: esp_restart_noos_dig at C:/Espressif/esp/esp-idf/components/esp_system/esp_system.c:46 (discriminator 1)`

esp-saurabhbansal, 2023-02-08:
Hello @skanky-dev 

When I was saying 'increasing the stack size of task usb_events from 2048 to 3072/4096', I meant to make the change in this line of file msc_example_main.c:
```
    task_created = xTaskCreate(handle_usb_events, "usb_events", 2048, NULL, 2, NULL);
```

Please try making the change from 2048 -> 4096 in the above line.

skanky-dev, 2023-02-09:
Well thats very interesting, with that xTaskCreate() changed to 4096 the resetting issue is gone. Its not working still, but the original issue appears solved. I now get:

`--- idf_monitor on \\.\COM17 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x167c
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x2e98
entry 0x403c9904
I (25) boot: ESP-IDF v5.0 2nd stage bootloader
I (25) boot: compile time 12:38:36
I (25) boot: chip revision: v0.1
I (26) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (91) boot_comm: chip revision: 1, min. application chip revision: 0
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0e4f8h ( 58616) map
I (117) esp_image: segment 1: paddr=0001e520 vaddr=3fc92100 size=01af8h (  6904) load
I (118) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=30614h (198164) map
I (159) esp_image: segment 3: paddr=0005063c vaddr=3fc93bf8 size=012b8h (  4792) load
I (160) esp_image: segment 4: paddr=000518fc vaddr=40374000 size=0e020h ( 57376) load
I (177) esp_image: segment 5: paddr=0005f924 vaddr=50000000 size=00010h (    16) load
I (183) boot: Loaded app from partition at offset 0x10000
I (183) boot: Disabling RNG early entropy source...
I (197) cpu_start: Pro cpu up.
I (197) cpu_start: Starting app cpu, entry point is 0x40375250
0x40375250: call_start_cpu1 at C:/Espressif/esp/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_stDr(19 )pefcpu up.U
V_BCA2H__ERR4 iEGlegal ed r biis tr clr g  ena ma k is[0 0
x f20V)CeCHsE_ERn :FUco_Be 2 access _rEGor intr c8rbi& enta minsk ith: 0 bf
[0m
D (220) clk: RTC_SLOW_CLK calibration value: 3704896
V CACHE_ERR: illegal error intr clr & ena mask is: 0x3f
V CACHE_ERR: core 0 access error intr clr & ena mask is: 0x1f
I (229) cpu_start: Pro cpu start user code
I (232) cpu_start: cpu freq: 160000000 Hz
I (237) cpu_start: Application information:
I (242) cpu_start: Project name:     usb-msc
I (247) cpu_start: App version:      1
I (251) cpu_start: Compile time:     Feb  9 2023 12:38:11
I (257) cpu_start: ELF file SHA256:  baa102a18b950b52...
I (263) cpu_start: ESP-IDF:          v5.0
V (268) memory_layout: reserved range is 0x3c04e4e8 - 0x3c04e508
D (274) memory_layout: Checking 5 reserved memory ranges:
D (280) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc92100
D (286) memory_layout: Reserved memory range 0x3fc92100 - 0x3fc95930
D (292) memory_layout: Reserved memory range 0x3fceee34 - 0x3fcf0000
D (299) memory_layout: Reserved memory range 0x40374000 - 0x40382100
0x40374000: _WindowOverflow4 at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1742

D (305) memory_layout: Reserved memory range 0x600fe000 - 0x600fe010
D (312) memory_layout: Building list of available memory regions:
V (318) memory_layout: Examining memory region 0x40374000 - 0x40378000
0x40374000: _WindowOverflow4 at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1742

0x40378000: spi_flash_mmap_pages at C:/Espressif/esp/esp-idf/components/spi_flash/flash_mmap.c:222

[... 77 lines cut ...]
V (688) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (688) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x402
D (688) intr_alloc: Connected src 57 to int 12 (cpu 0)
V (688) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (698) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (698) intr_alloc: Connected src 80 to int 2 (cpu 1)
I (708) cpu_start: Starting scheduler on APP CPU.
V (708) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (718) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x402
D (728) intr_alloc: Connected src 58 to int 3 (cpu 1)
D (728) heap_init: New heap initialised at 0x3fce9710
V (738) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (748) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (748) intr_alloc: Connected src 52 to int 13 (cpu 0)
I (758) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
V (768) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (768) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x802
D (778) intr_alloc: Connected src 38 to int 17 (cpu 0)
I (818) example: Waiting for USB stick to be connected
D (1128) HUB: Root port reset
D (1128) HUB: Stage done: START
D (1128) HUB: Stage done: GET_SHORT_DEV_DESC
E (1128) HUB: Bad transfer status: CHECK_SHORT_DEV_DESC
E (1128) HUB: Stage failed: CHECK_SHORT_DEV_DESC
D (1138) HUB: Stage done: CLEANUP_FAILED`

esp-saurabhbansal, 2023-02-10:
@skanky-dev 

I observed that you are working on IDF v5.0
The issue is under investigation.
Please give me some time. I will get back to you.

skanky-dev, 2023-02-16:
Thanks @esp-saurabhbansal, I'm working towards completing my project at the end of next week. Can I ask what sort of timeline you think there might be on being able to look at this as I'm not sure if I should wait or if I should move to an older IDF to get the product out the door?

Dazza0, 2023-02-28:
@skanky-dev Sorry for the delay. Given that the device is failing at the `GET_SHORT_DEV_DESC`, we suspect this issue might be related to reset recovery time.

In `hcd.c:33` ([link](https://github.com/espressif/esp-idf/blob/release/v5.0/components/usb/hcd.c#L33)), could you try increasing the value of the `#define RESET_RECOVERY_MS` macro to something like `500` and see if that resolves your issue?

skanky-dev, 2023-02-28:
Hi, that has solved something, I now get:
D (627) cpu_start: calling init function: 0x42001dc4
0x42001dc4: esp_init_app_elf_sha256 at C:/Espressif/esp/esp-idf/components/esp_app_format/esp_app_desc.c:69

D (632) cpu_start: calling init function: 0x42006eac on core: 0
0x42006eac: __esp_system_init_fn_esp_timer_startup_init at C:/Espressif/esp/esp-idf/components/esp_timer/src/esp_timer.c:464

V (638) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (644) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xC02
D (652) intr_alloc: Connected src 59 to int 3 (cpu 0)
D (657) cpu_start: calling init function: 0x42002810 on core: 0
0x42002810: __esp_system_init_fn_init_components0 at C:/Espressif/esp/esp-idf/components/esp_system/startup.c:477

V (664) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (670) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (678) intr_alloc: Connected src 79 to int 9 (cpu 0)
I (683) cpu_start: Starting scheduler on PRO CPU.
V (688) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (688) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x402
D (688) intr_alloc: Connected src 57 to int 12 (cpu 0)
V (688) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (698) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (698) intr_alloc: Connected src 80 to int 2 (cpu 1)
I (708) cpu_start: Starting scheduler on APP CPU.
V (708) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (718) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x402
D (728) intr_alloc: Connected src 58 to int 3 (cpu 1)
D (728) heap_init: New heap initialised at 0x3fce9710
V (738) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (748) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (748) intr_alloc: Connected src 52 to int 13 (cpu 0)
I (758) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
V (768) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (768) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x802
D (778) intr_alloc: Connected src 38 to int 17 (cpu 0)
I (818) example: Waiting for USB stick to be connected
D (1598) HUB: Root port reset
D (1598) HUB: Stage done: START
D (1598) HUB: Stage done: GET_SHORT_DEV_DESC
E (1598) HUB: Bad transfer status: CHECK_SHORT_DEV_DESC
E (1598) HUB: Stage failed: CHECK_SHORT_DEV_DESC
D (1608) HUB: Stage done: CLEANUP_FAILED

Dazza0, 2023-02-28:
@skanky-dev Look like enumeration is still failing at the `CHECK_SHORT_DEV_DESC` stage. Any chance you could attach a logic analyzer to see packets are failing?

From our past experience, failing at `CHECK_SHORT_DEV_DESC` is usually due to the device not acknowledge the Host's control transfers.

skanky-dev, 2023-02-28:
Unfortunately my logic analyser isn't fast enough for USB

Dazza0, 2023-02-28:
One thing that could help is to print the actual transfer status on `E (1598) HUB: Bad transfer status:` (note: it should already do so on the latest master, see commit https://github.com/espressif/esp-idf/commit/6ae151db8c6f06158aba510b4f9c62ed6e4bc01a, line https://github.com/espressif/esp-idf/commit/6ae151db8c6f06158aba510b4f9c62ed6e4bc01a#diff-165a79f61ddffd65fd0111c9b542af693bfc228eb8230e62d21ff4999a8983acR423).

If the erroneous transfer's status is `1` (i.e., `USB_TRANSFER_STATUS_ERROR`), then it is usually indicative of the device not responding to a particular control transfer (i.e., not acknowledging a USB packet). The USB protocol requires the Host retry a transaction 3 times before reporting an error.

skanky-dev, 2023-02-28:
Yes its giving a 1:
I (819) example: Waiting for USB stick to be connected
D (1599) HUB: Root port reset
D (1599) HUB: Stage done: START
D (1599) HUB: Stage done: GET_SHORT_DEV_DESC
E (1599) HUB: Bad transfer status 1: CHECK_SHORT_DEV_DESC
E (1599) HUB: Stage failed: CHECK_SHORT_DEV_DESC
D (1609) HUB: Stage done: CLEANUP_FAILED

Dazza0, 2023-02-28:
@skanky-dev Could try if any of the following changes allows the enumeration to succeed (or get past the `CHECK_SHORT_DEV_DESC` stage).

- Increasing the following time macros in `hcd_dwc.c`
```c
#define DEBOUNCE_DELAY_MS                       500
#define RESET_HOLD_MS                           500
#define RESET_RECOVERY_MS                       500
```

- Try keep the USB stick/flash drive already plugged into the USB port (i.e., already powered), then reset the ESP32-S3 (either via the `EN` button on the development board, or via `idf.py monitor`).
- Does another USB stick/flash drive from a different vendor work?

skanky-dev, 2023-02-28:
Thanks for those suggestions. They haven't fixed the issue, there's just a delay before it gets to the same point.
If I change the "Bad transfer status" return value to true I get:
I (819) example: Waiting for USB stick to be connected
D (2319) HUB: Root port reset
D (2319) HUB: Stage done: START
D (2319) HUB: Stage done: GET_SHORT_DEV_DESC
E (2319) HUB: Bad transfer status 1: CHECK_SHORT_DEV_DESC
D (2319) HUB: Stage done: CHECK_SHORT_DEV_DESC
D (3329) HUB: Stage done: SECOND_RESET
E (3329) HUB: Failed to submit: SET_ADDR
E (3329) HUB: Stage failed: SET_ADDR
D (3329) HUB: Stage done: CLEANUP_FAILED

I need to order an alternative USB memory stick to test with something different as I don't have another USB-C one here. I should get that in tomorrow (UK time here) and I'll test with that plus on a ESP32-S3-DevKitC-1 board that I should get in too, just to ensure its not my custom PCB at fault

wakwak-koba, 2023-02-28:
I was searching on 'CHECK_SHORT_LANGID_TABLE' and landed here.
I was having trouble with that error with the hub connected, and have yet to resolve it.

The memory stick you are using may be the Hub
Isn't it built in?
When you connect it to your PC, why don't you check the device manager to see if there are more hubs?

skanky-dev, 2023-03-03:
I've tested now with a ESP32-S3-DevKitC-1 and found that it worked so it's turned out there was an issue with my custom PCB. One of my D connections must have been bad, resulting in enough of a data transfer to occur for the ESP32 to detect a USB stick being connected, but not so that the data transfer would be good.  So my apologies for that, I had checked and double checked, but the bad solder joint was in an inaccessible area under the USB connector. Replacing it and reflowing resolved the issue.
Working backwards through the issue then, this change in hcd.c is now no longer required for the USB stick to work.  
RESET_RECOVERY_MS                       30 > 500
If I change it back to 30 it still works. However I'd argue a change is still needed, because in my faulty setup this value was resulting in a ESP32 reset, whereas changing it to 500 resulted in the driver correctly determining there was a problem and giving up, so maybe it should be changed anyway? One for you with your expertise in the driver to decide on. I'm going to leave it changed in my setup.

This change in the sample project msc_example_main.c was still needed though:
  task_created = xTaskCreate(handle_usb_events, "usb_events", 2048, NULL, 2, NULL);
Changing this to 4096 resolved my original issue.

Thanks very much for all your help

skanky-dev, 2023-03-03:
> 

Thanks wakwak-koba, didn't turn out to be my issue but a good idea for anyone experiencing a similar issue

esp-saurabhbansal, 2023-03-07:
@skanky-dev 
Good to know that your issue is resolved.
Can this issue be closed if there is nothing more to discuss.

closed 2023-03-08 by skanky-dev

esp-saurabhbansal, 2023-04-14:
Hello,

Just for your information, We have recently released USB Device MSC driver. Please check if it is useful to you. We highly appreciate the feedback.

The driver makes ESP based device recognizable by USB-hosts as a USB Mass Storage Device.
It either allows the embedded application to access the partition or Host PC accesses the partition over USB MSC. They can't be allowed to access the partition at the same time.

The driver supports storage media of two types:
1. SPI Flash
2. SD MMC Card

Driver can be found [here](https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/tusb_msc_storage.c).
Sample application can be found [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc).

Thanks.

skanky-dev, 2023-04-19:
Thanks for letting me know. At  the moment I'm only using USB to access USB
mass storage devices, so in host mode, but good to know for the future
Thanks
Adam

Adam Pulley
IBEX - Technology Development Specialists
Tel: +44 (0)20 8087 2178
Email: ***@***.***
Web: https://ibex.tech
This email and any files transmitted with it are private and confidential.

Asanga-SPLat, 2025-05-09:
Hi Guys,

I am experiencing the same issue failing at the `GET_SHORT_DEV_DESC` for my one of USB Flash drives. It is imported from china with custom labelled. I am using idf5.4.1, ESP32-S3-DevKitC-V1.0 board with OTG cable and usb/host/msc example for testing. When I am plugin the custom made USB stick, some times it works fully and some times ESP stucks at `ENUM: Bad transfer status 1: CHECK_SHORT_DEV_DESC`. I checked with commercial available other USB sticks, and they didn't fail at any time. 

I changed debug output into Debug and here is the output when plug the USB stick and when it fails.
> I (612) main_task: Calling app_main()
D (612) intr_alloc: Connected src 38 to int 5 (cpu 0)
D (612) EXT_PORT: Driver installed
D (622) EXT_HUB: Driver installed
I (622) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
D (632) intr_alloc: Connected src 16 to int 8 (cpu 0)
I (632) example: Waiting for USB flash drive to be connected
D (652) USB_MSC: USB MSC handling start
D (4882) HUB: Root port reset
D (4882) HUB: Device tree node (uid=1): new
D (4882) ENUM: [0:0] Start processing, device address 0
E (4882) USBH: Dev 0 EP 0 Error
D (4882) ENUM: [0:0] GET_SHORT_DEV_DESC OK
D (4882) USBH: Processing actions 0xe
D (4892) USBH: Default pipe device 0
E (4892) ENUM: Bad transfer status 1: CHECK_SHORT_DEV_DESC
E (4892) ENUM: [0:0] CHECK_SHORT_DEV_DESC FAILED
D (4902) ENUM: [0:0] CANCEL OK
D (4902) HUB: Disabling root port

Here is the debug output when success on plugging the USB stick (Custom labelled)

> D (612) EXT_PORT: Driver installed
D (622) EXT_HUB: Driver installed
I (622) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
D (632) intr_alloc: Connected src 16 to int 8 (cpu 0)
I (632) example: Waiting for USB flash drive to be connected
D (652) USB_MSC: USB MSC handling start
D (5702) HUB: Root port reset
D (5702) HUB: Device tree node (uid=1): new
D (5702) ENUM: [0:0] Start processing, device address 0
D (5702) ENUM: [0:0] GET_SHORT_DEV_DESC OK
D (5702) USBH: Processing actions 0x4
D (5702) USBH: Default pipe device 0
D (5712) ENUM: [0:0] CHECK_SHORT_DEV_DESC OK
D (5772) ENUM: [0:0] SECOND_RESET OK
D (5772) ENUM: [0:0] SECOND_RESET_COMPLETE OK
D (5772) ENUM: [0:0] SET_ADDR OK
D (5772) USBH: Processing actions 0x4
D (5772) USBH: Default pipe device 0
D (5772) ENUM: Assign address (dev_addr=1)
D (5782) ENUM: [0:0] CHECK_ADDR OK
D (5792) ENUM: [0:0] SET_ADDR_RECOVERY OK
D (5792) ENUM: [0:0] GET_FULL_DEV_DESC OK
D (5792) USBH: Processing actions 0x4
D (5792) USBH: Default pipe device 1
D (5792) ENUM: [0:0] CHECK_FULL_DEV_DESC OK
D (5802) ENUM: Selected bConfigurationValue=1
D (5802) ENUM: [0:0] SELECT_CONFIG OK
D (5802) ENUM: [0:0] GET_SHORT_CONFIG_DESC OK
D (5812) USBH: Processing actions 0x4
D (5812) USBH: Default pipe device 1
D (5812) ENUM: [0:0] CHECK_SHORT_CONFIG_DESC OK
D (5822) ENUM: [0:0] GET_FULL_CONFIG_DESC OK
D (5822) USBH: Processing actions 0x4
D (5832) USBH: Default pipe device 1
D (5832) ENUM: [0:0] CHECK_FULL_CONFIG_DESC OK
D (5832) ENUM: [0:0] GET_SHORT_LANGID_TABLE OK
D (5842) USBH: Processing actions 0x4
D (5842) USBH: Default pipe device 1
D (5842) ENUM: [0:0] CHECK_SHORT_LANGID_TABLE OK
D (5852) ENUM: [0:0] GET_FULL_LANGID_TABLE OK
D (5852) USBH: Processing actions 0x4
D (5862) USBH: Default pipe device 1
D (5862) ENUM: [0:0] CHECK_FULL_LANGID_TABLE OK
[... 98 lines cut ...]
I (6402) example: Writing to file /usb/esp/dummy
I (7422) example: Write speed 1.03 MiB/s
D (7422) vfs_fat: vfs_fat_lseek: offset=0, filesize:=1048576
I (7422) example: Reading from file /usb/esp/dummy
I (8392) example: Read speed 1.08 MiB/s
I (8402) example: Example finished, you can disconnect the USB flash drive

Like @Dazza0 said, I increased the `CONFIG_USB_HOST_RESET_RECOVERY_MS` to 500 and problem solved. But, some times, when start the ESP from reset state (like power up), ESP stuck at getting `GET_SHORT_DEV_DESC` and does nothing. Here is the debug output of a such event.

> I (612) main_task: Calling app_main()
D (612) intr_alloc: Connected src 38 to int 5 (cpu 0)
D (612) EXT_PORT: Driver installed
D (622) EXT_HUB: Driver installed
I (622) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
D (632) intr_alloc: Connected src 16 to int 8 (cpu 0)
I (632) example: Waiting for USB flash drive to be connected
D (652) USB_MSC: USB MSC handling start
D (962) HUB: Root port reset
D (962) HUB: Device tree node (uid=1): new
D (962) ENUM: [0:0] Start processing, device address 0
D (962) ENUM: [0:0] GET_SHORT_DEV_DESC OK

I have an application to start logging data into USB stick. According to my application, there is nobody to unplug the USB stick and plug it back after turning on the machine. So, mounting the USB stick file system on power up it important. Like I said, this is only happens with my custom logo USB stick, which is working fine in my Win PC without any driver issue.

Thanks.