espressif/esp-idf #10719: HTTPS OTA fails with SIM7600

AAl-Hadi, 2023-02-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

For OTA to be working fine

### What is the actual behavior?

I get "esp-modem: HW FIFO Overflow" error

### Steps to reproduce.

1. Use SIM7600G with esp-modem and HTTPS_OTA 

### Debug Logs.

```plain
W (340813) esp-modem: HW FIFO Overflow
I (340823) ota_service: heap = 11500
W (340833) esp-modem: HW FIFO Overflow
I (340833) ota_service: heap = 11500
W (340863) esp-modem: HW FIFO Overflow
W (340883) esp-modem: HW FIFO Overflow
I (340883) ota_service: heap = 11500
I (340883) ota_service: heap = 11500
W (340923) esp-modem: HW FIFO Overflow
W (340943) esp-modem: HW FIFO Overflow
```

### More Information.

So, I have an Ethernet project that is running MQTT. OTA is working on HTTPS (same as the example).
I changed the project to work with SIM7600G instead of Ethernet. All parts work fine except OTA (was working with Ethernet).
When I do OTA with cell I get these errors:
```
16:50:05.470 -> [0;33mW (929738) esp-modem: HW FIFO Overflow[0m
16:50:07.414 -> [0;33mW (931718) esp-modem: HW FIFO Overflow[0m
16:50:07.503 -> [0;33mW (931798) esp-modem: HW FIFO Overflow[0m
16:50:07.589 -> [0;33mW (931868) esp-modem: HW FIFO Overflow[0m
16:50:07.689 -> [0;33mW (931948) esp-modem: HW FIFO Overflow[0m
16:50:07.740 -> [0;33mW (932038) esp-modem: HW FIFO Overflow[0m
```
Now, if I leave it for like 10-15 minutes it keeps printing these errors and then eventually works. But I want it to work flawlessly if possible
I looked into these comments [here](https://github.com/espressif/esp-idf/issues/4582) and [here](https://github.com/espressif/esp-idf/issues/7205)

I tried to move UART to ISR
The changes are:
1. I get this error instead of the previous one:
```
16:54:14.490 -> [0;31mE (81789) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1[0m
16:54:14.490 -> [0;31mE (81789) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1[0m
16:54:14.490 -> [0;31mE (81809) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1[0m
16:54:14.490 -> [0;31mE (81809) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1[0m
16:54:14.490 -> [0;31mE (81809) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1[0m
16:54:14.572 -> [0;31mE (81889) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1[0m
16:54:16.542 -> [0;31mE (83849) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1[0m
```
2. Instead of taking 10-15 minutes to work, now it takes 5-10 minutes. Still too long and still something is wrong

I then tried to increase the timeout using:
`uart_set_rx_timeout(dte_config.port_num, 255);` 
but I get this error:
`E (21263) uart: tout_thresh = 255 > maximum value = 1`

Finally, I tried to migrate to ESP-IDF release/v5.0 but I still get he same errors
What is the problem and how to fix it?

david-cermak, 2023-02-08:
Hi @AAl-Hadi 

After moving the UART's ISR to `IRAM`, the UART reception is no longer the bottleneck, so trying to adjust Rx timetout wouldn't help.
The errors you're seeing are from lwIP, which is struggling to accept (a lot of?) packets to the stack internals. To address this, you can either increase number of Rx buffers, or reduce the re-transmit timeout of TCP config.
Could you please unselect:
```
[ ] TCP high speed retransmissions
```
located under `(Top) → Component config → LWIP → TCP` component configuration menu.

and see if it helps?

AAl-Hadi, 2023-02-09:
Thank you for your reply @david-cermak 
I tried to uncheck "TCP high speed retransmissions" I still get the same problem
How do I increase the number of Rx buffers?
Btw, this is the config I use for the modem:
Default:
```
#define ESP_MODEM_DTE_DEFAULT_CONFIG()          \
    {                                           \
        .port_num = UART_NUM_1,                 \
        .data_bits = UART_DATA_8_BITS,          \
        .stop_bits = UART_STOP_BITS_1,          \
        .parity = UART_PARITY_DISABLE,          \
        .baud_rate = 115200,                    \
        .flow_control = MODEM_FLOW_CONTROL_NONE,\
        .tx_io_num = 13,                        \
        .rx_io_num = 15,                        \
        .rts_io_num = 27,                       \
        .cts_io_num = 23,                       \
        .rx_buffer_size = 1024,                 \
        .tx_buffer_size = 512,                  \
        .pattern_queue_size = 20,               \
        .event_queue_size = 30,                 \
        .event_task_stack_size = 2048,          \
        .event_task_priority = 5,               \
        .line_buffer_size = 512                 \
    }
```
And I changed the following from default:
```
  esp_modem_dte_config_t dte_config = ESP_MODEM_DTE_DEFAULT_CONFIG();
  dte_config.tx_io_num = CONFIG_EXAMPLE_MODEM_UART_TX_PIN;
  dte_config.rx_io_num = CONFIG_EXAMPLE_MODEM_UART_RX_PIN;
  dte_config.event_task_stack_size = 3024;
  dte_config.rx_buffer_size = 2048;
```
This is the [HW ](https://pt.aliexpress.com/item/1005001705250713.html?gatewayAdapt=glo2bra) I'm currently using. And [GitHub ](https://github.com/Xinyuan-LilyGO/T-SIM7600X) page

And here's the sdkconfig, assuming I messed something
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10694612/sdkconfig.txt)

david-cermak, 2023-02-09:
> I tried to uncheck "TCP high speed retransmissions" I still get the same problem

But could still see from your sdkconfig:
```
CONFIG_LWIP_TCP_RTO_TIME=1500
```
(which is still the IDF default)

Could you please adjust this option to at lease 3 seconds? (you can find it again in the [TCP menu](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-tcp-rto-time))

> How do I increase the number of Rx buffers?

Here you can adjust the TCP's recv mbox: [CONFIG_LWIP_TCP_RECVMBOX_SIZE](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-tcp-recvmbox-size), but this probably won't help in your case.

AAl-Hadi, 2023-02-12:
Thank you for your reply @david-cermak 
I changed RTO to 3sec and SYN to 6
The same errors
I get 2 more lines printed with them:
```
E (98314) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (98314) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (98414) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (98414) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1
E (108624) TRANSPORT_BASE: esp_tls_conn_read error, errno=No more processes
W (108624) HTTP_CLIENT: esp_transport_read returned:-26880 and errno:11
```

david-cermak, 2023-02-12:
@AAl-Hadi Could you please check how much free heap you have before starting the OTA?
Those failures in the logs mean that lwIP cannot obtain new PBUF (packet buffer) and the new errors indicate that the connection was interrupted.

Could you please again try to adjust some of the lwip options per below?
```
CONFIG_LWIP_TCP_MAXRTX=12
CONFIG_LWIP_TCP_SYNMAXRTX=12
CONFIG_LWIP_TCP_RTO_TIME=3000
CONFIG_TCP_MSS=640
```
(now wanted to keep RTO higher, while max SYN retransmit number default, and decrease MSS to save on average pbuf size)

AAl-Hadi, 2023-02-13:
Thank you for your reply @david-cermak 
The heap before starting OTA is ~56000 bytes
```
I (96234) ota_service: heap = 56472
I (96244) ota_service: heap = 56256
```
After starting it it varies between 3800-9000 bytes
```
I (100284) ota_service: heap = 9108
I (100354) ota_service: heap = 8336
I (100364) ota_service: heap = 8336
I (100444) ota_service: heap = 6888
I (100454) ota_service: heap = 6888
I (100544) ota_service: heap = 6164
I (100544) ota_service: heap = 6164
I (100624) ota_service: heap = 3992
I (100634) ota_service: heap = 3992
I (101944) ota_service: heap = 8336
I (101944) ota_service: heap = 7612
I (102034) ota_service: heap = 6164
I (102034) ota_service: heap = 6164
I (102114) ota_service: heap = 5440
I (102124) ota_service: heap = 5440
I (102214) ota_service: heap = 3992
I (102214) ota_service: heap = 3992
I (102304) ota_service: heap = 5440
I (103664) ota_service: heap = 8336
I (103754) ota_service: heap = 7612
I (103754) ota_service: heap = 6888
I (103834) ota_service: heap = 6164
I (103844) ota_service: heap = 6164
```
I tried the options you mentioned. Everything is still the same

david-cermak, 2023-02-13:
Thanks for looking into heap sizes, 56kB sounds quite enough, but I think this might really become marginal and causing issues during OTA processing.
The lowest number, about 3.8k of *total* free size, we have still to allocate some 1.5k packets, which need to be contiguous. (Note that it's also possible to check the [`heap_caps_get_largest_free_block()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html#_CPPv232heap_caps_get_largest_free_block8uint32_t), which gives very useful information on these marginal scenarios).

Could you please try to apply the following patch to see if it helps?
```patch
diff --git a/components/esp_netif/lwip/esp_netif_lwip_ppp.c b/components/esp_netif/lwip/esp_netif_lwip_ppp.c
index fc27232eff..ca02911d40 100644
--- a/components/esp_netif/lwip/esp_netif_lwip_ppp.c
+++ b/components/esp_netif/lwip/esp_netif_lwip_ppp.c
@@ -269,10 +269,25 @@ esp_err_t esp_netif_start_ppp(esp_netif_t *esp_netif)
     return ESP_OK;
 }
 
+err_t pppos_input_tcpip2(ppp_pcb *ppp, u8_t *s, int l)
+{
+    struct pbuf *p = pbuf_alloc(PBUF_RAW, l, PBUF_RAM);
+    if (!p) {
+        return ERR_MEM;
+    }
+    memcpy(p->payload, s, l);
+
+    err_t err = tcpip_inpkt(p, ppp_netif(ppp), pppos_input_sys);
+    if (err != ERR_OK) {
+        pbuf_free(p);
+    }
+    return err;
+}
+
 void esp_netif_lwip_ppp_input(void *ppp_ctx, void *buffer, size_t len, void *eb)
 {
     struct lwip_peer2peer_ctx * obj = ppp_ctx;
-    err_t ret = pppos_input_tcpip(obj->ppp, buffer, len);
+    err_t ret = pppos_input_tcpip2(obj->ppp, buffer, len);
     if (ret != ERR_OK) {
         ESP_LOGE(TAG, "pppos_input_tcpip failed with %d", ret);
     }
```
(Here we try to allocate exactly the size, we'd like to use -- since in OTA we're mostly receiving, so the packets are more or less short TCP ACKs)

Another idea might be to try to free up some of the heap used, as we're going to restart anyway after the OTA, so you might be able to find a kilobyte or two, perhaps?

AAl-Hadi, 2023-02-14:
Thank you for your reply @david-cermak 
I'll try that solution and will get back to you
I'll also restart a new project and add only MQTT and OTA components. Let's see if that may solve the problem. Maybe another component is behind this>?

david-cermak, 2023-02-14:
> Maybe another component is behind this>?

Should be probably the application designer who takes care of keeping the system healthy, including free heap.
Unfortunately we dont demonsttrate this approach (init - deinit) consitently in IDF examples (and some components might not even support deinit, like lwip).

As for the MQTT+OTA project, it's probably just enough to call `esp_mqtt_client_destroy()` before starting the OTA; no need to keep MQTT up and running and helps to free up ~3 KB, which would make a difference in such corner cases of ~60KB free, largely fragmented heap.

AAl-Hadi, 2023-02-26:
Hello @david-cermak 
Sorry I was a bit late
I copied the MQTT example "ssl_mutual_auth" and added OTA to it
I used it initially without the fixes mentioned bove
I got this error:
`HW FIFO Overflow`
Then I placed the UART ISR in the ram and now everything is working with a small note
With Ethernet, OTA takes less than a minute, with cell it takes ~2.5 minutes. 
And it's not that the whole process is slow. It initiate OTA and wait for ~2.5 minutes then start writing the sectors which will take few seconds only. I don't know why it happens this way but I'd accept this as a solution.
Other than that I have 2 notes:
1. Since now UART ISR has to be in ram for this to work, is there a downside for this? I don't want the other functionalities of FW to be affected
2. I'm going to reintroduce component by component to my project until I find what is causing this problem.

david-cermak, 2023-02-27:
Hello @AAl-Hadi 

1) The only downside is that `IRAM` space is limited, so if you use some other IRAM optimization you might run out of available space (`idf.py size` to help)
2) You'd probably find out that the more components you add, the more memory gets used (and deinit of the component to help here -- please don't forget to look at the free heap size and `heap_caps_get_largest_free_block()`)

---
PS: The [above snippet](https://github.com/espressif/esp-idf/issues/10719#issuecomment-1428243589) should help when you're running out of the free heap
PPS: About OTA: You can also check the exact stages, what the OTA is currently doing using [this event handler](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_https_ota.html), just to see the differences between OTA over Ethernet and PPPoS. You can also try to increase the baudrate for PPPoS-OTA to check the difference.

AAl-Hadi, 2023-02-27:
Thank you @david-cermak for all of the feedback.
I'm going to close this as solved for now.
After I test all of these, I'll post the results here for reference.

closed 2023-02-27 by AAl-Hadi

AAl-Hadi, 2023-02-28:
I've solved the problem but will writ it here for reference:
1. UART must be in ISR for this to work
2. Above [snippet ](https://github.com/espressif/esp-idf/issues/10719#issuecomment-1428243589) works but I prefer not modify the ESP library
3. The main problem is the resources being used. So, I removed BT code, and I destroy MQTT and MODBUS before OTA. That makes everything work. There's a minor problem, is that that OTA HTTP connection takes 2.5 minutes and I don't know why. The OTA writing itself takes few seconds.
Overall, I'm OK with the result