espressif/esp-idf #10720: dhcps: The subnet mask does not match the A address.

0xFEEDC0DE64, 2023-02-07:
Since our latest esp-idf rebase the DHCP server on the AP interface stopped working, we have had a quite unusual ip address range to avoid conflicts of customer's smart phones simulaniusly connected to multiple networks (cellular network and home networks).

Our selected AP ip range was 10.128.250.181 (255.255.255.0).

The error log in question:

```
I (2869) WIFI_STACK espwifistack.cpp:1604 wifi_event_cb(): event_base=WIFI_EVENT event_id=WIFI_EVENT_AP_START                                                 
I (2889) WIFI_STACK espwifistack.cpp:2178 wifi_set_ap_ip(): ip=10.128.250.181 subnet=255.255.255.0 gateway=10.128.250.181                                     
I (2889) esp_netif_lwip esp_netif_lwip.c:1037 esp_netif_start_api(): DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1                        
I (2909) WIFI_STACK espwifistack.cpp:2194 wifi_set_ap_ip(): esp_netif_dhcps_get_status() resulted in STARTED                                                  
dhcps: The subnet mask does not match the A address.                                                                                                          
E (2929) esp_netif_lwip esp_netif_lwip.c:1440 esp_netif_dhcps_start_api(): DHCP server cannot be started                                                      
E (2939) WIFI_STACK espwifistack.cpp:2270 wifi_set_ap_ip(): esp_netif_dhcps_start() failed with ESP_ERR_ESP_NETIF_DHCPS_START_FAILED                          
E (2949) WIFI_STACK espwifistack.cpp:328 init(): wifi_set_ap_ip() failed with ESP_ERR_ESP_NETIF_DHCPS_START_FAILED                                            
W (4229) wifi: [ieee80211_power.c,111] Total power save buffer number: 4                                                                                      
I (4229) WIFI_STACK espwifistack.cpp:1609 wifi_event_cb(): event_base=WIFI_EVENT event_id=WIFI_EVENT_AP_STOP                                                  
I (4229) WIFI_STACK espwifistack.cpp:1604 wifi_event_cb(): event_base=WIFI_EVENT event_id=WIFI_EVENT_AP_START                                                                                                                                                                                                                
dhcps: The subnet mask does not match the A address.
```

I can avoid this error by commenting out a line of code in esp-idf (but I would like to discuss some other solution, maybe sdkconfig?)

```
$ git diff
diff --git a/components/lwip/apps/dhcpserver/dhcpserver.c b/components/lwip/apps/dhcpserver/dhcpserver.c
index 1475b49c1a..1ff632ada6 100644
--- a/components/lwip/apps/dhcpserver/dhcpserver.c
+++ b/components/lwip/apps/dhcpserver/dhcpserver.c
@@ -1272,7 +1272,7 @@ err_t dhcps_start(dhcps_t *dhcps, struct netif *netif, ip4_addr_t ip)
     IP4_ADDR(&dhcps->broadcast_dhcps, 255, 255, 255, 255);
 
     dhcps->server_address.addr = ip.addr;
-    DHCP_CHECK_SUBNET_MASK_IP(htonl(dhcps->dhcps_mask.addr), htonl(dhcps->server_address.addr));
+    // DHCP_CHECK_SUBNET_MASK_IP(htonl(dhcps->dhcps_mask.addr), htonl(dhcps->server_address.addr));
     dhcps_poll_set(dhcps, dhcps->server_address.addr);
 
     dhcps->client_address_plus.addr = dhcps->dhcps_poll.start_ip.addr;
```

igrr, 2023-02-07:
This might be the same issue as reported in https://github.com/espressif/esp-idf/issues/10559.

AxelLin, 2023-02-13:
@0xFEEDC0DE64 
Could you please always show the esp-idf version (git describe --tags output) in your issue report?

AxelLin, 2023-03-19:
@0xFEEDC0DE64 
The fix is in master(4f0732c), v5.0 (392533f) and v4.4 (109eec403334), please check if it fixed your issue.

xueyunfei998, 2023-03-30:
hi @AxelLin 
The internal mr has been submitted and will be merged soon.

0xFEEDC0DE64, 2023-04-04:
> you please always show the esp-idf version (git describe --tags output) in y

https://github.com/0xFEEDC0DE64/esp-idf

i will check if the latest master has it fixed

AxelLin, 2023-10-19:
@0xFEEDC0DE64 Do you still hit the issue?

0xFEEDC0DE64, 2023-10-23:
we still have the issue
i have turned on logging again to cirumvent the issue

closed 2023-10-23 by 0xFEEDC0DE64

xueyunfei998, 2023-10-31:
hi @0xFEEDC0DE64 
The latest idf has fixed this problem.