espressif/esp-idf #10726: esp_http_client_read seems to return wrong data in OTA Update Process

franz-ms-muc, 2023-02-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF Pre-release v5.0-rc1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I expected a clean OTA Process. 

### What is the actual behavior?

but sometimes there are 16bit Portions in the Flash corrupted, and therefore the OTA Checksum failes. 

sometimes also sectors gets filled with FF, 

[image]

### Steps to reproduce.

In the Moment we cannot let you reproduce this, 
it is happening in our closed-Source software, 
and we could not see it in the Samples, nor reproduce it. 
so we need Support in detecting and fixing the Error.

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1838
load:0x403c9700,len:0xbd0
load:0x403cc700,len:0x2f7c
entry 0x403c9900
I (25) boot: ESP-IDF v5.0-rc1-dirty 2nd stage bootloader
I (25) boot: compile time 17:34:39
I (25) boot: chip revision: V001
I (27) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
[... 456 lines cut ...]

0x40382b79: esp_system_abort at C:/Espressif/frameworks/esp-idf/components/esp_system/esp_system.c:135

0x40389f85: __assert_func at C:/Espressif/frameworks/esp-idf/components/newlib/assert.c:78

0x40388a17: tlsf_free at C:/Espressif/frameworks/esp-idf/components/heap/tlsf/tlsf.c:1049 (discriminator 1)

0x40388354: multi_heap_free_impl at C:/Espressif/frameworks/esp-idf/components/heap/multi_heap.c:241

0x40376976: heap_caps_free at C:/Espressif/frameworks/esp-idf/components/heap/heap_caps.c:353

0x40389fb5: free at C:/Espressif/frameworks/esp-idf/components/newlib/heap.c:39

0x4209c8e7: esp_https_ota_abort at C:/Espressif/frameworks/esp-idf/components/esp_https_ota/src/esp_https_ota.c:639

0x42026788: OTA_AbortUpdate at C:/Projects/ksc-2n/src/APPL/OTA/OTA.c:772

0x42026ae1: OTA_MainFunction at C:/Projects/ksc-2n/src/APPL/OTA/OTA.c:415

0x42126657: osw::Event::run() const at C:/Projects/ksc-2n/src/OS/os_wrapper/src/Event.cpp:27

0x42010cdd: osw::Task<1u>::run(void*) at C:/Projects/ksc-2n/src/OS/os_wrapper/include/Task.hpp:213

0x40386516: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```

Alvin1Zhang, 2023-02-09:
Thanks for reporting.

franz-ms-muc, 2023-02-09:
Hello Franz,

 

Changing the TLS memory setting from SPIRAM to Internal RAM seems to solve the problem.

Made longterm flash in the loop test with one application updating the other => no checksum verification failures.

However, with the same setting in the example project advanced_https_ota I can’t still force it to break so that I can show the issue to ESP.

 

Best regards,

Martin

hmalpani, 2023-02-17:
Hello @franz-ms-muc
Could you also share the sdkconfig file? It will be helpful in reproducing the issue. Thanks!

mahavirj, 2023-02-27:
@franz-ms-muc 

I suspect this could be related to memory corruption. Please use guidelines from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html.

Alternatively, if you would recreate the issue under any IDF example then please share us the instructions.

hmalpani, 2023-03-15:
Hello @franz-ms-muc 
Any update on this issue?

Alvin1Zhang, 2023-03-27:
@franz-ms-muc Thanks for reporting, would you please help share if any updates for the issue? Thanks.

mahavirj, 2023-04-03:
Closing. Please drop a comment if you are still facing a issue here..

closed 2023-04-03 by mahavirj