espressif/esp-idf #10728: How to Scan the QR Code on the ESP32S3 Module ?

franz-ms-muc, 2023-02-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://www.espressif.com/sites/default/files/documentation/espressif_module_packaging_information_en.pdf

need to scan the QR Code. 

which App does do this good ? 
i tested a couple, and none seems to work. 

Thanks, 
Franz

igrr, 2023-02-08:
I'm sorry, this really is not an ESP-IDF related question. Please use https://www.espressif.com/en/contact-us/technical-inquiries/other-technical-issues or https://www.espressif.com/en/contact-us/technical-inquiries/hardware-issues for questions related to the modules or hardware.

(I have personally used https://play.google.com/store/apps/details?id=com.datamatrix.qrscanner which worked okay. It helps if you point the camera at such an angle that the light source reflects off the module surface, resulting in higher contrast.)

closed 2023-02-08 by igrr

diplfranzhoepfinger, 2023-02-15:
Thanks a lot for the Answer!!

i wrote to the Espressif Support: 

The Answer was: 

> Hi Franz,
> 
> Thanks for your mail.
> 
> Kindly please note that the QR code scanning on the mobile phone is limited by the resolution of the mobile phone camera, so the APP may not scan it sucessfully.
> 
> We would recommend you to use a barcode scanner to scan the QR code on the module.
> 
> Thank you.
> 
> 
> 
> Best regards,
> Emily Cai 蔡娟
> Business Support | Espressif Systems

i will order a Barcode Scanner, therefore i have sent some Modules to my Barcode House for Evaluation.

diplfranzhoepfinger, 2023-02-15:
But i found another Method: 

this App on Android can read the Barcode, (Data Matrix Code)
hard to read, but can: 

https://f-droid.org/packages/com.atharok.barcodescanner/

unfortunately it is not official Google Play Store App. 

Thanks, 
Franz

franz-ms-muc, 2023-02-16:
we found a 2nd way: 

[https://online-barcode-reader.inliteresearch.com/](https://online-barcode-reader.inliteresearch.com/)