espressif/esp-idf #10731: Wrong wakeup cause in example deep sleep

Lerqiu, 2023-02-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3278-g668eb92d4f

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Adafruit QT Py ESP32-S2

### Power Supply used.

USB

### What is the expected behavior?

Print at second wake up: "Wake up from timer."

### What is the actual behavior?

Print at second wake up: "Not a deep sleep reset"

### Steps to reproduce.

1. At menuconfig "Example Configuration" disable everything (this way only timer is left)
2. At menuconfig set "Channel for console output" to  "USB CDC"

### Debug Logs.

```plain
Waiting for the device to reconnect.
I (34) boot: ESP-IDF v5.1-dev-3278-g668eb92d4f 2nd stage bootloader
I (35) boot: compile time Feb  8 2023 19:59:41
I (35) boot: chip revision: v0.0
I (35) boot.esp32s2: SPI Speed      : 80MHz
I (35) boot.esp32s2: SPI Mode       : DIO
I (35) boot.esp32s2: SPI Flash Size : 2MB
I (37) boot: Enabling RNG early entropy source...
I (37) boot: Partition Table:
I (37) boot: ## Label            Usage          Type ST Offset   Length
I (38) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (39) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (40) boot:  2 factory          factory app      00 00 00010000 00100000
I (42) boot: End of partition table
I (42) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=082dch ( 33500) map
I (49) esp_image: segment 1: paddr=00018304 vaddr=3ff9e02c size=00004h (     4) load
I (50) esp_image: segment 2: paddr=00018310 vaddr=3ffc0160 size=017e8h (  6120) load
I (52) esp_image: segment 3: paddr=00019b00 vaddr=40024000 size=06518h ( 25880) load
I (58) esp_image: segment 4: paddr=00020020 vaddr=40080020 size=1329ch ( 78492) map
I (74) esp_image: segment 5: paddr=000332c4 vaddr=4002a518 size=05c40h ( 23616) load
I (80) esp_image: segment 6: paddr=00038f0c vaddr=40070000 size=0002ch (    44) load
I (81) esp_image: segment 7: paddr=00038f40 vaddr=50000200 size=00010h (    16) load
I (88) boot: Loaded app from partition at offset 0x10000
I (89) boot: Disabling RNG early entropy source...
I (110) main_task: Started on CPU0
I (110) main_task: Calling app_main()
Not a deep sleep reset
Enabling timer wakeup, 20s
Entering deep sleep
None
Waiting for the device to reconnect........................................
I (34) boot: ESP-IDF v5.1-dev-3278-g668eb92d4f 2nd stage bootloader
I (35) boot: compile time Feb  8 2023 19:59:41
I (35) boot: chip revision: v0.0
I (35) boot.esp32s2: SPI Speed      : 80MHz
I (35) boot.esp32s2: SPI Mode       : DIO
I (35) boot.esp32s2: SPI Flash Size : 2MB
I (36) boot: Enabling RNG early entropy source...
I (36) boot: Partition Table:
I (36) boot: ## Label            Usage          Type ST Offset   Length
I (37) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (38) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (39) boot:  2 factory          factory app      00 00 00010000 00100000
I (41) boot: End of partition table
I (41) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=082dch ( 33500) map
I (48) esp_image: segment 1: paddr=00018304 vaddr=3ff9e02c size=00004h (     4) load
I (49) esp_image: segment 2: paddr=00018310 vaddr=3ffc0160 size=017e8h (  6120) load
I (52) esp_image: segment 3: paddr=00019b00 vaddr=40024000 size=06518h ( 25880) load
I (59) esp_image: segment 4: paddr=00020020 vaddr=40080020 size=1329ch ( 78492) map
I (75) esp_image: segment 5: paddr=000332c4 vaddr=4002a518 size=05c40h ( 23616) load
I (82) esp_image: segment 6: paddr=00038f0c vaddr=40070000 size=0002ch (    44) load
I (84) esp_image: segment 7: paddr=00038f40 vaddr=50000200 size=00010h (    16) load
I (90) boot: Loaded app from partition at offset 0x10000
I (91) boot: Disabling RNG early entropy source...
I (112) main_task: Started on CPU0
I (112) main_task: Calling app_main()
Not a deep sleep reset
Enabling timer wakeup, 20s
Entering deep sleep
```

### More Information.

Board info: https://www.adafruit.com/product/5325

Disabling "Brownout Detector" in "ESP System Settings" don't change anything. I also tried with "Use RTC watchdog in start code" at off.

Lerqiu, 2023-02-18:
Indeed connection to machine via usb was the problem. Test with battery and communication via WiFi return correct value.

ESP-Marius, 2023-04-27:
As mentioned [here](https://docs.espressif.com/projects/esp-idf/en/v4.3.1/esp32s2/api-guides/usb-console.html): 

> If the application enters light sleep (including automatic light sleep) or deep sleep mode, USB CDC device will disappear from the system.

So `idf.py monitor` will lose the connection and try to reconnect. The default reconnect behavior is to reset the chip, which is what you are seeing here.

If you connect with `idf.py monitor --no-reset ` then I think you should be able to see the correct reset cause, even with USB CDC.

closed 2023-04-27 by ESP-Marius

AxelLin, 2023-04-27:
> If you connect with `idf.py monitor --no-reset ` then I think you should be able to see the correct reset cause, even with USB CDC.

No, this won't work.
After lightsleep or deepslep, if USB CDC device is disappeared then "idf.py monitor --no-reset" won't work.
 (It works sometimes if USB CDC device is not disappear during lightsleep)
I have to unplug and replug the USB cable to make it work.

ESP-Marius, 2023-04-27:
@AxelLin ah, thanks for helping confirm this. Then yeah, no easy way around this hardware limitation then.