espressif/esp-idf #10732: Soft AP on ESP-WIFI-MESH

eduardobehr, 2023-02-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to create a mesh network with an exposed soft AP interface serving a page via HTTP (so that other devices may connect to the mesh nodes and root).

__My question is: is it possible to make such a server with AP from the nodes of a mesh?__

The docs for `esp_netif_create_default_wifi_mesh_netifs()` mention that the ESP-MESH API disables both DHCP client and server:

```
Creates default STA and AP network interfaces for esp-mesh. 
Both netifs are almost identical to the default station and soft AP, 
but with DHCP client and server disabled. 
Please note that the DHCP client is typically enabled only if the device is promoted to a root node. 
Returns created interfaces which could be ignored setting parameters to NULL if an 
application code does not need to save the interface instances for further processing.
```

Here is an example code:

```c++
// initialize netif
ESP_ERROR_CHECK(esp_netif_init());
ESP_ERROR_CHECK(esp_event_loop_create_default());
ESP_ERROR_CHECK(esp_netif_create_default_wifi_mesh_netifs(&m_netif_sta, &m_netif_ap));

// initialize and start WiFi
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init(&cfg));
ESP_ERROR_CHECK(esp_wifi_start());

// initialize mesh
esp_mesh_init();
ESP_ERROR_CHECK(esp_netif_dhcps_start(m_netif_ap));
```

When I try to enable the DHCP server to use the ESP as an access point, it crashes:

```sh
dhcps = 0x0, netif = 0x3ffc4dec
E (2292) esp_netif_lwip: DHCP server cannot be started (err -16)
ESP_ERROR_CHECK failed: esp_err_t 0x500d (ESP_ERR_ESP_NETIF_DHCPS_START_FAILED) at 0x400d7c25
```

It seems the dhcps is not being initialized by the API.

---

Also, I've done __another test__ by using `esp_netif_create_default_wifi_sta` and `esp_netif_create_default_wifi_ap` instead of `esp_netif_create_default_wifi_mesh_netifs` without calling `esp_mesh_init` and it worked (I could access the AP with its SSID and password, and open the HTML page). But when I call `esp_mesh_init`, I can no longer connect to its SSID and my external device stays in an infinite reconnection loop with timeout, despite not logging any error.

This is my second attempt:

```c++
ESP_ERROR_CHECK(esp_netif_init());
ESP_ERROR_CHECK(esp_event_loop_create_default());

// ESP_ERROR_CHECK(esp_netif_create_default_wifi_mesh_netifs(&m_netif_sta, &m_netif_ap));

m_netif_sta = esp_netif_create_default_wifi_sta();
m_netif_ap = esp_netif_create_default_wifi_ap();

wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

ESP_ERROR_CHECK(esp_wifi_init(&cfg));

ESP_ERROR_CHECK(esp_wifi_start());

wifi_config_t ap_cfg{};

memcpy(
    ap_cfg.ap.ssid,
    "MyNetwork",
    32
);

memcpy(
    ap_cfg.ap.password,
    "MyPassword",
    64
);

ap_cfg.ap.max_connection = 3;

ap_cfg.ap.authmode = WIFI_AUTH_WPA_WPA2_PSK;

esp_wifi_set_config(wifi_interface_t::WIFI_IF_AP, &ap_cfg);

esp_err_t ret = ESP_ERR_MESH_NOT_INIT;
// ret = esp_mesh_init(); // AP only works if mesh is not initialized

```

zhangyanjiaoesp, 2023-05-24:
@eduardobehr I think the official example `esp-idf/examples/mesh/ip_internal_network` can meet your requirements.

eduardobehr, 2023-07-13:
Thanks, I took a look at the example and I'm trying to make it work for my case.

Do you know if it's possible to access the mesh nodes' access points with external devices? There is the following struct which I think should be used here:

`esp_mesh.h`
```c
/**
 * @brief Mesh softAP configuration
 */
typedef struct {
    uint8_t password[64];              /**< mesh softAP password */
    /**
     * max number of stations allowed to connect in, default 6, max 10
     * = max_connection + nonmesh_max_connection
     */
    uint8_t max_connection;            /**< max mesh connections */
    uint8_t nonmesh_max_connection;    /**< max non-mesh connections */
} mesh_ap_cfg_t;
```

It mentions "non-mesh connections", which I understand by external devices (e.g. Android phone). However, the AP layer doesn't seem to emit an SSID, which could be used by other devices. It seems to generate a hidden network.

eduardobehr, 2023-07-13:
I also asked this on the following commit, which introduced the supposed feature:
https://github.com/espressif/esp-idf/commit/42a1be10db3fdba1c7f2ba4b8a542bc72371ad6e#commitcomment-121519972

zhangyanjiaoesp, 2024-03-05:
@eduardobehr About the non-mesh device connects to mesh network, you can refer to this https://github.com/espressif/esp-idf/pull/8158

zhangyanjiaoesp, 2024-04-23:
@eduardobehr Can this issue be closed?

eduardobehr, 2024-04-23:
I wasn't able to test it for now, but you may close it.

Sherry616, 2024-04-24:
Thanks for reporting, we will close this issue. Feel free to reopen if have more updates.

closed 2024-04-24 by Sherry616