espressif/esp-idf #10734: mqtt(or wifi)speed unstable?

weizzh, 2023-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi

In our project, I watched the network is not smooth enough, so I made a test for this. 
hdware: esp32s3, skd: 4.4.2
See line (10382) and (10612). Sometimes it took too much more time to publish same size data.
Code attached below, just modify wifi configuration in menuconfig and mqtt config in app_main.c(line 107-110).
[mqtt_test.zip](https://github.com/espressif/esp-idf/files/10693191/mqtt_test.zip)
[image]

nishanth-radja, 2023-02-09:
@weizzh can you help me more about the network-env to check if its wifi issue.
1.Is there other wifi devices around,Is it a very noisy env?
2.what AP you are using and what security is used.
3.Can we have a sniffer capture during the data transfer.

weizzh, 2023-02-09:
@nishanth-radja 
1. I'm afraid yes. I did this test in our office.There are several devices, cellphones and laptops connecting to the router.
2. I checked our wireless router. It's TL-XDR5430, and the security type currently used is WPA2.
3. I didn't have a snifer on hand. If we cann't fix this problem through coding, I will get one sniffer to continue test.

nishanth-radja, 2023-02-09:
@weizzh can you try increasing the buffer size and see if there is any improvement.

CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=12
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_TX_BUFFER_TYPE=1
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=y
CONFIG_ESP32_WIFI_TX_BA_WIN=6
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=6
CONFIG_ESP32_WIFI_MGMT_SBUF_NUM=32

CONFIG_LWIP_TCP_SND_BUF_DEFAULT=12000
CONFIG_LWIP_TCP_WND_DEFAULT=12000

david-cermak, 2023-02-24:
@weizzh MQTT runs on top of TCP and TCP is not a real-time communication protocol. (but from the numbers you shared I would say it worked *pretty well*)

Are you connecting to a local MQTT broker? If not, then you have to find out if the issues are in the WiFi/local network or somewhere else. Wireshare is a good tool to quickly check and identify the issues in your network. Alternatively, you can setup a local broker and test with it.

It's also good to think about a better metrics to evaluating network performance. From what I can see in your screenshot, I could only guess that you start the timer before `esp_mqtt_publish()` and stop the timer after it. This is not very scalable, since it only calculates the time of copying your local buffers to lwIP buffers. The actual sending takes place in an asynchronous task.
One possible reason of slower publish after each ~ 20 attempts could be running out of lwip TCP buffer and waiting for the asynchronous sending. Increasing lwip buffers would help, but IMO with almost zero effect on network performance.

Alvin1Zhang, 2023-08-07:
@weizzh Thanks for reporting, would please help share if any updates for the issue? Thanks.

closed 2024-07-01 by euripedesrocha