espressif/esp-idf #10736: transaction buffer size for spi_bus_initialize

niexuzhong, 2023-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

there may be error in function spi_bus_initialize, there is an parameter named bus_config, an structure with an parameter called max_transfer_sz, even if you use spi_bus_initialize with your own transancation buffer size in bus_config and disable dma. this function will directly set the transaction buffer size as SOC_SPI_MAXIMUM_BUFFER_SIZE so if you set buffer size to be larger than SOC_SPI_MAXIMUM_BUFFER_SIZE(default value is 64) in this function  when you transmit the larger data than 64 bytes later,it will still display tx data > host maxium. then i think that you should change the function to set transaction size as buffer size in bus_config. or if it is user use an buffer size  larger than SOC_SPI_MAXIMUM_BUFFER_SIZE.please return error,otherwise. the initialization will be ok. but it will report an error when transmit large data.

wanckl, 2023-02-09:
@niexuzhong Hello? :smile: 
In actually, you can simply set DMA option to `SPI_DMA_CH_AUTO`, in this case you can set any value larger than your buffer size to `max_transfer_sz` because DMA will control the data during transaction,,,,If any reasons you must disable DMA for SPI, then CPU need reconfigure a transaction for SPI peripheral after every transaction finished, and the max trans length for a transaction is `SOC_SPI_MAXIMUM_BUFFER_SIZE(64)` bytes,,,,,esp-idf is a up layer driver for embedded development, you don't need to configure the DMA or some channel boringly.
So, if you want transact a buffer larger than 64 bytes, you can init spi bus as `SPI_DMA_CH_AUTO`, or split into several transactions without DMA,,
If still any question, feel free to ask me again :grinning:

Maksons, 2023-02-13:
+1 
Nice catch @niexuzhong  
"spi_bus_initialize()" should retturn error if "max_transfer_sz" exceeds hardware limits.
"max_transfer_sz" should be taken in account even if lower then hardware limits and without DMA.

But take a note, "max_transfer_sz" is limitation which is by default equal to real hardware limitation.
So, user could set this limit lower than HW but not higher to get working result.
For SPI without DMA this limit is 64 bytes (it's size of SPI controller internal buffer hardware work with).
For SPI with DMA, the data comes from DMA and and with it limits (buffer must be 4 bytes alligned, etc).

If you are interested in both type transactions with and without DMA to same device, vote for pull request #9069 and maybe try it.

closed 2023-02-18 by niexuzhong