espressif/esp-idf #10738: ./esp-idf/export.sh: line 204: complete: nosort: invalid option name

MATTYGILO, 2023-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5bb59b00e72f8f91eb24d8c65bf9a7ea2b8a4f5f

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

I want to run `source ./esp-idf/export.sh` to run all the required code for a ci

### What is the actual behavior?

The command fails and returns this error

> ./esp-idf/export.sh: line 204: complete: nosort: invalid option name

### Steps to reproduce.

Run esp-idf/export.sh idf 4.4 on a Mac m1

dobairoland, 2023-02-10:
@MATTYGILO Are you sure that you have one copy of ESP-IDF on your computer and you are using the right one? The reported issue is for line 204 but the file has only 188 lines in the v4.4 branch (https://github.com/espressif/esp-idf/blob/release/v4.4/export.sh).

Could you please share also you bash version? (if you use other shell then please state which one and also share the version)

dobairoland, 2023-02-10:
Ok, I see from the commit that you are actually using the master ESP-IDF (v5.1 in development). Is this intentional?

closed 2023-02-27 by dobairoland