espressif/esp-idf #10739: ESP32 freezes when wrong wifi password is used with WPA3

0xFEEDC0DE64, 2023-02-09:
We have rebased our esp-idf recently and we are facing a strange issue with wifi connection. A customer enters a wrong wifi password and we try to connect to it and the main task completely freezes until the watchdog kicks in and reboots.

We were able to capture the coredump remotely on the customers site and the stack shows code in regards to WPA3.

We cannot reproduce this on the developers side, we also have WPA3 access points but we receive the correct events in the case of wrong passwords.

This is the stacktrace from the customer with a freezing ESP32:
[image]

Any ideas?

nishanth-radja, 2023-02-13:
@0xFEEDC0DE64 what is the IDF version you are using?

0xFEEDC0DE64, 2023-02-13:
My fork bases on a 200 commits old esp-idf master

https://github.com/0xFEEDC0DE64/esp-idf/commits/fixes

0xFEEDC0DE64, 2023-02-13:
Customer reported that he has a german telekom.de modem/router/access-point device (no wonder everything goes bad when the deutsche telekom is involved :D)

Sadly we only get the coredumps of the product, when the correct password is entered and the esp32 connects to our cloud and send the file. The coredump does not contain any log output right before the device freezes or the crash. Thus it is very hard to debug.

vik-gokhale, 2023-02-14:
@0xFEEDC0DE64 
Is being on Master branch needed?

We don't remember having this problem with wrong WPA3 password on v4.4/v5.0.
We are sure this problem does not occur on the latest commits of these branches.
Please move the devices to the latest commit on v4.4 / v5.0 branch and keep this thread posted?

0xFEEDC0DE64, 2023-02-27:
yes, we are using latest API of the esp-idf, and it would probably also happen with older IDF versions, it just doesnt happen with every access point, only with this customers one. I just saw a wifi blob commit that has to do with WPA3 SAE PK, we will try again.

0xFEEDC0DE64, 2023-02-28:
New wifi blob seems to fix the issue :+1:  (or deutsche telekom installed an update on the customer's router without him knowing).

I still dont understand why it froze in the first place, I hope some day some hackers publish a decompilation of the closed source wifi blobs and we can take a look at it ourself.

closed 2023-02-28 by 0xFEEDC0DE64