espressif/esp-idf #10742: Pair Bluetooth Classic with more than 20 devices

samir213, 2023-02-10:
### Is your feature request related to a problem?

I have a project in which I want to share an information twice or more aday via Bluetooth Classic with about 50 Android devices that use "Simple Bluetooth Terminal" android App (Every device holder opens the App in proximity of the ESP32 board to get the information). My concern is that ESP32 Bt classic pairing is limited to 20 devices only.

### Describe the solution you'd like.

I think it would be useful to allow user to deal with the pairing process, or at least set this limit at his convinience.

### Describe alternatives you've considered.

In the mean time, I am just using it as it is so that at every new pairing obove the limit new devices overwrite old ones, but doubt wether this would harm the ESP32 microcontroller or is it safe for a long term run.

xiongweichao, 2023-03-30:
Hi @samir213 ,

You can try to increase the [value.](https://github.com/espressif/esp-idf/blob/81e1e6599553a646a689ad51e32a5d48b34cfec5/components/bt/host/bluedroid/common/include/common/bt_target.h#L855)

Thanks

xiongweichao, 2024-02-22:
@samir213 On the latest branch you can modify the number of allowed pairs through menuconfig.

Sherry616, 2024-03-28:
Thanks for reporting and we will close this issue. Feel free to reopen with more updates, thanks.

closed 2024-03-28 by Sherry616