espressif/esp-idf #10744: Flash onto the Device

osmond2023, 2023-02-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

the board will reboot and start up the “hello_world” application.

### What is the actual behavior?

 fatal error occurred: Could not open ttyUSB0, the port doesn't exist
CMake Error at run_serial_tool.cmake:55 (message):
  
  /home/os/.espressif/python_env/idf5.0_py3.11_env/bin/python;;/home/os/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s2
  failed

FAILED: CMakeFiles/flash /home/os/esp/hello_world/build/CMakeFiles/flash 

### Steps to reproduce.

[os@fedora hello_world]$ idf.py -p ttyUSB0 flash

### Build or installation Logs.

```plain
[os@fedora hello_world]$ idf.py -p ttyUSB0 flash
Executing action: flash
Running ninja in directory /home/os/esp/hello_world/build
Executing "ninja flash"...
[1/5] cd /home/os/esp/hello_world/build/esp-idf/esptool_py && /home/os/.espressif/python_env/idf5.0_py3.11_env/bin/python /home/os/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/os/esp/hello_world/build/partition_table/partition-table.bin /home/os/esp/hello_world/build/hello_world.bin
hello_world.bin binary size 0x274f0 bytes. Smallest app partition is 0x100000 bytes. 0xd8b10 bytes (85%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/os/esp/hello_world/build/bootloader/esp-idf/esptool_py && /home/os/.espressif/python_env/idf5.0_py3.11_env/bin/python /home/os/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/os/esp/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x5430 bytes. 0x1bd0 bytes (25%) free.
[3/5] No install step for 'bootloader'
[4/5] Completed 'bootloader'
[4/5] cd /home/os/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/os/esp/esp-idf -D "SERIAL_TOOL=/home/os/.espressif/python_env/idf5.0_py3.11_env/bin/python;;/home/os/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s2" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/os/esp/hello_world/build -P /home/os/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32s2 -p ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.4
Serial port ttyUSB0

A fatal error occurred: Could not open ttyUSB0, the port doesn't exist
CMake Error at run_serial_tool.cmake:55 (message):
  
  /home/os/.espressif/python_env/idf5.0_py3.11_env/bin/python;;/home/os/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s2
  failed

FAILED: CMakeFiles/flash /home/os/esp/hello_world/build/CMakeFiles/flash 
cd /home/os/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/os/esp/esp-idf -D "SERIAL_TOOL=/home/os/.espressif/python_env/idf5.0_py3.11_env/bin/python;;/home/os/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s2" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/os/esp/hello_world/build -P /home/os/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/os/esp/hello_world/build/log/idf_py_stderr_output_16850 and /home/os/esp/hello_world/build/log/idf_py_stdout_output_16850
```

### More Information.

ESP32-S2-DevKitM-1
windows 10 
Oracle VM VirtualBOX
Fedora-Workstation-Live-x86_64-37-1.7

first with the board / dongle unplugged :
/dev/tty    /dev/tty16  /dev/tty24  /dev/tty32  /dev/tty40  /dev/tty49  /dev/tty57  /dev/tty8    /dev/ttyS15  /dev/ttyS23  /dev/ttyS31
/dev/tty0   /dev/tty17  /dev/tty25  /dev/tty33  /dev/tty41  /dev/tty5   /dev/tty58  /dev/tty9    /dev/ttyS16  /dev/ttyS24  /dev/ttyS4
/dev/tty1   /dev/tty18  /dev/tty26  /dev/tty34  /dev/tty42  /dev/tty50  /dev/tty59  /dev/ttyS0   /dev/ttyS17  /dev/ttyS25  /dev/ttyS5
/dev/tty10  /dev/tty19  /dev/tty27  /dev/tty35  /dev/tty43  /dev/tty51  /dev/tty6   /dev/ttyS1   /dev/ttyS18  /dev/ttyS26  /dev/ttyS6
/dev/tty11  /dev/tty2   /dev/tty28  /dev/tty36  /dev/tty44  /dev/tty52  /dev/tty60  /dev/ttyS10  /dev/ttyS19  /dev/ttyS27  /dev/ttyS7
/dev/tty12  /dev/tty20  /dev/tty29  /dev/tty37  /dev/tty45  /dev/tty53  /dev/tty61  /dev/ttyS11  /dev/ttyS2   /dev/ttyS28  /dev/ttyS8
/dev/tty13  /dev/tty21  /dev/tty3   /dev/tty38  /dev/tty46  /dev/tty54  /dev/tty62  /dev/ttyS12  /dev/ttyS20  /dev/ttyS29  /dev/ttyS9
/dev/tty14  /dev/tty22  /dev/tty30  /dev/tty39  /dev/tty47  /dev/tty55  /dev/tty63  /dev/ttyS13  /dev/ttyS21  /dev/ttyS3
/dev/tty15  /dev/tty23  /dev/tty31  /dev/tty4   /dev/tty48  /dev/tty56  /dev/tty7   /dev/ttyS14  /dev/ttyS22  /dev/ttyS30

then with plugged in :
/dev/tty    /dev/tty16  /dev/tty24  /dev/tty32  /dev/tty40  /dev/tty49  /dev/tty57  /dev/tty8    /dev/ttyS15  /dev/ttyS23  /dev/ttyS31
/dev/tty0   /dev/tty17  /dev/tty25  /dev/tty33  /dev/tty41  /dev/tty5   /dev/tty58  /dev/tty9    /dev/ttyS16  /dev/ttyS24  /dev/ttyS4
/dev/tty1   /dev/tty18  /dev/tty26  /dev/tty34  /dev/tty42  /dev/tty50  /dev/tty59  /dev/ttyS0   /dev/ttyS17  /dev/ttyS25  /dev/ttyS5
/dev/tty10  /dev/tty19  /dev/tty27  /dev/tty35  /dev/tty43  /dev/tty51  /dev/tty6   /dev/ttyS1   /dev/ttyS18  /dev/ttyS26  /dev/ttyS6
/dev/tty11  /dev/tty2   /dev/tty28  /dev/tty36  /dev/tty44  /dev/tty52  /dev/tty60  /dev/ttyS10  /dev/ttyS19  /dev/ttyS27  /dev/ttyS7
/dev/tty12  /dev/tty20  /dev/tty29  /dev/tty37  /dev/tty45  /dev/tty53  /dev/tty61  /dev/ttyS11  /dev/ttyS2   /dev/ttyS28  /dev/ttyS8
/dev/tty13  /dev/tty21  /dev/tty3   /dev/tty38  /dev/tty46  /dev/tty54  /dev/tty62  /dev/ttyS12  /dev/ttyS20  /dev/ttyS29  /dev/ttyS9
/dev/tty14  /dev/tty22  /dev/tty30  /dev/tty39  /dev/tty47  /dev/tty55  /dev/tty63  /dev/ttyS13  /dev/ttyS21  /dev/ttyS3   /dev/ttyUSB0
/dev/tty15  /dev/tty23  /dev/tty31  /dev/tty4   /dev/tty48  /dev/tty56  /dev/tty7   /dev/ttyS14  /dev/ttyS22  /dev/ttyS30

dobairoland, 2023-02-10:
Hi @osmond2023. The command should be `idf.py -p /dev/ttyUSB0 flash` including the `/dev/` part.

osmond2023, 2023-02-13:
Hi dobairoland,
thank you.

[os@fedora hello_world]$ idf.py -p /dev/ttyUSB0 flash
Executing action: flash
Running ninja in directory /home/os/esp/hello_world/build
Executing "ninja flash"...
[1/5] cd /home/os/esp/hello_world/build/esp-idf/esptool_py && /home/os/.espressif/python_env/idf5.0_py3.11_env/bin/python /home/os/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/os/esp/hello_world/build/partition_table/partition-table.bin /home/os/esp/hello_world/build/hello_world.bin
hello_world.bin binary size 0x25f10 bytes. Smallest app partition is 0x100000 bytes. 0xda0f0 bytes (85%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/os/esp/hello_world/build/bootloader/esp-idf/esptool_py && /home/os/.espressif/python_env/idf5.0_py3.11_env/bin/python /home/os/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/os/esp/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x5570 bytes. 0x1a90 bytes (24%) free.
[3/5] No install step for 'bootloader'
[4/5] Completed 'bootloader'
[4/5] cd /home/os/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/home/os/esp/esp-idf -D "SERIAL_TOOL=/home/os/.espressif/python_env/idf5.0_py3.11_env/bin/python;;/home/os/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s2" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/os/esp/hello_world/build -P /home/os/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32s2 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.4
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-S2FH4 (revision v0.0)
Features: WiFi, Embedded Flash 4MB, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
MAC: 84:f7:03:e3:37:7a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00010000 to 0x00035fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 21872 bytes to 13839...
Writing at 0x00001000... (100 %)
Wrote 21872 bytes (13839 compressed) at 0x00001000 in 0.6 seconds (effective 284.0 kbit/s)...
Hash of data verified.
Compressed 155408 bytes to 87650...
Writing at 0x00010000... (16 %)
Writing at 0x0001a3de... (33 %)
Writing at 0x0001fe1c... (50 %)
Writing at 0x000257c9... (66 %)
Writing at 0x0002ba1c... (83 %)
Writing at 0x00033dc5... (100 %)
Wrote 155408 bytes (87650 compressed) at 0x00010000 in 2.2 seconds (effective 562.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 354.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done

dobairoland, 2023-02-13:
Ok, great. I'm happy that it works.

closed 2023-02-13 by dobairoland