espressif/esp-idf #10748: ESP32 WROOM STA and DHCP waits for IP address indefinitely

sergioaiello, 2023-02-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROOM-32E

### Power Supply used.

USB

### What is the expected behavior?

This case might be in between a feature request. Sorry if I wrongly consider it a case. Since it is very easy to reproduce, I looked for a former existing. I did not find any, sorry if I missed it.

According to my observation: the example_connect does not return if DHCP server does not deliver an IP address (IPv4)

In IDF 4.4 I got the same behavior in case the ESP cannot associate to the AP.
In 5.0 it is possible to define the retries' number and do not stop indefinitely.
I expect the same behavior applies for IP address assignment.
Rationale: even if the ESP32 with no IP address has a very limited value, in my implementations samples are collected in an internal log-file cyclically uploaded. Therefore it is preferable the ESP32 never stops indefinitely. I think this behavior should generally rule as in principle.

### What is the actual behavior?

example_connect does not return so the ESP32 waits indefinitely for an IP address.

### Steps to reproduce.

1. provide a WIFI radio acting as a AP but with no DHCP service
2. unplug the ethernet cable
3. let the ESP32 associate to the AP and monitor its state

### Debug Logs.

```plain
Please, let me know which log do you need and I'll provide them to you.
Thank you.
```

igrr, 2023-02-11:
Hi @sergioaiello,

In the readme of the [protocols-related examples](https://github.com/espressif/esp-idf/tree/master/examples/protocols#establishing-wi-fi-or-ethernet-connection) we do point out that:

> Protocols examples use a simple helper function, example_connect(), to establish Wi-Fi and/or Ethernet connection. This function is implemented in [protocol_examples_common.h](https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/include/protocol_examples_common.h), and has a very simple behavior: block until connection is established and IP address is obtained, then return. This function is used to reduce the amount of boilerplate and to keep the example code focused on the protocol or library being demonstrated.

You shouldn't use this function in your real application. Instead, please study the Wi-Fi or Ethernet related examples and implement the connection management logic suitable for your application.

sergioaiello, 2023-02-13:
@Igrr, thank you for your useful message and sorry so much, my limited investigations lead me to the protocol example only.
Based on your message other examples are available.
Could you suggest the most flexible (hence complex) one?
I'll take the suggested example as my new reference.
If no more hints, then I'll close this case.

closed 2023-02-14 by sergioaiello