espressif/esp-idf #10749: [linux] Driver component no longer builds

higaski, 2023-02-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3296-g76433e4cb2

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The driver component no longer builds when setting `linux` as target.

### What is the actual behavior?

CMake produces an error when trying to build the project

```cmake
CMake Error at /home/vinci/esp/esp-idf/tools/cmake/component.cmake:479 (add_library):
  Cannot find source file:

    /home/vinci/esp/esp-idf/components/bootloader_support/src/bootloader_random_linux.c
```

### Steps to reproduce.

1. Copy [hello_world_linux_compatible](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/linux_compatible/hello_world_linux_compatible)
2. Add driver to requirements in main/CMakeLists.txt
`idf_component_register(SRCS "hello_world_main.c" INCLUDE_DIRS "" REQUIRES driver)`
3. idf.py --preview set-target linux
4. idf.py build

ESP-Marius, 2023-02-13:
Hi @higaski

As mentioned [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/host-apps.html#component-linux-mock-support-overview), `driver` is not supported for `simulation`. You could add the cmock driver mocks we have by adding

```list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/driver/")```

to you project CMake file, which would allow you to link the driver (mock) component. 

If you describe what you are trying to achieve by linking the `driver` component then maybe I can suggest a better suited  approach for you.

higaski, 2023-02-13:
I just need the driver headers to not brick a x86 build, that's all.

I've previously been using a combination of the following two lines:

```cmake
list(APPEND EXTRA_COMPONENT_DIRS tests $ENV{IDF_PATH}/tools/mocks/driver)
```

and

```cmake
idf_component_register(
  ...
  REQUIRES
  driver
  ...
)
```

But that stopped working during the last couple of weeks. It produces the following CMake error
```cmake
[0/1] Re-running CMake...-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target linux
-- building DRIVER MOCKS (only SPI, I2C and GPIO driver)
-- Project sdkconfig file /home/vinci/Develop/VSCode/hello_world_linux_compatible/sdkconfig
-- building DRIVER MOCKS (only SPI, I2C and GPIO driver)
CMake Error at /home/vinci/esp/esp-idf/tools/cmake/component.cmake:313 (message):
  Include directory
  '/home/vinci/esp/esp-idf/components/driver/include/driver' is not a
  directory.
Call Stack (most recent call first):
  /home/vinci/esp/esp-idf/tools/cmake/component.cmake:481 (__component_add_include_dirs)
  /home/vinci/esp/esp-idf/tools/cmake/component.cmake:558 (idf_component_register)
  /home/vinci/esp/esp-idf/tools/mocks/driver/CMakeLists.txt:15 (idf_component_mock)

-- Configuring incomplete, errors occurred!
See also "/home/vinci/Develop/VSCode/hello_world_linux_compatible/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
```

Probably the folder structure of the component has changed but no one thought to adjust the mock? The driver mock components CMake sets an include_dirs variable like this
```cmake
set(include_dirs
    "${original_driver_dir}/include"
    "${original_driver_dir}/include/driver"
    "${CMAKE_CURRENT_SOURCE_DIR}/../hal/include"
    "${CMAKE_CURRENT_SOURCE_DIR}/../esp_hw_support/include")
```

But the "/include/driver" folder doesn't seem to exist at the current master?

ESP-Marius, 2023-02-13:
You are right, someone recently reorganized the driver and broke this. We'll take a look at it.

Meanwhile, another option when building for the linux target is to make requirements dynamic depending on the target, e.g. something like:

```
idf_build_get_property(target IDF_TARGET)

if(${target} STREQUAL "linux")
    set(REQS "")
else()
    list(APPEND REQS driver)
endif()

idf_component_register(...
                  REQUIRES ${REQS})
```
Of course, this may require some changes to your component as well if you are including/calling any of the driver functions.

closed 2023-02-22 by espressif-bot

Silur, 2023-07-27:
`list(APPEND EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/tools/mocks/driver)` still doesnt work on linux simulation even after  https://github.com/espressif/esp-idf/commit/b68b0ea714e664f39801121f2786f447e2252e9c 
cloning a fresh copy and building the hello-world with the preview target results in the same error

higaski, 2023-07-27:
Works for me with `ESP-IDF v5.2-dev-1890-g28167ea5a3`

Silur, 2023-07-27:
`idf.py --preview set-target linux`
halts with:

```
Cannot find source file:

    <redacted>/esp-idf/components/esp_mm/port/linux/ext_mem_layout.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm
  .ccm .cxxm .c++m .h .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90
  .f95 .f03 .hip .ispc
Call Stack (most recent call first):
```

AFAIK there is no esp_mm mock?