espressif/esp-idf #10752: Unnecessary assertion in esp_blufi.c ?

someburner, 2023-02-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom

### Power Supply used.

External 5V

### What is the expected behavior?

Not crash when a client disconnects prematurely.

### What is the actual behavior?

Assertion is triggered when `ble_gattc_notify_custom` returns an error.

### Steps to reproduce.

1. Send high rate of data over blufi (BLE) to some connected device

2. Disconnect the device- e.g. turn off bluetooth or send a disconnect.

3. crash

### Debug Logs.

```plain
I (136976) BLUFI_EXAMPLE: subscribe event; conn_handle=0 attr_handle=14 reason=1 prevn=0 curn=1 previ=0 curi=0

I (136993) blufi: testing 123
I (137243) blufi: testing 123
I (137456) blufi: Recv Custom Data 2
blufi logging: about to switch

assert failed: esp_blufi_send_notify esp_blufi.c:437 (rc == 0)

Backtrace: 0x4008246a:0x3ffe1ce0 0x400935f5:0x3ffe1d00 0x4009cbca:0x3ffe1d20 0x400f3843:0x3ffe1e40 0x400ecdb5:0x3ffe1e60 0x400f3896:0x3ffe1e90 0x400ecfd3:0x3ffe1eb0 0x400eda2a:0x3ffe1f10 0x400eccad:0x3ffe1f50 0x400edeef:0x3ffe1f70 0x40096a61:0x3ffe1fa0
0x4008246a: panic_abort at /COMPONENT_ESP_SYSTEM_DIR/panic.c:412

0x400935f5: esp_system_abort at /COMPONENT_ESP_SYSTEM_DIR/esp_system.c:135

0x4009cbca: __assert_func at /COMPONENT_NEWLIB_DIR/assert.c:78

0x400f3843: esp_blufi_send_notify at /COMPONENT_BT_DIR/common/btc/profile/esp/blufi/nimble_host/esp_blufi.c:437 (discriminator 1)

0x400ecdb5: btc_blufi_send_notify at /COMPONENT_BT_DIR/common/btc/profile/esp/blufi/blufi_prf.c:82

0x400f3896: esp_blufi_send_encap at /COMPONENT_BT_DIR/common/btc/profile/esp/blufi/nimble_host/esp_blufi.c:456

0x400ecfd3: btc_blufi_send_encap at /COMPONENT_BT_DIR/common/btc/profile/esp/blufi/blufi_prf.c:245

0x400eda2a: btc_blufi_send_custom_data at /COMPONENT_BT_DIR/common/btc/profile/esp/blufi/blufi_prf.c:405
 (inlined by) btc_blufi_call_handler at /COMPONENT_BT_DIR/common/btc/profile/esp/blufi/blufi_prf.c:814

0x400eccad: btc_thread_handler at /COMPONENT_BT_DIR/common/btc/core/btc_task.c:204

0x400edeef: osi_thread_run at /COMPONENT_BT_DIR/common/osi/thread.c:165

0x40096a61: vPortTaskWrapper at /COMPONENT_FREERTOS_DIR/FreeRTOS-Kernel/portable/xtensa/port.c:151
```
```

### More Information.

I'm extending blufi- using the custom data field to send various things to/from a phone or laptop. In this case I'm forwarding `ESP_LOGx` logs to the connected central device. It seems due to the way events are propagated through blufi / ble stack (nimble in this case), it's possible for one to send data before getting the `ESP_BLUFI_EVENT_BLE_DISCONNECT` event.

This is the assertion that is triggered:

https://github.com/espressif/esp-idf/blob/bb25d6abd3398d0f06fb09fdc9935b791f9344ff/components/bt/common/btc/profile/esp/blufi/nimble_host/esp_blufi.c#L436

Commenting that out and things work fine. I'd prefer to keep assertions enabled, but it doesn't seem like this should be a fatal assertion- rather and error should be logged or an error event should be sent so the application knows something bad happened.

someburner, 2023-02-12:
In fact- even with fairly low send frequency the assert can be triggered. This could probably result in crashes even with normal blufi provisioning operations like getting the wifi scan list.

someburner, 2023-02-14:
Looks like it's just from exhausted mbuf pool from nimble. It's easy to fix by doing a bit of rate limiting in application code, but there should probably be a check for a failed buffer alloc. e.g.

```c
void esp_blufi_send_notify(void *arg)
{
    struct pkt_info *pkts = (struct pkt_info *) arg;
    struct os_mbuf *om;
    om = ble_hs_mbuf_from_flat(pkts->pkt, pkts->pkt_len);
    if (!om) {
        ESP_LOGE(TAG, "mbuf_from_flat failed");
        return;
    }
    int rc = 0;
    rc = ble_gattc_notify_custom(conn_handle, gatt_values[1].val_handle, om);
    assert(rc == 0);
}
```

IshaESP, 2023-02-20:
Hi @someburner 
Can you please try out this [patch](https://github.com/espressif/esp-idf/files/10779965/IDFGH-10752.zip) ?

Thanks.

AxelLin, 2023-04-21:
@IshaESP 
This issue has been marked as "Resolution: Done" for 2 Months. Is this fixed? Which commit fix it?

rahult-github, 2023-04-21:
Fixed via 14aa55b3531f0a1b419c38ca1c74617d1bad1c65 .

closed 2023-04-21 by rahult-github