espressif/esp-idf #10753: Check if factory reset occurs in app?

wuyuanyi135, 2023-02-12:
### Is your feature request related to a problem?

I need to perform some extra initialization after factory reset. I want to check if factory reset occurs in the app. 

### Describe the solution you'd like.

Provide an API to tell if this is the first boot-up after factory reset.

### Describe alternatives you've considered.

I have considered using a dummy nvs partition that will be set after boot. On factory reset this partition will be erased, so that the app will know this is the first boot after factory reset. However, this solution seems a bit tedious.

igrr, 2023-02-12:
> I have considered using a dummy nvs partition that will be set after boot. On factory reset this partition will be erased, so that the app will know this is the first boot after factory reset. However, this solution seems a bit tedious.

Unfortunately I don't think there's anything significantly simpler in terms of the underlying implementation. You could replace an NVS partition with a single-sector data partition, and write data into it as a token of the "first boot". But if you are using NVS for any other purpose, it's probably easier to reuse that partition and store the "first boot" token there.

wuyuanyi135, 2023-02-12:
@igrr Thanks for confirming!

closed 2023-02-12 by wuyuanyi135

chipweinberger, 2023-02-13:
I personally use a boot count.

I increment the nvs boot count setting on each boot.

If zero, i know it's the first boot after reset.

wuyuanyi135, 2023-02-13:
> nt.
> 
> I increment the nvs boot count setting on each boot.
> 
> If zero, i know it's the first boo

Thanks for your suggestion!