espressif/esp-idf #10754: mdns: mdns_hostname_set() should not use task notifications (xTaskNotifyWait)

boborjan2, 2023-02-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.x-5.x

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

Mdns shall not use the task notification bits of the calling task. If the calling task use the task notification bits, it is potential source of malfunction. (it was indeed in my case)

### What is the actual behavior?

mdns_hostname_set() (and _mdns_execute_action) use the global task notification bits of the task that calls mdns_hostname_set(). This causes potential issues with tasks that make use of the task notification bits themselves. The application writer cannot be aware of this internal behavior of mdns.c. The internal inter-task communication of mdns shall not interfere with the user tasks of the application. Mdns should use another forms of ipc, e.g. a semaphore.

### Steps to reproduce.

You need a task that makes use of BIT0 of the task notification bits and call mdns_hostname_set().

0xFEEDC0DE64, 2023-02-13:
They moved mdns stuff to a different repo: https://github.com/espressif/esp-protocols

Maybe create the issue there?

boborjan2, 2023-02-13:
I can create one there as well. The issue is also present in 4.4 where mdns is not separated yet. I don't know if fixes from the separate repo are backported to 4.4.

david-cermak, 2023-02-13:
Thanks for the report, will fix!
No need to raise another issues in the protocols repo, I'll fix it there first and then backport to IDF `v4.4`.

PS: PRs welcome :-)

boborjan2, 2023-02-13:
I will provide you will a pr in the protocols repo then.

david-cermak, 2023-06-09:
Backported as 0050fa940c7d719a5f7f182e7b51bfdfb2a49fad

closed 2023-06-09 by david-cermak