espressif/esp-idf #10755: Scheduler not starting

ClaesIvarsson, 2023-02-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-319-g26960f1734

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Program should start executing

### What is the actual behavior?

Program execution is blocked after:
```
I (105) boot: Defaulting to factory image
I (107) esp_image: segment 0: paddr=00400020 vaddr=3f400020 size=aec48h (715848) map
I (327) esp_image: segment 1: paddr=004aec70 vaddr=3ffbdb60 size=013a8h (  5032) load
I (329) esp_image: segment 2: paddr=004b0020 vaddr=400d0020 size=241fc0h (2367424) map
I (1045) esp_image: segment 3: paddr=006f1fe8 vaddr=3ffbef08 size=05808h ( 22536) load
I (1053) esp_image: segment 4: paddr=006f77f8 vaddr=40080000 size=1d504h (120068) load
I (1110) boot: Loaded app from partition at offset 0x400000
I (1110) boot: Disabling RNG early entropy source...
I (1116) psram: This chip is ESP32-D0WD
I (1117) spiram: Found 64MBit SPI RAM device
I (1117) spiram: SPI RAM mode: flash 80m sram 80m
I (1119) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (1123) cpu_start: Pro cpu up.
I (1125) cpu_start: Starting app cpu, entry point is 0x40081760
I (0) cpu_start: App cpu up.
I (1671) spiram: SPI SRAM memory test OK
I (1680) cpu_start: Pro cpu start user code
I (1680) cpu_start: cpu freq: 240000000
I (1680) cpu_start: Application information:
I (1681) cpu_start: Project name:     ElsIotModule
I (1684) cpu_start: App version:      1.1.1-beta-14-g5541c492
I (1687) cpu_start: Compile time:     Feb 10 2023 14:36:51
I (1690) cpu_start: ELF file SHA256:  5adf6b06c3673f43...
I (1693) cpu_start: ESP-IDF:          v4.4.3-319-g26960f1734
I (1697) heap_init: Initializing. RAM available for dynamic allocation:
I (1700) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1703) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1706) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1709) heap_init: At 3FFD0710 len 0000F8F0 (62 KiB): DRAM
I (1713) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1716) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1719) heap_init: At 4009D504 len 00002AFC (10 KiB): IRAM
I (1722) spiram: Adding pool of 4083K of external SPI memory to heap allocator
I (1727) spi_flash: detected chip: generic
I (1729) spi_flash: flash io: dio
I (1756) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

### Steps to reproduce.

I updated my stable project from version v4.4.3-1-gcc5640aef3 to v4.4.3-319-g26960f1734 and then this happens
Changing back to commit v4.4.3-319-g7dde97d2d6 restores my project to stable again.

I have not been able to find a newer commit on branch release/v4.4 which makes execution working again. I have tried using v4.4.4.

### More Information.

- efuses is used in the project!
- ESP32-WROVER

Dazza0, 2023-02-13:
@ClaesIvarsson Could you provide more details such as...

- Your project configuration (`sdkconfig.h`)
- What efuses have been set and their values (suggest running `espefuse.py summary`)

ClaesIvarsson, 2023-02-13:
> @ClaesIvarsson Could you provide more details such as...
> 
> * Your project configuration (`sdkconfig.h`)
> * What efuses have been set and their values (suggest running `espefuse.py summary`)

[sdkconfig_and_efuse_summary.zip](https://github.com/espressif/esp-idf/files/10721959/sdkconfig_and_efuse_summary.zip)

AxelLin, 2023-02-13:
@ClaesIvarsson 
You can run 'git bisect' to quickly find the first bad commit.

ClaesIvarsson, 2023-02-13:
> @ClaesIvarsson You can run 'git bisect' to quickly find the first bad commit.

Thanks, that is exactly what I did, and then I found that 26960f17346ca8230171148e087fc1b38aab8bbd is breaking my build. 
I as not able to find any configuration parameter that I have changed related to the changes in this commit. Also, there is very little help from the log output to help guide me in the right direction

KonstantinKondrashov, 2023-02-14:
@ClaesIvarsson 
Hi! I am not sure why you get this issue. I run some examples on this version (v4.4.3-319-g26960f1734) and I do not see such an issue. If you suspect that the issue is in 26960f17346ca8230171148e087fc1b38aab8bbd then I can suggest to try this patch.   

```diff
diff --git a/components/hal/efuse_hal.c b/components/hal/efuse_hal.c
index 061d1be..69f909f 100644
--- a/components/hal/efuse_hal.c
+++ b/components/hal/efuse_hal.c
@@ -10,9 +10,9 @@
 #include "hal/efuse_ll.h"
 #include "hal/assert.h"
 #include "hal/efuse_hal.h"
+#include "esp_attr.h"
 
-
-uint32_t efuse_hal_chip_revision(void)
+IRAM_ATTR uint32_t efuse_hal_chip_revision(void)
 {
     return efuse_hal_get_major_chip_version() * 100 + efuse_hal_get_minor_chip_version();
 }
diff --git a/components/hal/esp32/efuse_hal.c b/components/hal/esp32/efuse_hal.c
index f0cfbfe..8cbae52 100644
--- a/components/hal/esp32/efuse_hal.c
+++ b/components/hal/esp32/efuse_hal.c
@@ -11,8 +11,9 @@
 #include "hal/assert.h"
 #include "hal/efuse_hal.h"
 #include "soc/syscon_reg.h"
+#include "esp_attr.h"
 
-uint32_t efuse_hal_get_major_chip_version(void)
+IRAM_ATTR uint32_t efuse_hal_get_major_chip_version(void)
 {
     uint8_t eco_bit0 = efuse_ll_get_chip_ver_rev1();
     uint8_t eco_bit1 = efuse_ll_get_chip_ver_rev2();
@@ -44,7 +45,7 @@ uint32_t efuse_hal_get_major_chip_version(void)
     return chip_ver;
 }
 
-uint32_t efuse_hal_get_minor_chip_version(void)
+IRAM_ATTR uint32_t efuse_hal_get_minor_chip_version(void)
 {
     return efuse_ll_get_chip_wafer_version_minor();
 }
```

ClaesIvarsson, 2023-02-15:
Tried this patch without success.

KonstantinKondrashov, 2023-02-15:
@ClaesIvarsson Could you try to enable debug log? It could help to reveal where exactly it is stacked.

ClaesIvarsson, 2023-02-16:
[boot.log](https://github.com/espressif/esp-idf/files/10754546/boot.log)
boot with verbose as default log level (CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y)

I also changed to verbose level for my stable branch (using esp-idf v4.4.3, 6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7).
This resulted in the same behavior. Software freezes just after starting scheduler. I guess commit 26960f17346ca8230171148e087fc1b38aab8bbd is not the root cause of this problem. Maybe some timing is changing causing this problem.

AxelLin, 2023-02-20:
> I also changed to verbose level for my stable branch (using esp-idf v4.4.3, [6407ecb](https://github.com/espressif/esp-idf/commit/6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7)). This resulted in the same behavior. Software freezes just after starting scheduler.

This is different from your original report that v4.4.3-1-gcc5640aef3 is good.
Maybe you should try to find an older version that does not has this issue first. e.g. 4.4.2 or 4.4.1.
Then run git bisect again between the good commit and v4.4.3 (bad).

ESP-Marius, 2023-03-06:
@ClaesIvarsson I think you are probably right that this is more likely due to some timings changing rather than a IDF commit breaking something.

What is interesting though is that the WDT is not triggering, which it should if you are stuck at that point. Some suggestions when you get back to investigating this:
1. If possible connect a JTAG and see what the CPU is doing when you are stuck
2. Check if the [main task](https://github.com/espressif/esp-idf/blob/4c2afac35502161ce473660c99e17feb013a378d/components/freertos/port/port_common.c#L94) starts running (e.g. by adding a print there)

ClaesIvarsson, 2023-03-17:
I connected the debugger and dumped the registers to file when execution is stuck.
[regs.txt](https://github.com/espressif/esp-idf/files/10999415/regs.txt)

I enabled full log (CONFIG_LOG_DEFAULT_LEVEL=5) using (esp-idf 4.4.3). Pls note that everything works just fine if CONFIG_LOG_DEFAULT_LEVEL=3

```
V (2450) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (2453) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xC02
D (2458) intr_alloc: Connected src 17 to int 2 (cpu 0)
D (2460) cpu_start: Setting C++ exception workarounds.
D (2463) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (2466) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
D (2470) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (2473) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
V (2477) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (2480) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (2484) intr_alloc: Connected src 24 to int 3 (cpu 0)
I (2487) cpu_start: Starting scheduler on PRO CPU.
D (0) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (2493) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
V (2496) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (2500) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (D (2504) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (2506) cpu_start: Starting scheduler on APP CPU.
2490) partition: Loading the partition table
V (2518) calculated md5: 0x3ffbb398   0c 21 a4 b8 0f e8 3c 44  e5 c0 87 60 d4 ac ae e8  |.!....<D...`....|
V (2518) stored md5: 0x3f4c81f0   0c 21 a4 b8 0f e8 3c 44  e5 c0 87 60 d4 ac ae e8  |.!....<D...`....|
D (2522) partition: Partition table MD5 verified
V (2525) spi_flash: allocate temp buffer: 0x3ffd4c44 (32)
V (2528) spi_flash: allocate temp buffer: 0x3ffd4cc8 (32)
```

main function is never reached!

I do not want to dump the full log and place it here, but can provide it directly to developers on request!

ESP-Marius, 2023-03-17:
@ClaesIvarsson the full log would be nice, you can find my email address with `git show dfedaa2b65` :smile: 
`

AxelLin, 2023-04-17:
@ClaesIvarsson @ESP-Marius 
Any update for this issue?

ESP-Marius, 2023-04-17:
The root cause seemed to be some tasks getting spinned up on core 1 before `app_main` ran and starving the idle task (cpu0 bootup process relies on the idle task for cpu1 to run at least once to signal that cpu1 boot up process is complete).

@ClaesIvarsson can we consider the problem solved on your end?

ClaesIvarsson, 2023-04-17:
> The root cause seemed to be some tasks getting spinned up on core 1 before `app_main` ran and starving the idle task (cpu0 bootup process relies on the idle task for cpu1 to run at least once to [...]

Sorry for not following up on this issue sooner (other more urgent tasks has been prioritized)

I was not able to confirm this thesis.
I am using dual core functionality which makes this a possibility.
The only tasks assigned to the second core is the WIFI task (CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1=y) and this is not started until later.
User-created tasks are currently created with tskNO_AFFINITY. 
I will have a go at this problem again asap.

ClaesIvarsson, 2023-04-17:
As I could not find any code that I implement that initialize  a task during startup, I went in another direction.
I added a portYIELD in the main_task where the code waits for idle hook and this changed the behavior to the better.
My conclusion is that the criterias are actually fulfilled to continue, but the task to inform about it is frozen out.

Please advice if this is a sustainable solution.

ESP-Marius, 2023-04-18:
> I added a portYIELD in the main_task where the code waits for idle hook and this changed the behavior to the better.
My conclusion is that the criterias are actually fulfilled to continue, but the task to inform about it is frozen out.

Yeah, the idle task for cpu 1 probably never runs because there is a task running on it that never yields. So when you yield on `main_task` I suspect the task that was running on cpu 1 might end up migrating to cpu 0 (where you yielded) and thus freeing up cpu 1 to run its idle task.

I would still suggest you try to find the task that is starving core 1 (seems to be a task related to reading the flash, using the partition API).

AxelLin, 2023-05-09:
> As I could not find any code that I implement that initialize a task during startup, I went in another direction. I added a portYIELD in the main_task where the code waits for idle hook and this [...]

@ClaesIvarsson 
This looks strange to me especially you mentioned it was working with older esp-idf version.
Could you share the code to reproduce this issue so others can debug this issue?

ClaesIvarsson, 2023-05-10:
> @ClaesIvarsson This looks strange to me especially you mentioned it was working with older esp-idf version. Could you share the code to reproduce this issue so others can debug this issue?

Sorry, sharing the code is not possible.

ClaesIvarsson, 2023-07-13:
After a lot of testing and investigating I finally found what is causing this problem.
I have some leftovers from esp-mdf still in my project and in mupgrade there is a feature of fallback version which creates a task in a function with attribute constructor:
`__attribute((constructor)) mdf_err_t mupgrade_partition_switch()`
When I disabled this feature, my problem with scheduler not starting disappeared.

[esp-mdf/components/mupgrade/mupgrade_check.c#L135](https://github.com/espressif/esp-mdf/blob/cf502740f5a6c82a0dc3059e7591c262795f70b2/components/mupgrade/mupgrade_check.c#L135)

As I was recommended to not start any tasks in code called in constructors or startup code, I think this is not a good design in the esp-mdf

Case closed from my side!

closed 2023-07-13 by ClaesIvarsson