espressif/esp-idf #10756: RMT

Imdad72, 2023-02-13:
### Is your feature request related to a problem?

I am not able to control non nec tpye remote control devices. 

### Describe the solution you'd like.

I should be able to vontrol all non nec devices

suda-morris, 2023-02-13:
Hi @Imdad72 our example provides the standard NEC encoder and decoder: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/ir_nec_transceiver

For other IR protocols, you need to implement the encoder and decoder on your own side, you can refer to [this documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#customize-rmt-encoder-for-nec-protocol) for how to customize an RMT encoder.

Imdad72, 2023-02-13:
Thanks for response,
i am using version 4.3 idf  i want to control non nec devices i am stuck in my project kindly help me out

suda-morris, 2023-02-13:
Then you may need to read your IR protocol spec and rewrite this function: https://github.com/espressif/esp-idf/blob/release/v4.3/examples/peripherals/rmt/ir_protocols/components/infrared_tools/src/ir_builder_rmt_nec.c#L98

suda-morris, 2023-02-14:
I'm afraid we can't prioritize that at the moment. It's more of an application layer stuff. You can google the keyword "IR Panasonic ESP32", I think there will be some example code to inspire you.

Imdad72, 2023-02-14:
i have searched all these things earlier, i am stuck at this panasonic point for a month so i want solution how to get library of all ir remote protocols.

deviousprophet, 2023-02-14:
@Imdad72 I think you can check this Arduino library [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266)
It does support Panasonic various (BUT not every) protocols:  https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Panasonic.cpp

Learn their source code and build the protocol in IDF on your own or just use ArduinoIDE and try that library if you want a faster way to run it

Imdad72, 2023-02-14:
i need espidf library

mulligan252, 2024-06-28:
@Imdad72 Did you manage to solve this ? 

I am also trying to get this working:) I have created an issue here hoping to get some support : https://github.com/espressif/esp-idf/issues/14074  

I am basing my implementation off the https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Panasonic.cpp  .

I have managed to adapt the _ir_nec_transceiver_ example to work with the EliteScreens protocol, but I have hit a brick wall with the Panasonic implementation for some time now. It appears that I am encoding according to the spec, but unfortunately the data on the IR receiving end is not  as expected.