espressif/esp-idf #10757: ESP32 (ESP-IDF) "driver/twai.h" CAN Data Transmission is working but Receive function is not working using CAN/TWAI

Kunaalkk1, 2023-02-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0 and v4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Expected Output:

```
Message received:
ID: 0x123 DLC: 8 Data: 1 2 3 4 5 6 7 8
```

In my source code, I have implemented random data bits, so data may vary

### What is the actual behavior?

Actual Output:

`Failed to receive message`

### Steps to reproduce.

1. I used ESP-IDF v5.0 with VS Code.
2. I used the documentation to obtain the code for sender and receiver.
3. I used a tested and functional MCP2515 module as my transceiver for the same.
4. The Tx is broadcasting perfectly, but the Rx is not working.

### More Information.

I saw a YouTube video with someone who used a similar code, and it was working perfectly for him (YouTube Video: link: https://youtu.be/bxzWuIqfn9Y). He used "driver/can.h" which has been removed from v5.0 of ESP-IDF because CAN and TWAI are the same.

This is my code:

```
#include "driver/gpio.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "stdio.h"
#include "stdlib.h"
#include "driver/twai.h"

void twai_setup_and_install(){
    //Initialize configuration structures using macro initializers
    twai_general_config_t g_config = {
        .mode = TWAI_MODE_NORMAL,
        .tx_io = GPIO_NUM_5,
        .rx_io = GPIO_NUM_4,
        .clkout_io = TWAI_IO_UNUSED,
        .bus_off_io = TWAI_IO_UNUSED,
        .tx_queue_len = 5,
        .rx_queue_len = 5,
        .alerts_enabled = TWAI_ALERT_NONE,
        .clkout_divider = 0
    };

    twai_timing_config_t t_config = TWAI_TIMING_CONFIG_500KBITS();
    twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();

    // Install TWAI driver
    if (twai_driver_install(&g_config, &t_config, &f_config) == ESP_OK) {
        printf("Driver installed\n");
    } else {
        printf("Failed to install driver\n");
        return;
    }

    // Start TWAI driver
    if (twai_start() == ESP_OK) {
        printf("Driver started\n");
    } else {
        printf("Failed to start driver\n");
        return;
    }
}

void new_message(twai_message_t *message, uint32_t id, uint8_t dlc, uint8_t *data)
{
    
    message->flags = TWAI_MSG_FLAG_NONE;
    message->identifier = id;
    message->data_length_code = dlc;
    for (int i = 0; i < dlc; i++) {
        message->data[i] = data[i];
    }
    printf("Message created\nID: %ld DLC: %d Data:\t", message->identifier, message->data_length_code);
    for (int i = 0; i < message->data_length_code; i++) {
        printf("%d\t", message->data[i]);
    }
    printf("\n");
}

void transmit_message(twai_message_t *message)
{
    if (twai_transmit(message, pdMS_TO_TICKS(1000)) == ESP_OK) {
        printf("Message queued for transmission\n");
    } else {
        printf("Failed to send message\n");
    }
}

void receive_message(twai_message_t *message)
{
    if (twai_receive(message, pdMS_TO_TICKS(1000)) == ESP_OK) {
        printf("Message received:\n");
        printf("ID: %ld DLC: %d Data:\t", message->identifier, message->data_length_code);
        for (int i = 0; i < message->data_length_code; i++) {
            (message->extd)?printf("Extended ID"):printf("Standard ID");
            printf("%d\t", message->data[i]);
        }
    } else {
        printf("Failed to receive message\n");
    }
}

void app_main()
{
    
    twai_setup_and_install();
    twai_message_t message;
    twai_message_t message1;
    // Set the data to send
    uint8_t data[8] = {rand() % 255, rand() % 255, rand() % 255, 
    rand() % 255, rand() % 255, rand() % 255, rand() % 255, rand() % 255};

    while(true){ 
    // Create a new message
    new_message(&message, 0x123, 8, data);

    // Transmit the message to a queue
    transmit_message(&message);

    // Receive the message from the queue
    receive_message(&message1);

    // Wait for 1 second
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    }

}
```

If anyone can help me as soon as possible, it would be really appreciated.

Sincere Regards.

Dazza0, 2023-02-13:
@Kunaalkk1 

I see that `twai_receive(message, pdMS_TO_TICKS(1000))`, which means the function will return an error if it has not received a message within 1 second.

Could you check...

- What `esp_err_t` value is being returned by `twai_receive()`
- You have another node on the bus actually sending messages at least once per second.

Kunaalkk1, 2023-02-14:
@Dazza0 

I believe that `esp_err_t` returns `ESP_OK` on successful transmission and 'ESP_ERR_TIMEOUT' if it does not retrieve any data from the Queue.

Further, I have used two nodes, one for transmitting the message, and other for receiving the message, they are both given 1 second. In fact, I even tried giving receive function 10 seconds, and sender function had been given 1 second, but still, it didn't seem to work, so the timeout or the two nodes are not the issue here.

In fact, I tried a different driver and source code for CAN available on the internet with Arduino IDE to test my hardware, and the code seemed to work perfectly, and the driver code was almost similar to the original ESP-IDF driver.

The transmitter code enqueues the messages as per the queue length, but since the receiver does not dequeue any message, the transmitter stops transmitting after 5 or 6 messages (Since I have set the `tx_queue_len` and `rx_queue_len` to 5 each).

Following is the receive function in the file twai.c of the ESP-IDF twai driver:

```
esp_err_t twai_receive(twai_message_t *message, TickType_t ticks_to_wait)
{
    //Check arguments and state
    TWAI_CHECK(p_twai_obj != NULL, ESP_ERR_INVALID_STATE);
    TWAI_CHECK(message != NULL, ESP_ERR_INVALID_ARG);

    //Get frame from RX Queue or RX Buffer
    twai_hal_frame_t rx_frame;
    if (xQueueReceive(p_twai_obj->rx_queue, &rx_frame, ticks_to_wait) != pdTRUE) {
        return ESP_ERR_TIMEOUT;
    }

    TWAI_ENTER_CRITICAL();
    p_twai_obj->rx_msg_count--;
    TWAI_EXIT_CRITICAL();

    //Decode frame
    twai_hal_parse_frame(&rx_frame, message);
    return ESP_OK;
}
```

Thanks in advance

Best Regards.

Kunaalkk1, 2023-02-14:
I did some debugging in the code, used a multimeter to test my hardware and all...

I also shifted to TJA1051 Transceiver (Datasheet: https://www.nxp.com/docs/en/data-sheet/TJA1051.pdf)

Turns out, the transmission function is also not working. I used a multimeter, so turns out the data is not being sent to the transceiver from the ESP32 itself.

Please help me with this.

@Dazza0

Dazza0, 2023-02-14:
@Kunaalkk1 Could you check the following things:

- Which ESP32 dev kit you are using? Which GPIOs are you using for TX and RX (is it GPIO4 and GPIO5 as illustrated in the provided example)?
    - I'm asking this because GPIO4 on the ESP32 is used as the `HSPIHD` signal
    - Could you try another pair of GPIOs (e.g., 21 and 22)?
- Have you tried running the `twai_self_test` example in ESP-IDF? This is a simple loopback example to test if the transceiver is connected properly and shouldn't require any other node on the bus.
- Could you enable all of the TWAI alerts and see what alerts (if any) are triggered by the TWAI driver? Is the TWAI driver entering error passive or bus off?

Kunaalkk1, 2023-02-14:
@Dazza0 I am using ESP32 Dev Module (ESP-WROOM-32), I have tried with GPIO 4 and 5, and as you suggested, I tried using GPIO 21 and 22 as well. Neither of the two pairs seem to be working for me.

I have enabled TWAI alerts by changing it in the header file from `TWAI_ALERTS_NONE` to `TWAI_ALERTS_ALL`, but as you can see in the screenshot, none of the alerts are being logged. I am currently using v4.4.4 of ESP-IDF, please let me know if I should change that.

I tried the loopback, and it doesn't go beyond "Driver Started", no matter how long I wait. You can see its output here:

[image]

Awaiting your response
Regards

Dazza0, 2023-02-14:
@Kunaalkk1 With the `twai_self_test` example, if you skip directly connect GPIO 21 to 22, does the example still not work?

Kunaalkk1, 2023-02-14:
Actually, I figured that my sensor was not working. I received a new TJA1051 sensor, and I tested the `twai_self_test`, and it works on GPIO5 (Tx) and GPIO4 (Rx). Turns out it wasn't the firmware or driver that was not functioning, it was the sensor that was faulty.

Nevertheless, @Dazza0, I am not closing this conversation here, as I anticipate that I might face other related issues in the next few days, hope you will help!

Thanks for engaging with this query and helping me out with debugging options so far!

Best regards.

Dazza0, 2023-02-15:
@Kunaalkk1 I'd prefer if this issue was closed as this is not a bug with the TWAI driver or IDF. If you find a confirmed bug with the TWAI driver, please feel free to open another Github issue. FOr discussions/Q&A, we also have the [esp32 forum](https://esp32.com/).

closed 2023-02-15 by Dazza0

Kunaalkk1, 2023-02-15:
@Dazza0 sure, but I hope you won't mind if I reach out to you if I have any queries?

Thanks