espressif/esp-idf #10758: [nvs_flash] NVSEncryptedPartition why not use [mbedtls](https://github.com/espressif/esp-idf/tree/master/components/mbedtls) for xts encryption/decryption?

rukai0712, 2023-02-13:
### Describe the solution you'd like.

The NVSEncryptedPartition is still using https://github.com/espressif/esp-idf/blob/9092b9396605908d6c3f4bfc24c7a56abebdb141/components/nvs_flash/src/nvs_encrypted_partition.cpp#L20-L21
Why not use [mbedtls](https://github.com/espressif/esp-idf/tree/master/components/mbedtls) component that supports hardware encryption and decryption?

igrr, 2023-02-13:
Could you please clarify the question? As the lines you have linked show, NVS _does_ use mbedtls for AES-XTS.

rukai0712, 2023-02-14:
In the [mbedtls](https://github.com/espressif/esp-idf/tree/master/components/mbedtls) component there are `esp_aes_*` functions which seems use the hardware for encryption and decryption, like this
https://github.com/espressif/esp-idf/blob/9092b9396605908d6c3f4bfc24c7a56abebdb141/components/mbedtls/port/include/aes/esp_aes.h#L345-L346

In the NVSEncryptedPartition implementation, it uses `mbedtls_aes_xts_*` functions. Does those functions leveraged on hardware?

ESP-Marius, 2023-02-14:
> Does those functions leveraged on hardware?

@rukai0712 Yes, if `CONFIG_MBEDTLS_HARDWARE_AES` is enabled (it is by default) then the mbedtls functions will call the HW accelerated crypto functions.

rukai0712, 2023-02-14:
Thanks for your clarification.

> > Does those functions leveraged on hardware?
> 
> @rukai0712 Yes, if `CONFIG_MBEDTLS_HARDWARE_AES` is enabled (it is by default) then the mbedtls functions will call the HW accelerated crypto functions.

Thanks for the clarification.

closed 2023-02-14 by rukai0712