espressif/esp-idf #10761: openthread API secondary encapsulation

HeFeng1947, 2023-02-14:
### Is your feature request related to a problem?

The native openthread API is too complex and involves ipv6-related network Settings, making it too difficult for the average developer. 

### Describe the solution you'd like.

I've seen ESP encapsulate parts of the API, but it's too small for secondary development at the moment. My last project use zephyr OS and their APIs are easy for developers to get started. I think you can learn from their API encapsulation. https://github.com/zephyrproject-rtos/zephyr/tree/main/subsys/net/l2/openthread

chshu, 2023-02-15:
@HeFeng1947 The ESP OpenThread APIs are provided here: https://github.com/espressif/esp-idf/tree/master/components/openthread

Just note that the ESP Openthread is based on LwIP, zephyr OS uses its own IP stack. The IP interface in our SDK is provided via LwIP APIs, which all are commonly used socket APIs. You may refer to the UDP example here: https://github.com/espressif/esp-thread-br/blob/main/components/esp_ot_cli_extension/src/esp_ot_udp_socket.c

Let us know if you have any detailed feature requests, we are more than happy to provide and accelerate your development.

HeFeng1947, 2023-02-15:
> @HeFeng1947 The ESP OpenThread APIs are provided here: https://github.com/espressif/esp-idf/tree/master/components/openthread
> 
> Just note that the ESP Openthread SDK is based on LwIP, zephyr OS uses its own IP stack. The IP interface in our SDK is provided via LwIP APIs, which all are commonly used socket APIs. You may refer to the UDP example here: https://github.com/espressif/esp-thread-br/blob/main/components/esp_ot_cli_extension/src/esp_ot_udp_socket.c
> 
> Let us know if you have any detailed feature requests, we are more than happy to provide and accelerate your development.

都是国人那我直接说中文了，需求内容表达更清晰些。
我们现在需要以单H2模块搭建Thread Router的原型机，需要OT Joiner、SRP Client、COAP Server和基于Thread的OTA升级四个主要功能。目前贵司封装的接口我大致review了一下（还没有时间全部细看，而且仓库太分散了，如有遗漏请包涵），好像没看到封装最简单的ot启动网络（包括各种ip栈初始化，参考zephyr的**int openthread_start(struct openthread_context *ot_context)**，会根据设置开启各种joiner、commissioner、FTD\MTD初始化）。
这一块基础通用的APIs，无论是TBR、TR或者TED都是需要的，现在ESP有提供吗？

chshu, 2023-02-15:
ot 启动网络主要是下面两个 API：

- [esp_openthread_init()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_openthread.html#_CPPv419esp_openthread_initPK32esp_openthread_platform_config_t)： ot instance 初始化
- [esp_openthread_launch_mainloop()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_openthread.html#_CPPv430esp_openthread_launch_mainloop)：开始 ot mainloop

可以参考这两个示例：
ot_cli: https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli
matter over thread: https://github.com/espressif/esp-matter/tree/main/components/esp_matter_openthread 

OT Joiner, SRP Client 等功能可以通过 menuconfig 来配置 (在 ot_cli 目录下输入 _idf.py menuconfig_).

zwx1995esp, 2023-02-15:
`好像没看到封装最简单的ot启动网络`
一个简单的实现形式可以参考:
https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_cli/main/esp_ot_cli.c#L57-L98
历程里面的函数在 `components/openthread/port` 下面都有具体的实现。
如果想要配置相关的网络参数可以参考 openthread src code里面的 cli 的readme，或者代码里面也可以直接配置，
参考example:
https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_br/main/esp_ot_br.c#L95-L199

HeFeng1947, 2023-02-16:
@chshu @zwx1995esp 
你们在port目录下实现的一些接口，目前可能对用户还不太友好。比如**esp_openthread_netif_glue.c**里面的**esp_openthread_netif_glue_init()**，里面使用了**otSetStateChangedCallback()**，注册了ot state相关的回调用来触发IP层相关的事件，但是ot state里面有非常多种的状态，比如我需要注册ot role相关（OT_CHANGED_THREAD_ROLE）的回调，这时候我就不得不在你们的**esp_openthread_netif_glue_state_callback()** 里面修改。这部分是建议你们再细封装一层回调让用户也可以直接获取，请参考zephyr的**openthread_set_state_changed_cb()**
目前我还在熟悉esp的ot相关组件，实现一些我们的需求。下周我再反馈其他部分。

chshu, 2023-02-17:
可以通过 `otSetStateChangedCallback()` API 注册应用层的 callback，不需要改动`esp_openthread_netif_glue.c` 中的代码。`otSetStateChangedCallback()` 可以注册多个 callback，每个 callback 可以处理各自感兴趣的事件。

HeFeng1947, 2023-02-21:
S3+H2的BR方案里，S3对H2 RCP的升级，是否需要绑定H2的UART0？能否用H2的UART1连接呢？ @chshu

chshu, 2023-02-21:
> S3+H2的BR方案里，S3对H2 RCP的升级，是否需要绑定H2的UART0？能否用H2的UART1连接呢？ @chshu

可以在 ot_rcp example 中的 ESP_OPENTHREAD_DEFAULT_HOST_CONFIG 来配置 uart port 和 GPIO pin 脚:
https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_rcp/main/esp_ot_config.h#L35

HeFeng1947, 2023-02-21:
> > S3+H2的BR方案里，S3对H2 RCP的升级，是否需要绑定H2的UART0？能否用H2的UART1连接呢？ @chshu
> 
> 可以在 ot_rcp example 中的 ESP_OPENTHREAD_DEFAULT_HOST_CONFIG 来配置 uart port 和 GPIO pin 脚: https://github.com/espressif/esp-idf/blob/master/examples/openthread/ot_rcp/main/esp_ot_config.h#L35

是的，我们已经切换UART调通了功能，现在是对RCP升级这里不太确定是不是绑定了U0

chshu, 2023-02-21:
UART0 和 UART1 都是可以配置的，如果功能没问题了，说明已经配置成功了。

如果暂时没有其他问题，帮忙关一下 issue。有后续问题，欢迎再开新 issue。

closed 2023-02-22 by HeFeng1947

chshu, 2023-02-23:
@HeFeng1947 更正一下：针对 RCP update 功能，是绑定 H2 Uart0 的，因为这个功能是 hardcode 在 H2 ROM 里。但是针对 ot_rcp，uart0/uart1 都可以用于 Host - RCP 通信。

liujunling0, 2023-02-26:
咨询个小白问题，用一块板子是不是没法通过 uart0 来调试 threadrouter
https://openthread.io/guides/border-router/docker/run 我想用这个方案做个测试，但是发现调不通

xieqinan, 2023-02-28:
首先，如果用USB直接连ESP32C6的uart接口的话，ESP23C6应该是可以支持这个测试的(本地linux运行良好)。所以是可以通过烧录口的uart0来作为spinel的通信口的。其次，通过你提供的LOG来看，似乎是MAC OS的USB serial的驱动不支持。你可以尝试更新一下USB 驱动，ESP3232C6默认的RCP波特率是115200。最后，你也可以先开一个端口检查一下ESP32C6上的RCP程序是否正常运行。

liujunling0, 2023-02-28:
您好，您的邮件已收到，我会尽快答复，谢谢!