espressif/esp-idf #10762: [ESP32S3] v4.4 DHCP Server failing to reuse port

KonssnoK, 2023-02-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-116-g00bb43ff24

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3-WROOM N8R2

### Power Supply used.

USB

### What is the expected behavior?

DHCP Server to work correctly upon multiple switch on/off.

This scenario happens in a mesh environment where a node can become ROOT and has to provide IP address multiple times to other nodes

### What is the actual behavior?

DHCP Server will fail on udp_bind because port 67 was already used.

### Steps to reproduce.

1. Start DHCPS
2. Stop
3. Start DHCPS
...

KonssnoK, 2023-02-14:
```
    ip_set_option(pcb_dhcps, SOF_REUSEADDR);

    err_t err = udp_bind(pcb_dhcps, &netif->ip_addr, DHCPS_SERVER_PORT);
#if DHCPS_DEBUG
    if(err){
        DHCPS_LOG("dhcps:dhcps_start->udp_bind error %X\n", err);
    }
#endif
```

Proposed change to DHCPs code

KonssnoK, 2023-02-14:
Reference commit with fix:
1f027d364a0091477f81157360493fdddef946f7

xueyunfei998, 2023-02-15:
hi  @KonssnoK 

I saw your reappearance steps.
1 Start DHCPS
2 Stop
3 Start DHCPS

You should not execute `dhcps_stop` when you stop.

KonssnoK, 2023-02-15:
hi @xueyunfei998 ,
what do you mean?
how do i stop using the dhcps if i don't stop it?
Consider that the device will become a mesh node, which does not require dhcps, but only dhcp.

xueyunfei998, 2023-02-15:
hi @KonssnoK 

`D (406106) lwip: udp_bind: local port 67 already bound by another pcb`

The reason why this log appears is that the `dhcps_start` function is called, and the problem will appear when `dhcps_start` is called without closing `dhcps_stop`.

I have a question. 

Your reproduction steps are as follows.

1 Start DHCPS
2 Stop
3 Start DHCPS

Which function did you stop calling in the second step?

What's wrong now is that the function you called at the time of stop in the second step did not close dhcps.

xueyunfei998, 2023-02-21:
hi @KonssnoK 

There is something wrong with this function `set_dhcps_dns `, and I will fix this bug later.

KonssnoK, 2023-02-21:
@xueyunfei998 ok thanks, keep us updated. Meanwhile, we continue to use the reuseaddr as workaround!

AxelLin, 2023-03-17:
> hi @KonssnoK
> 
> There is something wrong with this function `set_dhcps_dns `, and I will fix this bug later.

@xueyunfei998 Any update?

closed 2023-03-27 by espressif-bot