espressif/esp-idf #10763: use twai ，how to change the receive buffer size.

giu-wxyao, 2023-02-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

use the twai commucation, how to change the receive buffer size.

giu-wxyao, 2023-02-14:
esp_idf-v4.4.   esp32    windows

i have a sensor , twai interface. MCU send a command to the sensor ,and the sensor will feedback 16 group data (128 byte) ,but the MCU only can receive the front 6 groups. and the last 10 group missing. so i think may be the receive buffer is not big enough. and i want to know how to change the buffer size..

troky, 2023-02-15:
You haven't read documentation well.

https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/peripherals/twai.html#_CPPv4N21twai_general_config_t12rx_queue_lenE

Dazza0, 2023-02-16:
@giu-wxyao Please see the response on the related [forum post](https://esp32.com/viewtopic.php?f=25&t=32137)

giu-wxyao, 2023-02-16:
TKS,I found where to change the parameter.

closed 2023-02-16 by giu-wxyao