espressif/esp-idf #10764: Possible bug spotted in mdns.c function calls to _mdns_get_service_item_instance

StevePeake, 2023-02-14:
I've been delving into mdns.c trying to trace a weird core panic unhandled exception. ( It was only caused when I tried to view the mDNS information using the "Service Browser" Android app - it all worked as expected using "mDNS Discovery").  This crash appears to have been resolved by moving from ESP-IDF v4.4.1 to v4.4.4

While fishing through mdns.c, I spotted two suspicious calls to _mdns_get_service_item_instance.

(line numbers refer to v4.4.4)

Definition on line 278:
```
static mdns_srv_item_t *_mdns_get_service_item_instance(const char *instance, const char *service, const char *proto,
                                                        const char *hostname)
```

Two of the calls to this function use the correct parameter order:

line 5056:
`mdns_srv_item_t * item = _mdns_get_service_item_instance(instance, service, proto, hostname);`

line 5125:
`return _mdns_get_service_item_instance(instance, service_type, proto, hostname) != NULL;`

The other two calls use a different parameter order:

line 1495:
`mdns_srv_item_t *service = _mdns_get_service_item_instance(q->host, q->service, q->proto, NULL);`

line 2753:
`service = _mdns_get_service_item_instance(name->host, name->service, name->proto, NULL);`

Is this deliberate? 
(If so, it would benefit from a comment explaining the intended effect of making instance into hostname and setting hostname = NULL)

I don't think this was related to my crash issue, as the oddly-ordered calls remain in v4.4.4

Regards

david-cermak, 2023-02-20:
@StevePeake I believe it was deliberate and commented few lines above:

https://github.com/espressif/esp-idf/blob/4c2afac35502161ce473660c99e17feb013a378d/components/mdns/mdns.c#L272-L273

Do you think it would be more readable if we comment it when calling this function as well?

david-cermak, 2023-03-24:
Hope this answers your question. Please reopen otherwise.

closed 2023-03-24 by david-cermak