espressif/esp-idf #10765: components/esp-tls/esp_tls.h is missing includes

rkit1, 2023-02-14:
When CONFIG_ESP_TLS_SERVER_SESSION_TICKETS is enabled

```
C:/.espressif/frameworks/esp-idf-v5.0/components/esp-tls/esp_tls.h:191:5: error: unknown type name 'mbedtls_entropy_context'
  191 |     mbedtls_entropy_context entropy;                                            /*!< mbedTLS entropy context structure */
      |     ^~~~~~~~~~~~~~~~~~~~~~~
C:/.espressif/frameworks/esp-idf-v5.0/components/esp-tls/esp_tls.h:193:5: error: unknown type name 'mbedtls_ctr_drbg_context'
  193 |     mbedtls_ctr_drbg_context ctr_drbg;                                          /*!< mbedTLS ctr drbg context structure.
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
```

AdityaHPatwardhan, 2023-02-15:
Hi @rkit1,
Thanks for the issue.
I agree that it is a bug.
I think it was introduced when we restructured the esp_tls.h file some
time back.
Can you please check if the patch that I have attached fixes your issue?
Thanks,
Aditya

[server_session_ticket_build_issue.patch.zip](https://github.com/espressif/esp-idf/files/10739454/server_session_ticket_build_issue.patch.zip)

rkit1, 2023-02-26:
Yes it fixed the issue.

closed 2023-02-27 by espressif-bot