espressif/esp-idf #10766: Feature Request: MQTT Broker

GeckoNickDeveloper, 2023-02-14:
### Describe the solution you'd like.

A simple solution like a component that can be configured with simple values, like a limited queue size (eg. max 5), max topic number (eg. 10) and simultaneous connections.  

### Describe alternatives you've considered.

There are other libraries, however I'd like to continue using ESP-IDF for a better integration.

### Additional context.

This would be usefull for projects where you do have multiple ESP32 but not a proper pc for handling the MQTT Broker locally .

This effect can be also achieved by raw sockets, however mqtt is a solid standard and I personally think ESP-IDF should include the Broker capability.

nopnop2002, 2023-02-19:
Try this:

https://github.com/nopnop2002/esp-idf-mqtt-broker

euripedesrocha, 2023-02-21:
Hi @GeckoNickDeveloper, this is an interesting feature. We'll put this in our pipeline of development. 

In the meantime, as you pointed, you can try some available libraries.

sramrajkar, 2023-04-25:
Hi. I am also looking at MQTT broker. Currently in the dev phase I can use the Mongoose to do a PoC. But it is GPL. I would like to have expressif provide this feature. This is kind of urgent for the project time lines

newcomertv, 2023-06-21:
+1 we have faced this issue multiple times that clients want an MQTT broker and we tell them there are no solid options to pick from.

@nopnop2002 great contribution! unfortunately mongoose which your based on is GPL

david-cermak, 2024-09-30:
FYI: There's an effort of porting mosquitto to IDF in https://github.com/espressif/esp-protocols/tree/master/components/mosquitto

It's not published yet, but we're planning on publishing some preliminary version soon, after adding TLS transport option (https://github.com/espressif/esp-protocols/pull/610)

david-cermak, 2024-10-30:
the mosquitto port has been released as [v2.028~0](https://github.com/espressif/esp-protocols/releases/tag/mosq-v2.0.28_0) and published in https://components.espressif.com/components/espressif/mosquitto

please report issues and post feature requests in https://github.com/espressif/esp-protocols/issues

closed 2024-10-30 by david-cermak