espressif/esp-idf #10767: Uploading OTA firmware dynamically

Benedito821, 2023-02-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have successfully runned the _**simple_ota_example**_ found on this repository. But I noticed that I can't change the .bin file that is hosted by HTTPS server without regenerating the certificate and reflashing the board through wire. Is there an example that allows me to flash the board once and henceforth just flash through wi-fi?

hmalpani, 2023-02-17:
Hello @Benedito821 
Could you please elaborate more on the issue? I am not sure what you mean by `regenerating the certificate and reflashing the board through wire`.

hmalpani, 2023-02-17:
If I understand correctly, what you mean is you are generating a new server certificate every time you want to host a binary on a server for OTA. That way you need the firmware to have that certificate with it so you have to build and flash a new binary to the board.
However you need not create new server certificate everytime. You can generate the certificate once and the host the server with that certificate. Then you just have to make sure that the binary has that certificate embedded with it. As the certificate does not change, you won't need to build and flash again and again.

Benedito821, 2023-02-17:
Ok. So where am I supposed to upload my new .bin to? I counted on a web
page where we could upload our files to and after that flash but it seems
not being available on this project.

hmalpani, 2023-02-17:
You need to upload the binary to the same server which was used before. You just have to make sure that the binary which was updated on the first OTA also has the server certificate.

Benedito821, 2023-02-18:
And I should upload it through, for instance, https://192.168.2.106:8070/hello_world.bin ?

hmalpani, 2023-02-20:
Hello @Benedito821 
You have to host the new binary on the same previous server.

Benedito821, 2023-02-20:
Yes, I'm aware of this, thank you ... but how to do it?

hmalpani, 2023-02-21:
You can refer https://github.com/espressif/esp-idf/tree/master/examples/system/ota#run-https-server on how to start a HTTPS server. You can start the webserver using command `openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070` in the the folder where the new binary is located. Hope it helps.

mahavirj, 2023-02-27:
Hopefully above answer helped you to correctly host the file on the HTTPS server. Please prefer using esp32.com forum for the `how-to` questions. Closing.

closed 2023-02-27 by mahavirj