espressif/esp-idf #10768: how to change spi speed without close spi

Dozingfiretruck, 2023-02-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

how to change spi speed without close spi

Icarus113, 2023-02-15:
@Dozingfiretruck  Hi, I assume that your question is `how to change spi speed without deinitialise SPI driver`? So if this is it, you'll need to call [`spi_bus_remove_device`](https://github.com/espressif/esp-idf/blob/master/components/driver/spi/include/driver/spi_master.h#L195), and [`spi_bus_add_device`](https://github.com/espressif/esp-idf/blob/master/components/driver/spi/include/driver/spi_master.h#L183), with an updated speed.

Dozingfiretruck, 2023-02-15:
> @Dozingfiretruck Hi, I assume that your question is `how to change spi speed without deinitialise SPI driver`? So if this is it, you'll need to call [...]

Hi,I tried this method, but asserted, using idf5.0

Icarus113, 2023-02-15:
Would you please help post your error log? I'm not quite sure where the program asserts. @Dozingfiretruck

Dozingfiretruck, 2023-02-15:
> Would you please help post your error log? I'm not quite sure where the program asserts. @Dozingfiretruck

[2023-02-15 16:00:44.400] assert failed: xQueueGenericCreate queue.c:424 (uxQueueLength > ( UBaseType_t ) 0)
[2023-02-15 16:00:44.400] Core  0 register dump:
[2023-02-15 16:00:44.400] MEPC    : 0x4038087e  RA      : 0x40389832  SP      : 0x3fcba140  GP      : 0x3fc93600  
[2023-02-15 16:00:44.400] TP      : 0x3fc87e8c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
[2023-02-15 16:00:44.400] S0/FP   : 0x3fcba2b2  S1      : 0x00000075  A0      : 0x3fcba198  A1      : 0x3fc953b5  
[2023-02-15 16:00:44.400] A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fcb4000  
[2023-02-15 16:00:44.400] A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcba198  S3      : 0x00000001  
[2023-02-15 16:00:44.400] S4      : 0x3fcba198  S5      : 0x00000001  S6      : 0x3fcc18f4  S7      : 0x00000080  
[2023-02-15 16:00:44.400] S8      : 0x00000000  S9      : 0x0100c0a4  S10     : 0x3fca8220  S11     : 0x00000010  
[2023-02-15 16:00:44.400] T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
[2023-02-15 16:00:44.400] MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
[2023-02-15 16:00:44.400] MHARTID : 0x00000000  
[2023-02-15 16:00:44.400] Stack memory:
[2023-02-15 16:00:44.400] 3fcba140: 0x3c105dd7 0x00000001 0x3c102700 0x403903d0 0xa5a5a5a5 0x00000073 0x00000000 0x3fc953b4
[2023-02-15 16:00:44.400] 3fcba160: 0x00343234 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc95574 0x3c102700 0x3fc958ec
[2023-02-15 16:00:44.400] 3fcba180: 0x3c10214a 0x3fc95584 0x3fcba160 0x3fc95588 0x3c1021a4 0x3fc953b4 0x65737361 0x66207472
[2023-02-15 16:00:44.400] 3fcba1a0: 0x656c6961 0x78203a64 0x75657551 0x6e654765 0x63697265 0x61657243 0x71206574 0x65756575
[2023-02-15 16:00:44.400] 3fcba1c0: 0x343a632e 0x28203432 0x75517875 0x4c657565 0x74676e65 0x203e2068 0x42552028 0x54657361
[2023-02-15 16:00:44.400] 3fcba1e0: 0x5f657079 0x20292074 0x3f002930 0x4038e7a0 0x3fcb4de0 0x0000000c 0x3fcb4a30 0x00000054
[2023-02-15 16:00:44.400] 3fcba200: 0x3fcb4de0 0x00000054 0x00001800 0xc0380000 0x00000000 0x3fcb4000 0xc0380000 0x00000060
[2023-02-15 16:00:44.400] 3fcba220: 0x3fcb4de0 0x0000005c 0x3fcb4a30 0x4038e7a0 0x00001800 0x00000054 0x00000000 0x0000005c
[2023-02-15 16:00:44.400] 3fcba240: 0x00000001 0x0000005c 0x00001800 0x40380b10 0x3fca8220 0x0100c0a4 0x00000000 0x00000080
[2023-02-15 16:00:44.400] 3fcba260: 0x3fcc18f4 0x00000001 0x00000000 0x3fcb46f4 0x00000000 0x3fcba318 0x3fcc1a08 0x40389cda
[2023-02-15 16:00:44.400] 3fcba280: 0x00000000 0x3fcba318 0x3fcc1a08 0x42052804 0x3fca8220 0x038040a4 0x00000001 0x3fcc1e08
[2023-02-15 16:00:44.400] 3fcba2a0: 0x00061a80 0x420f1631 0x00000000 0xffffffff 0x00000000 0x04c4b400 0x00000000 0x00000000
[2023-02-15 16:00:44.400] 3fcba2c0: 0x00061a80 0x00000080 0x00000000 0x00000001 0x00061a80 0x3fcc1a18 0x20000000 0x00000008
[2023-02-15 16:00:44.400] 3fcba2e0: 0x00000026 0x00000016 0x3fcba3f8 0x00000002 0x00061a80 0x00000000 0x00000002 0x42039a52
[2023-02-15 16:00:44.400] 3fcba300: 0x00000000 0x00000208 0x00001800 0x3fcb46f4 0x3fca8220 0x0100c0a4 0x00000000 0x00000000
[2023-02-15 16:00:44.400] 3fcba320: 0x00000000 0x00061a80 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[2023-02-15 16:00:44.400] 3fcba340: 0x3fcba3ac 0x3fcb07a8 0x3fcb07a8 0x4203fd12 0x3fcba3ac 0x00000000 0x3fcc1bd0 0x42040a2a
[2023-02-15 16:00:44.400] 3fcba360: 0x00000000 0x00000000 0x3fcc1e20 0x40389e9e 0x00000026 0x00000016 0x3fcba3f8 0x00000002
[2023-02-15 16:00:44.400] 3fcba380: 0x3fca8488 0x3fcba3fc 0x3fca8a64 0x420424d8 0x00000026 0x00000016 0x3fcab870 0x3fcab870
[2023-02-15 16:00:44.400] 3fcba3a0: 0x3c1099e4 0x3c1099b4 0x3fca8a64 0x3fcc1bd0 0x3fca8488 0x0100c0a4 0x20000000 0x00000008
[2023-02-15 16:00:44.400] 3fcba3c0: 0x00000026 0x00000016 0x420202d4 0x00000002 0x3fca8488 0x3fca8220 0x3fca8a64 0x420202f8
[2023-02-15 16:00:44.400] 3fcba3e0: 0x3fcc1bd0 0x01430087 0xfffe0002 0x3fca8220 0x3ffe0001 0x1fffffff 0x3fca8b00 0x3fca8200
[2023-02-15 16:00:44.400] 3fcba400: 0x3fca8488 0x3fca8220 0x3fca8a64 0x4200b598 0x00000000 0x0100c0a4 0xfffe0002 0x00000002
[2023-02-15 16:00:44.400] 3fcba420: 0x3ffe0001 0x1fffffff 0x3fca8b00 0x3fcab8e0 0x3fca8a64 0x00000002 0x3fca84b8 0x42013738
[2023-02-15 16:00:44.400] 3fcba440: 0x3fcab918 0x3fca8500 0x3fca8a64 0x4200fddc 0x00000000 0x3fcaee7c 0x3fca8a64 0x4200b4b0
[2023-02-15 16:00:44.400] 3fcba460: 0xffffffff 0x000000e4 0x20000000 0x00000018 0x3fca8fa0 0x000000e4 0x20000000 0x00000008
[2023-02-15 16:00:44.400] 3fcba480: 0x00000026 0x00000016 0x4202f29c 0x00000001 0x3fcaee04 0x00000000 0x3fca8a64 0x4200afc4
[2023-02-15 16:00:44.400] 3fcba4a0: 0x3fca8a64 0x00000003 0x4200b694 0x3fcba60c 0x00000000 0x3fcaec50 0x4200afba 0x3fca8a64
[2023-02-15 16:00:44.400] 3fcba4c0: 0x00000000 0x3fcaee04 0x00000001 0x4202f29c 0x00000016 0x00000026 0x00000008 0x20000000
[2023-02-15 16:00:44.400] 3fcba4e0: 0x000000e4 0x3fca8fa0 0x00000018 0x3fcba4a0 0x00000000 0x00000000 0x00000000 0x00000000
[2023-02-15 16:00:44.400] 3fcba500: 0x00000001 0x3fcaee04 0x3fca8a30 0x4200afc4 0x3fcaee04 0x00000002 0x42010734 0x3fcba678
[2023-02-15 16:00:44.400] 3fcba520: 0x3fcba730 0x3fcae3a8 0x4200afba 0x3fca8a30 0x3fcaee04 0x00000001 0x00000000 0x00000000

[2023-02-15 16:00:44.400] ELF file SHA256: d0d4dfdcaaa19d81
[2023-02-15 16:00:44.400] 
[2023-02-15 16:00:44.400] Rebooting...

Icarus113, 2023-02-15:
@Dozingfiretruck  From your post, it shows when creating a freertos queue (this is what `spi_bus_add_device` will call, internally), the given `uxQueueLength` which stands for the maximum number of queue items isn't larger than 0. This is unexpected behaviour. 

Could you check the `spi_device_interface_config_t: queue_size` ?

Dozingfiretruck, 2023-02-15:
> @Dozingfiretruck From your post, it shows when creating a freertos queue (this is what `spi_bus_add_device` will call, internally), the given `uxQueueLength` which stands for the maximum number of queue items isn't larger than 0. This is unexpected behaviour.
> 
> Could you check the `spi_device_interface_config_t: queue_size` that you pass to `spi_bus_add_device()`?

Hi,queue_size set 7,the log is

[2023-02-15 16:39:52.178] Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
[2023-02-15 16:39:52.178] 
[2023-02-15 16:39:52.178] Core  0 register dump:
[2023-02-15 16:39:52.178] MEPC    : 0x40382f8e  RA      : 0x40382dce  SP      : 0x3fcba250  GP      : 0x3fc93600  
[2023-02-15 16:39:52.178] TP      : 0x3fc87ebc  T0      : 0xdfffffff  T1      : 0xdf000040  T2      : 0xfffff000  
[2023-02-15 16:39:52.178] S0/FP   : 0x3fcc1a28  S1      : 0xffffff1f  A0      : 0x3fcc1e28  A1      : 0x3fcc20f0  
[2023-02-15 16:39:52.178] A2      : 0x3fcc20f0  A3      : 0x00000000  A4      : 0x3fcb4570  A5      : 0x00000000  
[2023-02-15 16:39:52.178] A6      : 0x0000000c  A7      : 0x00000040  S2      : 0x0000003c  S3      : 0xffffffff  
[2023-02-15 16:39:52.197] S4      : 0x3fcc1934  S5      : 0x00000200  S6      : 0x00000040  S7      : 0x00000001  
[2023-02-15 16:39:52.197] S8      : 0xffffffff  S9      : 0x00000091  S10     : 0x000000fe  S11     : 0x000000ff  
[2023-02-15 16:39:52.197] T3      : 0xff000fff  T4      : 0x00000ffc  T5      : 0x00000fff  T6      : 0xbfffffff  
[2023-02-15 16:39:52.197] MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0xffffff23  
[2023-02-15 16:39:52.197] MHARTID : 0x00000000  
[2023-02-15 16:39:52.197] 
[2023-02-15 16:39:52.197] Stack memory:
[2023-02-15 16:39:52.197] 3fcba250: 0x0000003c 0x3fcc20f0 0x3fcb4a50 0x4038e754 0x0000003c 0x3fcc1914 0x3fcc1a28 0x40382dce
[2023-02-15 16:39:52.197] 3fcba270: 0x0000016f 0x3fcc1e35 0x3fcb4714 0x3fcc1df5 0x0000016f 0x3fcc1e35 0xfffffeef 0x42039574
[2023-02-15 16:39:52.197] 3fcba290: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000200 0x00000200 0x00000000 0x3fcc1df5
[2023-02-15 16:39:52.197] 3fcba2b0: 0x3fcc1df5 0x00000000 0x3fcb07d0 0x42036b8c 0x00000026 0x3fc96d32 0x3fcba418 0x4204f024
[2023-02-15 16:39:52.197] 3fcba2d0: 0x3c110000 0x0000009f 0x3fcb07d0 0x420168b0 0x00000012 0x3fcb07d0 0x0000e9b4 0x00000001
[2023-02-15 16:39:52.197] 3fcba2f0: 0x3c110000 0x00001021 0x00000100 0x00000000 0x00000001 0x0000016f 0x3fcb07d0 0x4203f7e4
[2023-02-15 16:39:52.197] 3fcba310: 0x00000000 0x3fcb4690 0x00000001 0x3fcc1c24 0x00000000 0x04c4b400 0x00000000 0x00000010
[2023-02-15 16:39:52.197] 3fcba330: 0x3fca8220 0x0100c0a4 0x20000000 0x00000008 0x00000026 0x00000016 0x3fcba418 0x00000002
[2023-02-15 16:39:52.197] 3fcba350: 0x3fcba3cc 0x00000000 0x3fcc1bf0 0x42040742 0x3fcb4570 0x00000000 0x3fcc1bf0 0x42040770
[2023-02-15 16:39:52.197] 3fcba370: 0x3fcba3cc 0x00000000 0x3fcc1bf0 0x42040b4e 0x00000000 0x00000000 0x3fcc1e40 0x40389e9e
[2023-02-15 16:39:52.197] 3fcba390: 0x00000026 0x00000016 0x3fcba418 0x00000002 0x3fca8488 0x3fcba41c 0x3fca8a64 0x420425e6
[2023-02-15 16:39:52.197] 3fcba3b0: 0x00000026 0x00000016 0x3fcab870 0x3fcab870 0x3c1099e4 0x3c1099b4 0x3fca8a64 0x3fcc1bf0
[2023-02-15 16:39:52.197] 3fcba3d0: 0x3fca8488 0x0100c0a4 0x20000000 0x00000008 0x00000026 0x00000016 0x420202d4 0x00000002
[2023-02-15 16:39:52.197] 3fcba3f0: 0x3fca8488 0x3fca8220 0x3fca8a64 0x420202f8 0x3fcc1bf0 0x01430087 0xfffe0002 0x3fca8220
[2023-02-15 16:39:52.197] 3fcba410: 0x3ffe0001 0x1fffffff 0x3fca8b00 0x3fca8200 0x3fca8488 0x3fca8220 0x3fca8a64 0x4200b598
[2023-02-15 16:39:52.197] 3fcba430: 0x00000000 0x0100c0a4 0xfffe0002 0x00000002 0x3ffe0001 0x1fffffff 0x3fca8b00 0x3fcab8e0
[2023-02-15 16:39:52.197] 3fcba450: 0x3fca8a64 0x00000002 0x3fca84b8 0x42013738 0x3fcab918 0x3fca8500 0x3fca8a64 0x4200fddc
[2023-02-15 16:39:52.197] 3fcba470: 0x00000000 0x3fcaee7c 0x3fca8a64 0x4200b4b0 0xffffffff 0x000000e4 0x20000000 0x00000018
[2023-02-15 16:39:52.197] 3fcba490: 0x3fca8fa0 0x000000e4 0x20000000 0x00000008 0x00000026 0x00000016 0x4202f29c 0x00000001
[2023-02-15 16:39:52.197] 3fcba4b0: 0x3fcaee04 0x00000000 0x3fca8a64 0x4200afc4 0x3fca8a64 0x00000003 0x4200b694 0x3fcba62c
[2023-02-15 16:39:52.197] 3fcba4d0: 0x00000000 0x3fcaec50 0x4200afba 0x3fca8a64 0x00000000 0x3fcaee04 0x00000001 0x4202f29c
[2023-02-15 16:39:52.197] 3fcba4f0: 0x00000016 0x00000026 0x00000008 0x20000000 0x000000e4 0x3fca8fa0 0x00000018 0x3fcba4c0
[2023-02-15 16:39:52.197] 3fcba510: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000001 0x3fcaee04 0x3fca8a30 0x4200afc4
[2023-02-15 16:39:52.197] 3fcba530: 0x3fcaee04 0x00000002 0x42010734 0x3fcba698 0x3fcba750 0x3fcae3a8 0x4200afba 0x3fca8a30
[2023-02-15 16:39:52.197] 3fcba550: 0x3fcaee04 0x00000001 0x00000000 0x00000000 0x00000000 0x3c1062b0 0x00000000 0xfffe0002
[2023-02-15 16:39:52.197] 3fcba570: 0x3fca8a30 0x3fca8f88 0x00000001 0x3fcba530 0x00000000 0x00000000 0x00000000 0x00000000
[2023-02-15 16:39:52.197] 3fcba590: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[2023-02-15 16:39:52.197] 3fcba5b0: 0x00000000 0x00000000 0x00000000 0x3fcabef8 0x3fcac80c 0x3fcaec50 0x3fcaca64 0x42010cfa
[2023-02-15 16:39:52.197] 3fcba5d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000045 0x00000006 0x000007d0
[2023-02-15 16:39:52.269] 3fcba5f0: 0x3fcae3a8 0x3fca8f78 0x3fca8f60 0x3fcaee04 0x3fca8f70 0x3fca8f60 0x00000000 0x42011cd0
[2023-02-15 16:39:52.269] 3fcba610: 0x00000026 0x01008064 0x20000000 0x4200b814 0x3c10db70 0x3c10db64 0x3fcaee04 0x00000000
[2023-02-15 16:39:52.269] 3fcba630: 0x3fcae320 0x3fca8fb0 0x3fcaee04 0x42036644 0x00000000 0x00000016 0x4202f29c 0xffffffff
[2023-02-15 16:39:52.269] 
[2023-02-15 16:39:52.269] 
[2023-02-15 16:39:52.269] 
[2023-02-15 16:39:52.269] ELF file SHA256: a651af6f156a574d
[2023-02-15 16:39:52.269] 
[2023-02-15 16:39:52.269] Rebooting...

Icarus113, 2023-02-15:
- `MCAUSE : 0x00000005`, this is load access fault.
- `MTVAL : 0xffffff23`, this means the program is accessing a memory that was not initialized.

I suggest you checking your data initialisation. Maybe it's because the `dev_config` isn't initialised?

Dozingfiretruck, 2023-02-15:
> 

It has been initialized and sent and received data

> * `MCAUSE : 0x00000005`, this is load access fault.
> * `MTVAL : 0xffffff23`, this means the program is accessing a memory that was not initialized.
> 
> I suggest you checking your data initialisation. Maybe it's because the `dev_config` isn't initialised?
> 
> @Dozingfiretruck

dev_config has been initialized and sent and received data

Dozingfiretruck, 2023-02-15:
> It has been initialized and sent and received data
>
>
> dev_config has been initialized and sent and received data

[image]
[image]

wanckl, 2023-02-16:
do you check if the index `spi_id-2` is now out the range of array `spi_config`

Dozingfiretruck, 2023-02-16:
> 

> do you check if the index `spi_id-2` is now out the range of array `spi_config`

spi_id is 2 ,so spi_id-2 is 0,that's right

closed 2023-10-02 by Dozingfiretruck